EHfive/pulseaudio-modules-bt #78: [alsa-sink-bcm2835 ALSA] module-loopback.c: Could not peek into queue

gpeter12, 2019-08-30:
Hi!

I'm trying to use a Raspberry Pi for bluetooth sink with aptX-HD. But I'm getting a lot of Synchronization error.

```
[aptx_hd @ 0x1444c70] Synchronization error
[aptx_hd @ 0x1444c70] Synchronization error
[aptx_hd @ 0x1444c70] Synchronization error
[aptx_hd @ 0x1444c70] Synchronization error
[aptx_hd @ 0x1444c70] Synchronization error
[aptx_hd @ 0x1444c70] Synchronization error
I: [alsa-sink-bcm2835 ALSA] module-loopback.c: Could not peek into queue
I: [alsa-sink-bcm2835 ALSA] module-loopback.c: Could not peek into queue
[aptx_hd @ 0x1444c70] Synchronization error
I: [alsa-sink-bcm2835 ALSA] module-loopback.c: Could not peek into queue
[aptx_hd @ 0x1444c70] Synchronization error
[aptx_hd @ 0x1444c70] Synchronization error
I: [alsa-sink-bcm2835 ALSA] module-loopback.c: Could not peek into queue
[aptx_hd @ 0x1444c70] Synchronization error
I: [alsa-sink-bcm2835 ALSA] module-loopback.c: Could not peek into queue
```
The most interesting thing is that on my laptop (also as a sink) with the same code and with (almost) the same libavcodec version it works just perfectly fine. With ZERO error messages in the log. I suspect that the error message comes from the 1039. line of libavcodec's aptx.c. But I assume the problem isn't in libavcodec.

I would really really appreciate the help because I spent almost 3 days on this issue already. It would be a waste...

My setup:
- Asus Zenfone 5Z (it has official support for AptX-HD and AAC)
    - Android 9
    - Bluetooth 5.0
- Raspberry Pi 3 B (with official PSU)
    - Bluetooth 4.1
- Laptop: Asus N550JK (Intel Core I7 4700HQ version)
    - Ubuntu 19.04
    - Bluetooth 4.0
- libavcodec-dev:armhf (7:4.1.4-1+rpt1~deb10u1)
- pulseaudio-modules-bt version is the currently most recent 3ecf1d7
- pulseaudio version is pulseaudio 12.2-dirty (git: ee910cf6d) (because with the current master I couldn't get it work at all)

asavah, 2019-08-30:
@gpeter12 raspberry pi 3 built-in bluetooth seems unusable for a2dp , see https://github.com/raspberrypi/firmware/issues/887 ,  issues have been open for years without any reaction or even visible intention to fix by the vendor(s)

gpeter12, 2019-08-30:
Is it possible to this issue only apllies to RPi 3B? Because on the last week I succesfully managed to make a really stable bluetooth sink with RPi Zero W with DietPi and BlueAlsa. The only problem is that BlueAlsa is only SBC compatibile. I do this "project" only for aptX-HD sink.

closed 2019-11-05 by stale[bot]