EHfive/pulseaudio-modules-bt #79: modules fail to load

daenney, 2019-09-06:
I'm trying to switch over to pulseaudio-modules-bt but I seem to get into trouble loading the new modules

```
~/D/u/bin $ pacman -Q pulseaudio-modules-bt-git
pulseaudio-modules-bt-git r141.3ecf1d7-1
~/D/u/bin $ pacman -Q pulseaudio
pulseaudio 12.2-2
~/D/u/bin $ pacman -Q libldac
libldac 2.0.2.3-2
```

The default config on Arch includes:

```
### Automatically load driver modules for Bluetooth hardware
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover
.endif
```

However, I don't see any bluez related sink, though bluetooth is active and a device is paired:

```
~/D/u/bin $ bluetoothctl
Agent registered
[CHG] Controller FC:77:74:A3:E5:11 Pairable: yes
[Daenney’s AirPods]# paired-devices
Device BC:B8:63:XX:XX:XX Daenney’s AirPods
```

If I look into the systemd journal for Pulse I see:

```
Sep 06 14:03:30 yunikoni pulseaudio[5157]: E: [pulseaudio] a2dp_aac.c: Invalid object type in AAC configuration
Sep 06 14:03:30 yunikoni pulseaudio[5157]: E: [pulseaudio] bluez5-util.c: Endpoint SetConfiguration(): invalid arguments
```

daenney, 2019-09-06:
I've tried this with both bluez and bluez-git, the result is the same

EHfive, 2019-09-06:
Your headphone is not compatible with this module.

Plz post you result of `avinfo`. #31

daenney, 2019-09-06:
```
$ avinfo BC:B8:63:19:49:34
Connecting ... 
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-250
Stream End-Point #2: Audio Sink 
        Media Codec: MPEG24
                Object Types: MPEG-2 AAC LC 
                Frequencies: 44.1kHz 48kHz 
                Channels: 1 2 
                Bitrate: 256000
                VBR: Yes
```

The device in particular are Apple AirPods

MateuszKubuszok, 2019-09-13:
I might have similar problem. After I upgraded by Arch yesterday, my headset stopped connecting (at all) and when I tried to debug it, I also got error messages saying that I cannot load/initialize `module-bluetooth-discover` and `module-bluetooth-policy`. Perhaps there was a breaking change somewhere?

EHfive, 2019-09-13:
@MateuszKubuszok No, it's not your case. Arch recently updates pulseaudio from 12.2 to 12.99 in stable branch, so you have to rebuild `pulseaudio-modules-bt-git` from PKGBUILD.

ekaradon, 2019-09-16:
@EHfive I think we cannot anymore build from the current package as Arch is now using the pulseaudio 13.0-2 version. And so we have the following libraries updated:

```
[  5%] Linking C shared library libbluez5-util.so
/usr/bin/ld: cannot find -lpulsecommon-12.2
/usr/bin/ld: cannot find -lpulsecore-12.2
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/bluez5-util.dir/build.make:219: libbluez5-util.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:138: CMakeFiles/bluez5-util.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
==> ERROR: A failure occurred in build().
```

- libpulsecommon-13.0.so
- libpulsecore-13.0.so

EHfive, 2019-09-16:
@ekaradon Try `makepkg --clean` or just make sure previous cached sources is deleted. (A clean build)

ekaradon, 2019-09-16:
Using the option was not enough but with the last update (9f17638..5022ac1) and by removing everything in the directory it did work. Thanks @EHfive!

dbernheisel, 2019-09-16:
Works for me:

```sh
rm -rfv ~/.cache/yay/pulseaudio-modules-bt-git
yay pulseaudio-modules-bt-git
reboot
```

barraponto, 2019-09-18:
If your package builder caches (just keeps) the build folder, `git submodule update` should fix it.
(although I fixed it as mentioned above)

closed 2020-01-14 by stale[bot]