EHfive/pulseaudio-modules-bt #80: Modules not working with pulse audio 13.0

alex14641, 2019-09-18:
The code compiles OK, but the modules don't load. Also when I run the pulseaudio command, I get the following error: 
pulseaudio: symbol lookup error: pulseaudio: undefined symbol: pa_a2dp_codec_sbc

mirlang, 2019-09-19:
similar issue here: pulseaudio 13 segfaults while loading the module

EHfive, 2019-09-19:
https://github.com/EHfive/pulseaudio-modules-bt/issues/79#issuecomment-531252991
> Arch recently updates pulseaudio from 12.2 to 12.99 in stable branch, so you have to rebuild pulseaudio-modules-bt-git from PKGBUILD.

Now it's 13.0.

You should run a cleanbuild. https://github.com/EHfive/pulseaudio-modules-bt/issues/79#issuecomment-531755460 https://github.com/EHfive/pulseaudio-modules-bt/issues/79#issuecomment-531776823

EHfive, 2019-09-19:
https://github.com/search?q=pa_a2dp_codec_sbc&type=Code

You probably messed up official bluez modules with `pulseaudio-modules-bt` modules.

There is no symbol named `pa_a2dp_codec_sbc` in https://github.com/EHfive/pulseaudio-modules-bt/search?q=pa_a2dp_codec_sbc&unscoped_q=pa_a2dp_codec_sbc

Make sure you have followed all steps of https://github.com/EHfive/pulseaudio-modules-bt#build.

Q&A may also gives you help. https://github.com/EHfive/pulseaudio-modules-bt/wiki#q--a

Close.

EHfive, 2019-09-19:
@mirlang Just rebuild pulseaudio-modules-bt-git from PKGBUILD.

alex14641, 2019-09-21:
@EHfive  I'm running Slackware current. The problem is that PulseAudio 13.0 expects that symbol to be present; while PA 12.x did not: the symbol is present in the original PA 13.0 code to build libbluez5-util.so.

EHfive, 2019-09-21:
Pulseaudio modules can be dynamically loaded or unloaded,
```
$ pactl list modules short | grep bluez
10      module-bluez5-discover
$ pactl unload-module module-bluez5-discover
$ pactl load-module module-bluez5-discover
``` 

The missing symbol only present in modules, otherwise your pulseaudio would not even running.

You might messed original BT modules with these modules.

You need to repace all of these files:
```
/usr/lib/pulse-13.0/
/usr/lib/pulse-13.0/modules/
/usr/lib/pulse-13.0/modules/libbluez5-util.so
/usr/lib/pulse-13.0/modules/module-bluetooth-discover.so
/usr/lib/pulse-13.0/modules/module-bluetooth-policy.so
/usr/lib/pulse-13.0/modules/module-bluez5-device.so
/usr/lib/pulse-13.0/modules/module-bluez5-discover.so
```

Aerol, 2019-09-21:
On Slackware64-current, I get the error trying to run "pulseaudio -k".
Install log from make install:

```
-- Installing: /usr/lib64/pulse-13.0/modules/libbluez5-util.so                                                                                                                                      
-- Set runtime path of "/usr/lib64/pulse-13.0/modules/libbluez5-util.so" to "/usr/lib64/pulseaudio:/usr/lib64/pulse-13.0/modules"                                                                   
-- Installing: /usr/lib64/pulse-13.0/modules/module-bluez5-discover.so                                                                                                                              
-- Set runtime path of "/usr/lib64/pulse-13.0/modules/module-bluez5-discover.so" to "/usr/lib64/pulseaudio:/usr/lib64/pulse-13.0/modules"                                                           
-- Installing: /usr/lib64/pulse-13.0/modules/module-bluez5-device.so                                                                                                                                
-- Set runtime path of "/usr/lib64/pulse-13.0/modules/module-bluez5-device.so" to "/usr/lib64/pulseaudio:/usr/lib64/pulse-13.0/modules"                                                             
-- Installing: /usr/lib64/pulse-13.0/modules/module-bluetooth-discover.so                                                                                                                           
-- Set runtime path of "/usr/lib64/pulse-13.0/modules/module-bluetooth-discover.so" to "/usr/lib64/pulseaudio:/usr/lib64/pulse-13.0/modules"                                                        
-- Installing: /usr/lib64/pulse-13.0/modules/module-bluetooth-policy.so                                                                                                                             
-- Set runtime path of "/usr/lib64/pulse-13.0/modules/module-bluetooth-policy.so" to "/usr/lib64/pulseaudio:/usr/lib64/pulse-13.0/modules"
```

alex14641, 2019-09-21:
@EHfive 
I followed the install instructions (Part A) verbatim. I deleted the pulseaudio-modules-bt directory, and  copied and pasted the instructions to avoid error. The problem is that the pulseaudio executable from 13.0 expects the libbluez5-util module to have the missing symbol. The stock libbluez5-util  module has this symbol: it comes from ./pa/src/modules/bluetooth/a2dp-codec-sbc.c; the libbluez5-util  module in your code doesn't have the symbol, hence the problem.

rj-jesus, 2019-09-23:
I have this same issue too.

pacho2, 2019-09-29:
It also segfaults in Gentoo, even after rebuilding media-sound/pulseaudio-modules-bt with pulseaudio-13.0

pacho2, 2019-09-29:
I get this:
Reading symbols from pulseaudio...
Reading symbols from /usr/lib64/debug//usr/bin/pulseaudio.debug...
(gdb) r
Starting program: /usr/bin/pulseaudio 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
W: [pulseaudio] pid.c: Stale PID file, overwriting.
[New Thread 0x7fffe8062700 (LWP 2343)]
[New Thread 0x7fffe7861700 (LWP 2344)]

Thread 1 "pulseaudio" received signal SIGSEGV, Segmentation fault.
pa_hook_connect (hook=0x555555579770, prio=PA_HOOK_NORMAL, cb=0x7fffe705c580, data=0x555555751f50) at /var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/pulsecore/hook-list.c:71
71	/var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/pulsecore/hook-list.c: No existe el fichero o el directorio.
(gdb) bt
#0  0x00007ffff7f49780 in pa_hook_connect (hook=0x555555579770, prio=PA_HOOK_NORMAL, cb=0x7fffe705c580, data=0x555555751f50)
    at /var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/pulsecore/hook-list.c:71
#1  0x00007fffe705ced5 in module_bluetooth_policy_LTX_pa__init () at /usr/lib64/pulse-13.0/modules/module-bluetooth-policy.so
#2  0x00007ffff7f4bbe6 in pa_module_load (module=module@entry=0x7fffffffcaf0, c=c@entry=0x5555555793f0, name=name@entry=0x5555556cbdf0 "module-bluetooth-policy", argument=0x0)
    at /var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/pulsecore/module.c:191
#3  0x00007ffff7f3831b in pa_cli_command_load (c=0x5555555793f0, t=0x5555556c9220, buf=0x55555556fba0, fail=0x55555556e4b5)
    at /var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/pulsecore/cli-command.c:437
#4  0x00007ffff7f4002e in pa_cli_command_execute_line_stateful
    (c=c@entry=0x5555555793f0, s=s@entry=0x7fffffffcc50 "load-module module-bluetooth-policy", buf=buf@entry=0x55555556fba0, fail=fail@entry=0x55555556e4b5, ifstate=ifstate@entry=0x7fffffffcc4c)
    at /var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/pulsecore/cli-command.c:2141
#5  0x00007ffff7f407e1 in pa_cli_command_execute_file_stream (c=0x5555555793f0, f=0x55555556e890, buf=0x55555556fba0, fail=0x55555556e4b5)
    at /var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/pulsecore/cli-command.c:2181
#6  0x000055555555b520 in main (argc=<optimized out>, argv=<optimized out>) at /var/tmp/portage/media-sound/pulseaudio-13.0/work/pulseaudio-13.0/src/daemon/main.c:1100
(gdb)

EHfive, 2019-10-02:
I figured out `--enable-force-preopen` is actived in Slackware64-current pulseaudio build, resulting all modules are linked to `pulseaudio` executeable, as well modules symbol.
```
# ldd pulseaudio | grep module-blu
	module-bluetooth-discover.so => /usr/lib64/pulse-13.0/modules/module-bluetooth-discover.so (0x00007f94bd28a000)
	module-bluetooth-policy.so => /usr/lib64/pulse-13.0/modules/module-bluetooth-policy.so (0x00007f94bd283000)
	module-bluez5-discover.so => /usr/lib64/pulse-13.0/modules/module-bluez5-discover.so (0x00007f94bd25f000)
	module-bluez5-device.so => /usr/lib64/pulse-13.0/modules/module-bluez5-device.so (0x00007f94bd24e000)
```

You have to rebuild pulseaudio with `--disable-force-preopen` configure flag to resovle this.

https://mirrors.slackware.com/slackware/slackware64-current/source/l/pulseaudio/pulseaudio.SlackBuild

EHfive, 2019-10-02:
`pulseaudio` symbols with `--enable-force-preope` actived:
```
# objdump -t pulseaudio | grep sbc
00000000004402e0 g     O .data.rel.ro	0000000000000088              pa_a2dp_codec_sbc

# objdump -t pulseaudio | grep  bluez5
000000000041c410       F *UND*	0000000000000000              module_bluez5_device_LTX_pa__get_description
000000000041c6a0       F *UND*	0000000000000000              module_bluez5_device_LTX_pa__init
000000000041ca10       F *UND*	0000000000000000              module_bluez5_discover_LTX_pa__get_version
000000000041d110       F *UND*	0000000000000000              module_bluez5_discover_LTX_pa__init
000000000041daf0       F *UND*	0000000000000000              module_bluez5_discover_LTX_pa__get_description
000000000041e1c0       F *UND*	0000000000000000              module_bluez5_device_LTX_pa__done
000000000041e460       F *UND*	0000000000000000              module_bluez5_discover_LTX_pa__load_once
000000000041ecf0       F *UND*	0000000000000000              module_bluez5_device_LTX_pa__get_version
...
```

As well other modules.

alex14641, 2019-10-02:
I'm going to check to see if pulseaudio 12.x was built with --enable-force-preopen.

alex14641, 2019-10-03:
Verified that PA 12.2 had the flag as well.

rj-jesus, 2019-10-03:
I can confirm that with `--disable-force-preopen` I can compile and load the modules. So far I haven't encountered any other issues.

mirlang, 2019-10-03:
> It also segfaults in Gentoo, even after rebuilding media-sound/pulseaudio-modules-bt with pulseaudio-13.0

the pulseaudio-modules-bt ebuild in gentoo simply misses a step to update the pa-submodule to the right version, so it's always compiled against pulseaudio v12.2 (which is checked out by default), that's why it segfaults

alex14641, 2019-10-03:
The real issue here is that PA changed the module API in 13.0: the modules should be rewritten. I may look into this if I get some free time.

EHfive, 2019-10-04:
No offense, not sure what you mean. 

I think the real issue is that official BT module symbols are different with this modules which causes symbol missing when pulseaudio is built with `force preopen` making all `dlopen` able modules symbol address appended to `pulseaudio`.

https://github.com/EHfive/pulseaudio-modules-bt/blob/d5b87e80cc8a294bbd949f9b89e8eb92f01b8ccf/src/modules/bluetooth/module-bluez5-device.c#L1339-L1344

module API changes in 13.0 has been handled already.

-------

Both issues in this thread are related to packaging, and root causes are found.
So I am going to close this.

closed 2019-10-04 by EHfive

alex14641, 2019-10-04:
@EHfive 

I disagree with your assessment. The built in modules work with --disable-force-preopen enabled. I believe these should work, as well,

EHfive, 2019-10-04:
> @EHfive
> 
> I disagree with your assessment. The built in modules work with --disable-force-preopen enabled. I believe these should work, as well,

I can't describe details in Eng; You can't separately rebuild a shared library which been dynamically linked because symbol address change.

I will make pulseaudio built with `--disable-force-preopen` as a prerequisite in README.