EHfive/pulseaudio-modules-bt #82: Pulseaudio a2dp_sink_aac profile unavailable for AAC compatible headset

cahorn, 2019-09-24:
Thanks to all who contribute to this awesome project.

Unfortunately, my Soundcore Spirit X headset, which is compatible with AAC (see FAQ, page 2 at: https://www.soundcore.com/products/variant/spirit-x/A3451011), does not appear to be supported.

Using latest bluez from git: yes
Using latest pulseaudio-modules-bt from git: yes
Bluez compiled with `--enable-experimental`: yes
`bluetoothd` executed with flag `-E`: yes

Pulseaudio states the `a2dp_sink_aac` profile is unavailable:

    $ pactl list cards
    Card #2
        Name: bluez_card.E8_07_BF_14_84_DA
        Driver: module-bluez5-device.c
        Owner Module: 25
        Properties:
                device.description = "Soundcore Spirit X "
                device.string = "E8:07:BF:14:84:DA"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_E8_07_BF_14_84_DA"
                bluez.class = "0x240404"
                bluez.alias = "Soundcore Spirit X "
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
        Profiles:
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink_sbc
        Ports:
                headset-output: Headset (priority: 0, latency offset: 0 usec, available)
                        Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac
                headset-input: Headset (priority: 0, latency offset: 0 usec)
                        Part of profile(s): headset_head_unit

Attempting to change to the `a2dp_sink_aac` profile errors:

    $ pactl set-card-profile 2 a2dp_sink_aac
    Failure: Invalid argument

... Giving the following system error:

    $ journalctl -f
    ...
    Sep 24 12:05:47 <hostname> pulseaudio[17430]: Refused to switch profile to a2dp_sink_aac: Unavailable profile

Any pointers for further troubleshooting would be much appreciated; thank you in advance.

EHfive, 2019-09-25:
Do #31，post result here

cahorn, 2019-09-25:
Thank you for the prompt reply. Unfortunately, it looks like `bluez` thinks that the headset only supports SBC:

    $ avinfo <device>
    Connecting ... 
    Stream End-Point #1: Audio Sink 
            Content Protection: SCMS-T
            Media Codec: SBC
                    Channel Modes: Mono DualChannel Stereo JointStereo
                    Frequencies: 32Khz 44.1Khz 48Khz 
                    Subbands: 4 8
                    Blocks: 4 8 12 16 
                    Bitpool Range: 8-53

I'm left wondering if this is an issue with `bluez`'s method of communicating with the device or if this is grounds for an accusation of false advertising on the part of the headset manufacturer... :(

closed 2019-09-27 by EHfive