EHfive/pulseaudio-modules-bt #83: didn't work on pulseaudio 13.0

YangMame, 2019-10-07:
System: gentoo with unstable 
```
mame@Laptop ~ $ pulseaudio --version
pulseaudio 13.0
mame@Laptop ~ $ tail -f /etc/pulse/default.pa 

# Bluetooth
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover
.endif

mame@Laptop ~ $ LANG="en_US.UTF" pulseaudio --start
E: [pulseaudio] main.c: Daemon startup failed.
```

EHfive, 2019-10-07:
https://github.com/EHfive/pulseaudio-modules-bt/issues/80#issuecomment-537955189
> the pulseaudio-modules-bt ebuild in gentoo simply is missing a step to update the pa-submodule to the right version, so it's always compiled against pulseaudio v12.2 (which is checked out by default), that's why it segfaults

Need someone to issue a bug on https://packages.gentoo.org/packages/media-sound/pulseaudio-modules-bt.

Jjeje007, 2019-10-07:
Bug report opened : [https://bugs.gentoo.org/696912](url)

EHfive, 2019-10-07:
Current workaround:

This way you can force checkout pulseaudio v13.0
```
# export EGIT_OVERRIDE_COMMIT_PULSEAUDIO_PULSEAUDIO=v13.0
# USE="dbus" emerge media-sound/pulseaudio media-sound/pulseaudio-modules-bt 
```

Pulseaudio v13 ebuild disable `dbus` if `bluetooth` is unused but `pulseaudio-modules-bt` still requires it, therefore you need to append `dbus` USE flag to `media-sound/pulseaudio`

EHfive, 2019-10-07:
https://github.com/EHfive/pulseaudio-modules-bt/issues/83#issuecomment-539131879
> Bug report opened : https://bugs.gentoo.org/696912

https://github.com/EHfive/pulseaudio-modules-bt/issues/83#issuecomment-539173067
> Current workaround:
> ........

@pacho2 I thought you could help on the essay.

pacho2, 2019-10-20:
It works fine

Thanks a lot for your help!

EHfive, 2019-11-08:
https://bugs.gentoo.org/696912#c4

Close.

closed 2019-11-08 by EHfive