EHfive/pulseaudio-modules-bt #85: PPA Ubuntu 19.10

zylxpl, 2019-10-18:
Ubuntu 19.10 is out. Will it be supporte via PPA?

lucasfijen, 2019-10-18:
After my upgrade to 19.10 from 19.04 ldac stopped working. When I try to connect a bluetooth headset I get the following error:
Connection Failed: blueman.bluez.errors.DBusFailedError: Protocol not available...

My guess is that with the upgrade pulseaudio is updated to version 13. However the PPA for 19.04 is still at  version 1.1.99-3, while pulseaudio-modules-bt v1.2 has in its release notes compatibility with pulseaudio 13.

berglh, 2019-10-23:
Please save me from myself, Eoan support will be very welcome.

Dark-Passenger, 2019-11-01:
followed the build instructions just updating the pa submodule to v13.0 tag and it's working fine so far.

non5ense, 2019-11-07:
> followed the build instructions just updating the pa submodule to v13.0 tag and it's working fine so far.

Could you please post the particular command set for this?

berglh, 2019-11-08:
@non5ense, here, hold my :beer:
> Could you please post the exact command set for this?

I read the main readme and ended up with these. It's possible you may need to troubleshoot other packages/modules you don't have available that I had already installed. This will probably fail in the `cmake` or `make` stages.

**Note**: Steps 1 and 2 may be superfluous if you already had it installed previously.

1. Add existing repo to install some of the required libraries
```bash
sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
```
2. Change eoan to disco to install old packages 
```bash
sudo vi /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
sudo apt update
sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt
```
3. Install build dependencies (these are the ones I needed that I didn't already have):
```bash
sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential python3-dev libdbus-glib-1-dev libgirepository1.0sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential python3-dev libdbus-glib-1-dev libgirepository1.0-dev pkg-config libtool sbc-tools libtool cmake-dev pkg-config libtool sbc-tools libtool cmake
```
4. Backup existing bluez/bluetooth modules:
```bash
sudo find /usr/lib -regex ".*\(bluez5\|bluetooth\).*\.so.bak" 
```
5. Checkout repo into a location you like building in and cd into that folder
```bash
git clone https://github.com/EHfive/pulseaudio-modules-bt.git
cd pulseaudio-modules-bt/
```
6. The pa folder is a _link_ to the official pulse audio repository. Update the pulseaudio submodule to point to the v13.0 tag of the pulse audio repo:
```bash
git submodule update --init
cd pa/
git stash (if you there is anything uncommited)
git fetch
git fetch --tags
git checkout v13.0
```
7. Build the bluetooth modules
```bash
cd ..
mkdir build
cd build/
cmake -DFORCE_LARGEST_PA_VERSION=ON ..
cd ..
cmake ..
make
```
8. Install the modules & restart pulseaudio
```bash
sudo make install
pulseaudio -k
pulseaudio --start
```
9. I had to restart bluetooth service cause it was being silly and repair my headphones

```bash
sudo systemctl restart bluetooth
```
Enjoy the AptX/LDAC goodness, until it breaks again or the `eoan` repo support is added by our developer overlords. :smiling_imp:

Lastique, 2019-11-08:
JFYI, doing `make install` is a bad idea as it will break your system eventually because of outdated stray binaries laying in your `/usr`.

berglh, 2019-11-23:
@EHfive Is there someway I can contribute to packaging this for `eoan`?

EHfive, 2019-11-23:
> @EHfive Is there someway I can contribute to packaging this for `eoan`?

It would be help if you could maintain `pulseaudio-modules-bt`  and `libldac` in your own PPA/source.

You can find original packaging infos in https://launchpad.net/~eh5/+archive/ubuntu/pulseaudio-a2dp/+packages

I just don't have enough time to do OS installing / packaging / testing for each Ubuntu release.

berglh, 2019-11-28:
@EHfive Do you have any recommendations on testing?

If I am going to build using CI/CD methods and publish the result to `launchpad`, apart from testing successful installation, it is difficult to actually test `libldac` and `pulseaudio-modules-bt` with audio devices for automated building, unless I do so on my own laptop (time consuming, space considerations with VMs)

I have some time coming up over December break, I might be able to spend some time to get a PPA going then. If you have any advice on how to get started, it would be appreciated.

EHfive, 2019-11-28:
@berglh 

Unfortunately，you can only do runtime tests.

### runtime test (if you can run bluez in your CI/CD testing instance)

0. start bluez & pulseaudio

1. test if modules can be loaded

```bash
$ pulseaudio --start
$ pactl unload-module module-bluetooth-discover
$ pactl unload-module module-bluetooth-policy
$ pactl load-module module-bluetooth-policy
$ pactl load-module module-bluetooth-discover
```

2. test if A2DP endpoints are registered by checking logs dumped by bluez or pulseaudio
```
$ journalctl --user --boot --no-tail --unit pulseaudio | grep register_endpoint

Nov 28 18:45:05 eh5-arch pulseaudio[3120]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSink/SBC on adapter /org/bluez/hci0
Nov 28 19:17:33 eh5-arch pulseaudio[3729]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSource/VENDOR/LDAC on adapter /org/bluez/hci0
Nov 28 19:17:33 eh5-arch pulseaudio[3729]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSource/VENDOR/APTXHD on adapter /org/bluez/hci0
Nov 28 19:17:33 eh5-arch pulseaudio[3729]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSource/VENDOR/APTX on adapter /org/bluez/hci0
Nov 28 19:17:33 eh5-arch pulseaudio[3729]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSource/SBC on adapter /org/bluez/hci0
Nov 28 19:17:33 eh5-arch pulseaudio[3729]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSink/VENDOR/APTXHD on adapter /org/bluez/hci0
Nov 28 19:17:33 eh5-arch pulseaudio[3729]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSink/VENDOR/APTX on adapter /org/bluez/hci0
Nov 28 19:17:33 eh5-arch pulseaudio[3729]: D: [pulseaudio][modules/bluetooth/bluez5-util.c:1191 register_endpoint()] Registering /MediaEndpoint/A2DPSink/SBC on adapter /org/bluez/hci0
```

ValZapod, 2019-12-06:
@mirh @berglh Can you also support debian or this is only tested for Ubuntu? Please))))))

solalatus, 2019-12-14:
Just to let you know: sadly the above manual method is broken. I've followed the instructions by @berglh and now have no sound over Bluetooth at all. Any PPA update would be much appreciated! Thx!

hellfi, 2019-12-15:
> Just to let you know: sadly the above manual method is broken. I've followed the instructions by @berglh and now have no sound over Bluetooth at all. Any PPA update would be much appreciated! Thx!

For me with 19.10/aeon this worked:
_sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt \
libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential \
libdbus-glib-1-dev libgirepository1.0-dev pkg-config libtool sbc-tools libtool pkg-config \
libtool sbc-tools libtool cmake libdbus-1-dev_

(this WILL install unneeded packages, but I'm not sorting them out now)
 and then instructions from install A. build for PulseAudio releases (e.g., v12.0, v12.2, etc.)
git clone https://github.com/EHfive/pulseaudio-modules-bt.git
cd pulseaudio-modules-bt
git submodule update --init
git -C pa/ checkout v`pkg-config libpulse --modversion|sed 's/[^0-9.]*\([0-9.]*\).*/\1/'`

mkdir build && cd build
cmake ..
make
sudo make install

pulseaudio -k
_AND_
pulseaudio --start

zylxpl, 2019-12-18:
> @non5ense, here, hold my 
> 
> > Could you please post the exact command set for this?
> 
> I read the main readme and ended up with these. It's possible you may need to troubleshoot other packages/modules you don't have available that I had already installed. This will probably fail in the `cmake` or `make` stages.
> 
> **Note**: Steps 1 and 2 may be superfluous if you already had it installed previously.
> 
> 1. Add existing repo to install some of the required libraries
> 
> ```shell
> sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
> ```
> 
> 1. Change eoan to disco to install old packages
> 
> ```shell
> sudo vi /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
> sudo apt update
> sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt
> ```
> 
> 1. Install build dependencies (these are the ones I needed that I didn't already have):
> 
> ```shell
> sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential python3-dev libdbus-glib-1-dev libgirepository1.0sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential python3-dev libdbus-glib-1-dev libgirepository1.0-dev pkg-config libtool sbc-tools libtool cmake-dev pkg-config libtool sbc-tools libtool cmake
> ```
> 
> 1. Backup existing bluez/bluetooth modules:
> 
> ```shell
> sudo find /usr/lib -regex ".*\(bluez5\|bluetooth\).*\.so.bak" 
> ```
> 
> 1. Checkout repo into a location you like building in and cd into that folder
> 
> ```shell
> git clone https://github.com/EHfive/pulseaudio-modules-bt.git
> cd pulseaudio-modules-bt/
> ```
> 
> 1. The pa folder is a _link_ to the official pulse audio repository. Update the pulseaudio submodule to point to the v13.0 tag of the pulse audio repo:
> 
> ```shell
> git submodule update --init
> cd pa/
> git stash (if you there is anything uncommited)
> git fetch
> git fetch --tags
> git checkout v13.0
> ```
> 
> 1. Build the bluetooth modules
> 
> ```shell
> cd ..
> mkdir build
> cd build/
> cmake ..
> make
> ```
> 
> 1. Install the modules & restart pulseaudio
> 
> ```shell
> sudo make install
> pulseaudio -k
> pulseaudio --start
> ```
> 
> 1. I had to restart bluetooth service cause it was being silly and repair my headphones
> 
> ```shell
> sudo systemctl restart bluetooth
> ```
> 
> Enjoy the AptX/LDAC goodness, until it breaks again or the `eoan` repo support is added by our developer overlords.

> For me with 19.10/aeon this worked:
> _sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential libdbus-glib-1-dev [...]

I was able to successfully build modules using both of your guides. However, there are couple more dependencies needed to compile with aptx/HD/ldac/aac support:
`sudo apt install libavcodec-dev libfdk-aac-dev libldac-dev libavutil-dev`

Without it modules gonna build, but without suport for specific codecs.

berglh, 2019-12-18:
Thanks for the feedback. I'm still trying to determine how I can automate build testing so I can produce deb for each dist. I think I can do it on Docker but I guess I need a physical Bluetooth adapter for it to work.

@EHfive have you come across a method for adding a pseudo/fake Bluetooth device in Linux?

solalatus, 2019-12-19:
@zylxpl This did the trick! Many thanks! (Complete brokenness was because of other reasons, KDE and Gnome dual install running two pulses at  parallel...)

ningfei, 2019-12-24:
Built & packaged deb in case anyone needs it.
[pulseaudio-modules-bt.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/3999110/pulseaudio-modules-bt.zip) contains `pulseaudio-modules-bt_1.1.99-3~eoan1_amd64.deb`

```
sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
sudo vim /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
sudo apt update
sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt
sudo dpkg -i pulseaudio-modules-bt_1.1.99-3~eoan1_amd64.deb
```

berglh, 2019-12-26:
OK, some news. I was hacking around this afternoon and in the Bluez repo I found this directory:
[https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/emulator](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/emulator)

After building this with the `--enable-testing` configuration flag, and running `btvirt`, I managed to get a virtual bluetooth adaptor in my machine:
```
$ sudo hciconfig -a
[sudo] password for [me]: 
hci1:   Type: Primary  Bus: Virtual
        BD Address: 00:AA:01:00:00:23  ACL MTU: 192:1  SCO MTU: 0:0
        UP RUNNING 
        RX bytes:0 acl:0 sco:0 events:61 errors:0
        TX bytes:4935 acl:0 sco:0 commands:61 errors:0
        Features: 0xa4 0x08 0x08 0xc0 0x58 0x1e 0x7b 0x83
        Packet type: DM1 DH1 HV1 
        Link policy: RSWITCH SNIFF 
        Link mode: SLAVE ACCEPT 
        Name: 'bxps #2'
        Class: 0x1c010c
        Service Classes: Rendering, Capturing, Object Transfer
        Device Class: Computer, Laptop
        HCI Version: 5.0 (0x9)  Revision: 0x0
        LMP Version: 5.0 (0x9)  Subversion: 0x0
        Manufacturer: Bluetooth SIG, Inc. (63)
```
bluetoothctl also shows this device as being present:
```
Controller 00:AA:01:00:00:23 mymachine #2 [default]  < this is the virtual adaptor
Controller 9C:B6:D0:FF:5E:3C mymachine 
```
Anyway, what this means is I will be able to run this binary to emulate bluetooth on my hosted server that doesn't have a bluetooth adaptor and automate the build using a privileged docker container that also has systemd installed. I *should* be able to generate binaries for each supported LTS and the current rolling release with specific containers for each.

So my next step is to create a repo to house the build scripts in and do some testing with the virtual BT adaptor with Docker. I should be able to satisfy the build tests @EHfive presented above all in Docker :thinking: 

After generating the debs for installation, I'll need to setup a PPA on launchpad, which I haven't done before and then create some post-deb build steps to publish the files directly to the PPA. Hopefully it will end up being a one click build and publish with the only intermission being dependency and build error debugging.

Anyway, thanks for all the work above, this is a great template for me to generate the debs for `eoan` and avoid some of the errors I made with my manual install (on my work machine :joy:) that you folks ironed out. :pray:

berglh, 2019-12-26:
So more Docker hacking and I have the emulated bluetooth device functioning which results in a working `systemd` controlled bluez service inside of a Docker container.  I'm kind of surprised this actually works.
```bash
/tmp/bluez/emulator# ./btvirt -l 1 &
[1] 16400
/tmp/bluez/emulator# hciconfig -a
hci0:   Type: Primary  Bus: Virtual
        BD Address: 00:AA:01:00:00:23  ACL MTU: 192:1  SCO MTU: 0:0
        DOWN 
        RX bytes:0 acl:0 sco:0 events:35 errors:0
        TX bytes:431 acl:0 sco:0 commands:35 errors:0
        Features: 0xa4 0x08 0x08 0xc0 0x58 0x1e 0x7b 0x83
        Packet type: DM1 DH1 HV1 
        Link policy: RSWITCH SNIFF 
        Link mode: SLAVE ACCEPT 
/tmp/bluez/emulator# systemctl start bluetooth
/tmp/bluez/emulator# systemctl status bluetooth
* bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-12-26 09:22:56 UTC; 9s ago
     Docs: man:bluetoothd(8)
 Main PID: 16411 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   Memory: 1.5M
   CGroup: /system.slice/containerd.service/system.slice/bluetooth.service
           -16411 /usr/lib/bluetooth/bluetoothd

Dec 26 09:22:56 systemd[1]: Starting Bluetooth service...
Dec 26 09:22:56 bluetoothd[16411]: Bluetooth daemon 5.50
Dec 26 09:22:56 systemd[1]: Started Bluetooth service.
Dec 26 09:22:56 bluetoothd[16411]: Starting SDP server
Dec 26 09:22:56 bluetoothd[16411]: Bluetooth management interface 1.14 initialized
/tmp/bluez/emulator# bluetoothctl
Agent registered
[bluetooth]# power on
Changing power on succeeded
```
I can now move on to building `pulseaudio-modules-bt` with the instructions above :raised_hands:

libussa, 2019-12-29:
> Built & packaged deb in case anyone needs it.
> [pulseaudio-modules-bt.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/4000393/pulseaudio-modules-bt.zip) contains `pulseaudio-modules-bt_1.3-0~eoan0_amd64.deb`
> 
> ```
> sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
> sudo vim /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
> sudo apt update
> sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt
> sudo dpkg -i pulseaudio-modules-bt_1.3-0~eoan0_amd64.deb
> ```

Tried this, and now I can't get the headset connected at all :cry: 
When I try I get `a2dp-sink profile connect failed for <MAC>: Protocol not available`
I get that when trying to connect using a2dp, headset,  handsfree, and with the headset on or off

ningfei, 2019-12-30:
> Tried this, and now I can't get the headset connected at all cry
> When I try I get `a2dp-sink profile connect failed for <MAC>: Protocol not available`
> I get that when trying to connect using a2dp, headset, handsfree, and with the headset on or off

weird...what's your `pulseaudio` version and `bluez` packages version? I had the same error iirc when I installed it on PopOS 19.10 which actually has `pulseaudio` `12.2` instead of `13.0`.

ArseniyShestakov, 2020-01-05:
I have PA 13.0 and bluetoothd 5.50 which are out of box versions for Ubuntu 19.10 and modules provided above dont work. Attempt to load modules into PA fails with:
```
Failure: Module initialisation failed
```

ArseniyShestakov, 2020-01-05:
Ok I got libraries above working by changing module file RPATH:
```
sudo chrpath -r  '/usr/lib/x86_64-linux-gnu/pulseaudio:/usr/lib/pulse-13.0/modules' /usr/lib/pulse-13.0/modules/module-blue*
```

Now after pulse restart my headset connected and working. :-)

mollat, 2020-01-26:
This was the complete process working for my Xubuntu 19.10:

```sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
sudo vim /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
sudo apt update
sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt
wget https://github.com/EHfive/pulseaudio-modules-bt/files/4000393/pulseaudio-modules-bt.zip
unzip pulseaudio-modules-bt.zip
sudo dpkg -i pulseaudio-modules-bt_1.3-0~eoan0_amd64.deb
sudo apt install chrpath
sudo chrpath -r  '/usr/lib/x86_64-linux-gnu/pulseaudio:/usr/lib/pulse-13.0/modules' /usr/lib/pulse-13.0/modules/module-blue*
sudo reboot
```
However - I can use aptx profile with my headphone (Beoplay H9 3rd gen.) only. Changing to sbc is not possible, but this is no show stopper for me... ;)

Thanks to all contributors!

uprilion, 2020-01-27:
> This was the complete process working for my Xubuntu 19.10:
> 
> ```
> sudo vim /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
> sudo apt update
> sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt
> wget https://github.com/EHfive/pulseaudio-modules-bt/files/4000393/pulseaudio-modules-bt.zip
> unzip pulseaudio-modules-bt.zip
> sudo dpkg -i pulseaudio-modules-bt_1.3-0~eoan0_amd64.deb
> sudo apt install chrpath
> sudo chrpath -r  '/usr/lib/x86_64-linux-gnu/pulseaudio:/usr/lib/pulse-13.0/modules' /usr/lib/pulse-13.0/modules/module-blue*
> sudo reboot
> ```
> 
> However - I can use aptx profile with my headphone (Beoplay H9 3rd gen.) only. Changing to sbc is not possible, but this is no show stopper for me... ;)
> 
> Thanks to all contributors!

This process worked for me with Kubuntu 19.10 with AAC, thanks everyone

martinhansen, 2020-01-27:
> This was the complete process working for my Xubuntu 19.10:
>
> ```
> sudo vim /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
> sudo apt update
> sudo apt install [...]

This worked for me too, Ubuntu 19.10. Can only select LDAC or HSP/HFP (using FiiO BTR3), but that's what i wanted (the LDAC) :)

ningfei, 2020-01-27:
> Built & packaged deb in case anyone needs it.
> [pulseaudio-modules-bt.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/4117035/pulseaudio-modules-bt.zip) contains [...]

I updated the package here to fix the runpath issue.  So hopefully there will be no need to do the chrpath after installation.

anadion, 2020-01-28:
> Built & packaged deb in case anyone needs it.
> [pulseaudio-modules-bt.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/4117035/pulseaudio-modules-bt.zip) contains `pulseaudio-modules-bt_1.3-0~eoan0_amd64.deb`

Got errors with this new file:
```
# dpkg -i pulseaudio-modules-bt_1.3-0~eoan0_amd64.deb 
(Reading database ... 256874 files and directories currently installed.)
Preparing to unpack pulseaudio-modules-bt_1.3-0~eoan0_amd64.deb ...
Unpacking pulseaudio-modules-bt (1.3-0~eoan0) over (1.3-0~eoan0) ...
dpkg: dependency problems prevent configuration of pulseaudio-modules-bt:
 pulseaudio-modules-bt depends on libfdk-aac1 (>= 0.1.4); however:
  Package libfdk-aac1 is not installed.
 pulseaudio-modules-bt depends on libpulse0 (= 13.0); however:
  Version of libpulse0:amd64 on system is 1:13.0-1ubuntu1.1.
 pulseaudio-modules-bt depends on pulseaudio (= 13.0); however:
  Version of pulseaudio on system is 1:13.0-1ubuntu1.1.

dpkg: error processing package pulseaudio-modules-bt (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 pulseaudio-modules-bt
```
Installed the old one and use `chrpath`

ningfei, 2020-01-28:
oh, updated again...[pulseaudio-modules-bt.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/4122156/pulseaudio-modules-bt.zip)

phoenixfirebrd, 2020-01-28:
@ningfei , Can you share the changes you made to the files to correct the rpath/runpath to obtain the above deb. So that I can build a proper deb.

ningfei, 2020-01-28:
@phoenixfirebrd actually I patched the runpath using `patchelf` after I compiled those libs. As I see, all the 5 libs need to be patched.

vide, 2020-01-29:
@ningfei Thanks a lot for the deb! Just installed it on Ubuntu 19.10 and now I can use the LDAC sink and oh boy the quality has improved!! Now it sounds like my Android phone or using the cable, and that background noise when the sound was more saturated is completely gone! 
And thanks to @EHfive too who did most of the work! :heart:

berglh, 2020-01-30:
I've been delayed on this project as I've had a new desktop machine at home I've been getting online. I'm still planning on publishing to a PPA for this once I'm properly setup again as the new machine is running Ubuntu 19.10 and I am unable to use HD bluetooth audio right now. :thinking: Maybe by the time I get publishing to the PPA, I can just jump straight to 20.04 building 😂

My project checklist looks like this:

- [x] Create a PPA in my namespace for EHfive's pulseaudio-modules
- [ ] Investigate testing the build in Docker
  - [x] Create Docker container that runs Systemd
  - [x] Find a solution for emulating a virtual bluetooth adapter
    - [ ] Figure out how to build the emulator without building the whole bluez package [current]
  - [ ] Write a build script to clone and build PulseAudio, pulseaudio-modules-bt, libldac and package as deb
    - [ ] Implement Ubuntu distribution to dependency version mapping to be used in a generic build script:
      - [ ]  Build source repos tag mapping to Ubunbu distributions i.e. Ubuntu 19.10 to PA repo v13.0, and relevant libldac and pulseaudio-module-bt repo tags.
      - [ ] Any build environment dependency versioning for the the PPA build manifest for the above source repos
  - [ ] Write test script to install debs and test bluetooth modules load on service start
  - [ ] Develop deploy script to upload source dirs tarballs and package manifests to PPA for publishing if the above tests are successful
    - [ ] Generate signing cert for the PPA
- [ ] Install drone.io on hosted server to build, test and publish to PPA on PR/Tag pushes (implement above scripts in pipeline)
- [ ] Profit

With this pipeline, I should hopefully be able to just change the versions of the source repo tags for each distribution and have it test build on my hosted server. If there are issues building and implementing this repo into PA (i.e. patching files fails due to significant upstream changes), then I can raise issues in this repo. I'm looking to catch build issues rather than just uploading the source tarballs to the PPA first and then waiting for them to fail when the PPA builds it in `chroot` environment, leaving potentially broken build packages in the PPA. I have a feeling the mapping of Docker to `chroot` build may not be super smooth, but I'll cross that bridge when I get to it.

monosoul, 2020-01-30:
Hey @berglh , thanks for investing your personal time into it, looking forward to have an up-to-date PPA :smile:

monosoul, 2020-01-30:
@berglh BTW, instead of maintaining and paying for your own hosted service to build this stuff, you may probably use github's integration with Travis CI, which is free for opensource projects

polesapart, 2020-01-30:
I've been sort of doing as  [this comment](https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-551390000) states and never had to do the chrpath stuff. Not sure why.

berglh, 2020-02-01:
> @berglh BTW, instead of maintaining and paying for your own hosted service to build this stuff, you may probably use github's integration with Travis CI, which is free for opensource projects

Thanks @monosoul, I've actually used Travis CI in the past. I have a hosted server for a bunch of other services I host for myself (personal git, DNS, private cloud sync etc), I recently increased the capacity of this machine so I have extra RAM and CPU to utilise now.

I guess, long term, it would make sense to have a CICD solution that can be easily contributed to and publicly visible, so people can help me maintain the Ubuntu PPA builds with PRs.

One of the challenges with actually testing the modules actually load correctly is getting the Bluetooth service to function and to enable the Bluetooth kernel modules. I'm not sure how much access TravisCI gives you to the machine (I am guessing this is an ephemeral VM) to make low level changes like loading kernel modules - it may very well suffice. For basic builds this would be perfectly fine, but for actually testing the pulseaudio-modules are being loaded correctly may not be so straight forward and would require some investigation.

Having a Ubuntu server that I control down to the OS level gives me that control and diagnostic ability of interacting directly with the container when things are not going well. Additionally, if I remember correctly, TravicCI free has some limitations with the length of jobs and concurrent jobs - which if I'm building for multiple targets simultaneously would require me to pay money to increase my accounts resource capacity in TravisCI anyway.

Last night I figured out how to build the `btvirt` binary from the `bluez` source code without building the whole project. I'm not entirely happy with it to be honest. It took a lot of grepping around in the `make -n` output to figure out the dependency build commands and the script only needs one thing to change upstream and the whole thing will fail. Regardless, it's functional, perhaps I'll package it in a Github release to ensure the `pulseaudio-modules-bt` build scripts don't fail on this process and update it using a separate repo in Github.

```shell
#!/bin/bash

echo "Making lib/bluetooth-internal.la deps"
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/bluetooth.h lib/bluetooth/bluetooth.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/hci.h lib/bluetooth/hci.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/hci_lib.h lib/bluetooth/hci_lib.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/sco.h lib/bluetooth/sco.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/l2cap.h lib/bluetooth/l2cap.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/sdp.h lib/bluetooth/sdp.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/sdp_lib.h lib/bluetooth/sdp_lib.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/rfcomm.h lib/bluetooth/rfcomm.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/bnep.h lib/bluetooth/bnep.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/cmtp.h lib/bluetooth/cmtp.h
/usr/bin/mkdir -p lib/bluetooth
ln -s -f /build/bluez/lib/hidp.h lib/bluetooth/hidp.h
./src/genbuiltin hostname wiimote autopair policy   a2dp avrcp network input hog  gap scanparam deviceinfo  battery > src/builtin.h
ln -s -f /build/bluez/lib/bluetooth.h lib/bluetooth/bluetooth.h
/bin/bash ./libtool --silent --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I./lib      -g -O2 -MT lib/hci.lo -MD -MP -MF $depbase.Tpo -c -o lib/hci.lo lib/hci.c &&\
/bin/bash ./libtool --silent --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I./lib      -g -O2 -MT lib/hci.lo -MD -MP -MF $depbase.Tpo -c -o lib/hci.lo lib/hci.c
/bin/bash ./libtool --silent --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I./lib      -g -O2 -MT lib/sdp.lo -MD -MP -MF $depbase.Tpo -c -o lib/sdp.lo lib/sdp.c
/bin/bash ./libtool --silent --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I./lib      -g -O2 -MT lib/uuid.lo -MD -MP -MF $depbase.Tpo -c -o lib/uuid.lo lib/uuid.c
/bin/bash ./libtool --silent --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I./lib      -g -O2 -MT lib/bluetooth.lo -MD -MP -MF $depbase.Tpo -c -o lib/bluetooth.lo lib/bluetooth.c

echo "Making lib/bluetooth-internal.la"
/bin/bash ./libtool --silent --tag=CC   --mode=link gcc     -g -O2   -o lib/libbluetooth-internal.la   lib/bluetooth.lo lib/hci.lo lib/sdp.lo  lib/uuid.lo

echo "Making mainloop-notify.lo Deps"
/bin/bash ./libtool --silent --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I./lib      -g -O2 -MT src/shared/queue.lo -MD -MP -MF .Tp [... cut ...]
gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I./lib      -g -O2 -MT emulator/le.o -MD -MP -MF $depbase.Tpo -c -o emulator/le.o emulator/le.c

echo "Building btvirt!"
/bin/bash ./libtool --silent --tag=CC   --mode=link gcc     -g -O2   -o emulator/btvirt emulator/main.o emulator/serial.o emulator/server.o emulator/vhci.o emulator/btdev.o emulator/bthost.o emulator/smp.o emulator/phy.o emulator/amp.o emulator/le.o lib/libbluetooth-internal.la src/libshared-mainloop.la
```

Lumca, 2020-02-20:
Hi guys,
thanks for making this possible.
I am new linux user and any of these guides are begginers friendly.
Is some eazy way to add support for aptX on ubuntu 19.10?
Like with deb file?

thimios, 2020-02-20:
@berglh We are looking forward to a 19.10 / 20.4 ppa is there any way we can support you on making this happen?

berglh, 2020-02-20:
Sorry peoples, I've been recently busy with the COVID-2019 epidemic and munging data to understand the impact of this at my work at work and have neglected this project. I also finally got some bits and pieces yesterday for my home rig which is now fully operational on 19.10 and not half open with an AIO hanging out on my desk. I also had a bunch of scripting to get basic tasks working that I use frequently that has also taken some time. I recently got sidetrack on starting a Hugo blog, as I have a bunch of projects that I've documented which I want to share.

Back to business, I hope to make some progress this weekend as 19.04 is going to be EOL soon and I'll need to upgrade my XPS. Time is ticking. March is also a busy month for me on the weekends, so I will try to make some time today and this weekend to make some progress here.

I'm at the point where I need to write a script to do the PulseAudio and libldac builds and generate a deb (for test installation). I haven't make any progress since my last update, but I've created a personal repo for now where I will start publishing my progress [pulseaudio-modules-bt-ppa](]https://git.bytefuse.net/berglh/pulseaudio-modules-bt-ppa) (this is my personal Git server where I keep things I'm ashamed of).

smarkusg, 2020-02-21:
Could you additionally generate deb PulseAudio for "fix mic problems for apple airpods huawei freebuds etc modern-headphones" based on hfpv7 ?
Then your everything would be complete on Linux - ACC + MIC+ *AirPods ;)

The mic fix on PulseAudio may not be fixed until 14.0.

"..No, there's no such intention. Not because we don't want the patches, I just don't think we'll have time to review them before the next release (which will be 14.0)..."

https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/84

Thank you very much for your help and work :-)

links:
https://github.com/kuraga/pulseaudio/releases/tag/v13.0-hfp-v7

https://forum.manjaro.org/t/patch-for-pulse-audio-for-hfp-fix-mic-problems-for-apple-airpods-huawei-freebuds-etc-modern-headphones/115865

berglh, 2020-02-22:
Thanks to some help from @EHfive I have the container now building and packaging `libldac` as a deb. It's install successfully and looks to have placed some files, but some of the modules with the version suffix don't appear to be in the correct arch folder `./x86_64-linux-gnu/` as on `disco`.

Most of the work is happening in the [Dockerfile](https://git.bytefuse.net/berglh/pulseaudio-modules-bt-ppa/src/master/Dockerfile) for the time being.

### eoan ldac files
```bash
/build# find /usr/lib -name "*ldac*"
/usr/lib/libldacBT_abr.so.2
/usr/lib/libldacBT_enc.so
/usr/lib/libldacBT_enc.so.2.0.2.3
/usr/lib/libldacBT_enc.so.2
/usr/lib/libldacBT_abr.so.2.0.2.3
/usr/lib/libldacBT_abr.so
/usr/lib/pkgconfig/ldacBT-abr.pc
/usr/lib/pkgconfig/ldacBT-enc.pc
```
### disco ldac files
```
/usr/lib/x86_64-linux-gnu/libldacBT_abr.so.2
/usr/lib/x86_64-linux-gnu/libldacBT_abr.so.2.0.2.3
/usr/lib/x86_64-linux-gnu/libldacBT_enc.so.2
/usr/lib/x86_64-linux-gnu/libldacBT_enc.so.2.0.2.3
/usr/lib/pkgconfig/ldacBT-enc.pc
/usr/lib/pkgconfig/ldacBT-abr.pc
/usr/lib/libldacBT_enc.so
/usr/lib/libldacBT_abr.so
```
Anyway, some progress has been made, I should have some time to continue this tomorrow.

berglh, 2020-02-29:
@EHfive I've made some more progress in making `libldac`. I managed to figure out my issues with the debain rules file. Does it matter that `libldacBT_enc.so` and `libldacBT_abr.so` is in `$INSTALL_LIBDIR`?

```bash
$ find /usr/lib -name "*ldac*"
/usr/lib/x86_64-linux-gnu/pkgconfig/ldacBT-enc.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/ldacBT-abr.pc
/usr/lib/x86_64-linux-gnu/libldacBT_enc.so
/usr/lib/x86_64-linux-gnu/libldacBT_enc.so.2.0.2.3
/usr/lib/x86_64-linux-gnu/libldacBT_abr.so
/usr/lib/x86_64-linux-gnu/libldacBT_abr.so.2.0.2.3
/usr/lib/x86_64-linux-gnu/libldacBT_enc.so.2
/usr/lib/x86_64-linux-gnu/libldacBT_abr.so.2
```
Looking in the CMakeFile.txt, these files seem to just get passed by `add_library`. `pckconfig` uses the `$INSTALL_LIBDIR` prfiex.

In the rules file I just needed to include for the `architecture.mk`, however looking at the make
```dh
#!/usr/bin/make -f
# coments ...
include /usr/share/dpkg/architecture.mk

%:
        dh $@

DEB_CONFIGURE_EXTRA_FLAGS = \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DINSTALL_LIBDIR=/usr/lib/$(DEB_HOST_MULTIARCH) \
    -DLDAC_SOFT_FLOAT=OFF

# dh_make generated override targets
# This is example for Cmake (See https://bugs.debian.org/641051 )
override_dh_auto_configure:
        dh_auto_configure -- $(DEB_CONFIGURE_EXTRA_FLAGS)
```
Will this still continue to work correctly? 

I'm just going to move on to building this repo now, so I can test the deb files on my `eoan` machine.

EHfive, 2020-02-29:
@berglh Doesn't matter if `/usr/lib/x86_64-linux-gnu/libldacBT_enc.so.2.0.2.3` correctly sym linked to `/usr/lib/libldacBT_enc.so` and `/usr/lib/pkgconfig/ldacBT-xxx.pc` pointing to `/usr/lib/x86_64-linux-gnu/libldacBT_xxx.so.2.0.2.3`

berglh, 2020-02-29:
Hello everyone.

So, some news so far is that today I've created some deb files similar to @ningfei but completely inside the Docker build container I've been developing. 

I've tested installing this on my new `eoan` workstation. The results appear to be good, I will continue testing over the next few days to ensure smooth operation.

I'd appreciate if people [@thimios, @Lumca, @ningfei et al] could test this on Ubuntu 19.10 `eaon` also. I would recommend commenting out/removing the EH5 repo from your `/etc/apt/sources.list.d/*.list` files and then doing an `apt update` before hand to avoid any package issues.

## Pre-Install
Remove existing `libldac` and `pulseaudio-modules-bt` installations if you already have them installed. I haven't had them installed
```
sudo apt purge libldac pulseaudio-modules-bt
```

## Install

**Note**: It's possible that pulseaudio might not work correctly after removing `libldac` or `pulseaudio-modules-bt`, I haven't tested this. You may need to also re-install `pulseaudio` before doing the following if you experience issues.

### 1. Install Run-time Dependencies 
```bash
sudo apt install -y libavcodec-extra58 libfdk-aac1 bluez pulseaudio
# Optional: I like using blueman for managing devices
sudo apt install -y blueman
```
### 2. Install `libldac` and `pulseaudio-modules-bt` debs I built
URL: https://f.bytefuse.net/pulseaudio-modules-bt-ppa/
Please download and install both files.
```
sudo dpkg -i <deb names>
```
### 3. Reboot

## My Results

I have a Intel Wi-Fi 6 AX200 chip with integrated Bluetooth 5.0 support.
After reboot I checked the bluetooth service registered the codec endpoints successfully.

### bluetooth Endpoint Registration
```shell
$ sudo journalctl -lu bluetooth | grep Endpoint
```
```
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSource/VENDOR/APTXHD
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSource/VENDOR/APTX
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSource/AAC
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSource/SBC
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSink/VENDOR/APTXHD
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSink/AAC
Feb 29 16:37:38 zen bluetoothd[896]: Endpoint registered: sender=:1.437 path=/MediaEndpoint/A2DPSink/SBC
```
### Check SBC, LDAC, Apt-X Operation

I checked on multiple devices with different codec support. You can see from `Active Profile` that the correct codecs have been use. Additionally, Apt-X Low Latency appear to be working based on how fast must and volume change is working. Looking at video with Low Latency seems to confirm how quick this is.

#### Sony SRS-HG1 (LDAC)
```shell
Card #4
        Driver: module-bluez5-device.c
        Owner Module: 34
        Properties:
                device.description = "bsrshg1"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "speaker"
                bluez.path = "/org/bluez/hci0/dev_E0_37_BF_09_F9_8B"
                bluez.class = "0x240414"
                bluez.alias = "bsrshg1"
[... 29 lines cut ...]

#### Sennheiser Momentum 3 (Apt-X Low Latency)
```
Card #7
        Name: bluez_card.00_1B_66_B1_01_55
        Driver: module-bluez5-device.c
        Owner Module: 37
        Properties:
                device.description = "MOMENTUM 3"
                device.string = "00:1B:66:B1:01:55"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_00_1B_66_B1_01_55"
                bluez.class = "0x240404"
                bluez.alias = "MOMENTUM 3"
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
        Profiles:
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: yes)
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink_aptx
```

berglh, 2020-02-29:
> @berglh Doesn't matter if `/usr/lib/x86_64-linux-gnu/libldacBT_enc.so.2.0.2.3` correctly sym linked to `/usr/lib/libldacBT_enc.so` and `/usr/lib/pkgconfig/ldacBT-xxx.pc` pointing to `/usr/lib/x86_64-linux-gnu/libldacBT_xxx.so.2.0.2.3`

Thanks @EHfive

ValZapod, 2020-02-29:
Debian also has libldac in wishlist, I think it would be nice to package it. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=922154
@rhertzog I think (IMHO at least as I have Sony Headphones with LDAC codec), this is very important... Dunno, can you make it happen?

berglh, 2020-02-29:
@ValZapod If you look at  https://f.bytefuse.net/pulseaudio-modules-bt-ppa/ you will see I packaged libldac as well as the pulseaudio-bluetooth modules. It's my intention to package both in a PPA.

ValZapod, 2020-02-29:
@berglh Yeah, I do not like this, obviously. https://linuxconfig.org/install-packages-from-an-ubuntu-ppa-on-debian-linux

rhertzog, 2020-03-01:
@ValZapod I don't have superpowers to make things happen, sorry. My free time is scarce enough. If you want to make this happen on my paid time, have a look at https://www.freexian.com/services/debian-packaging.html

katsar0v, 2020-03-06:
> This was the complete process working for my Xubuntu 19.10:
>
> ```
> sudo vim /etc/apt/source.list.d/eh5-ubuntu-pulseaudio-a2dp-eoan.list # change eoan to disco
> sudo apt update
> sudo apt install [...]

H9 3rd gen has aptX LL which is not listed in this repo. I have H9 as well, do you see any difference? I prefered to buy a usb dongle instead of compiling this repo. But a ppa or deb would definitely be better

berglh, 2020-03-07:
@katsar0v Even if your dongle supports AptX, it doesn't mean that PulseAudio will encode AptX on the Bluetooth stream AFAIK. ffmpeg codecs library does include AptX codec. If you follow the instructions above in comment https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-592915350 where it says to install `libavcodec-extra58`, you should get the Apt-X Bluetooth modules if you install the deb files I built and linked in that post. 

My Momentum 3 headphones have AptX LL, and even though there was no indication from the software modules that this was in use, I can confirm the latency is infact very low. My lower than my LDAC Bluetooth speaker. I'd almost say, it seems like I'm hearing audio before I'm seeing the picture rendered on the screen, it feels that quick. There is also the mute button test, when listening to audio, press the mute button on your keyboard, you should be able tap it time with the tempo of the music and perfectly cut one beat out of the song, with an almost direct feeling.

All of that being said, I DO have a CSR Bluetooth 4 dongle at work, and I seem to recall seeing AptX LL in the Sound settings as an option after building libldac and this repo manually. I can't remember if this was actually the case, but I will check when I'm back in the office. Regardless, as far as I can tell, the audio quality is definitely AptX and the latency seems just as low as it does at work on my home machine without a CSR chipset bluetooth chip - just try the instructions in my big comment above.

katsar0v, 2020-03-10:
> @katsar0v Even if your dongle supports AptX, it doesn't mean that PulseAudio will encode AptX on the Bluetooth stream AFAIK - maybe @EHfive can confirm this. ffmpeg codecs library does include AptX codec. If you follow the instructions above in comment [#85 (comment)](https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-592915350) where it says to install `libavcodec-extra58`, you should get the Apt-X Bluetooth modules if you install the deb files I built and linked in that post.
> 
> My Momentum 3 headphones have AptX LL, and even though there was no indication from the software modules that this was in use, I can confirm the latency is infact very low. Way lower than my LDAC Bluetooth speaker. I'd almost say, it seems like I'm hearing audio before I'm seeing the picture rendered on the screen, it feels that quick. There is also the mute button test, when listening to audio, press the mute button on your keyboard, you should be able tap it time with the tempo of the music and perfectly cut one beat out of the song, with an almost direct feeling.
> 
> All of that being said, I DO have a CSR Bluetooth 4 dongle at work, and I seem to recall seeing AptX LL in the Sound settings as an option after building libldac and this repo manually. I can't remember if this was actually the case, but I will check when I'm back in the office. Regardless, as far as I can tell, the audio quality is definitely AptX and the latency seems just as low as it does at work on my home machine without a CSR chipset bluetooth chip - just try the instructions in my big comment above.

https://www.amazon.de/Avantree-Bluetooth-Transmitter-Lautsprecher-REICHWEITE/dp/B01G3J1I5M/ref=sr_1_6?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=avantree&qid=1583844550&sr=8-6

It is rather transmitter recognized as a sound card, not a dongle. You are correct - when using dongles, the software being used is the one on you pc, but with transmitters it is different

PrincessOfHell, 2020-03-24:
ppa has worked for me in ubuntu 19.10  I have a bluetooth dac shanling up2 that supports ldac. A led indicates the codec used and it works, but pulseaudio won't let me pass to another codec like aptx.  It always stays in ldac.

PrincessOfHell, 2020-03-24:
Is there any chance to include lhdc from huawei?

patrolez, 2020-03-26:
Good news everyone!
https://github.com/EHfive/pulseaudio-modules-bt/pull/101

I would also like to know if it is working not only on my machine :> @berglh @EHfive

berglh, 2020-04-05:
@patrolez 
This is pretty nice for generating debs, but it's not testing and packaging for Ubuntu PPA. I am getting pretty close on being able to package to a PPA soon. 

Things have been very busy with my work recently, so I haven't made much time for this project, but hopefully after I finish another big project this week for work I can focus on this a little more in the next couple of weeks.

As an update, the container I have is building the debs and installing them fine to test. However, when I start pulseaudio in the container it is unable to find the libldacBT_enc files as shown by @ArseniyShestakov. 

```
W: [pulseaudio] ldac_libs.c: Cannot open LDAC encoder library: libldacBT_enc.so.2. libldacBT_enc.so.2: cannot open shared object file: No such file or directory
W: [pulseaudio] ldac_libs.c: Cannot open LDAC encoder library: libldacBT_enc.so. libldacBT_enc.so: cannot open shared object file: No such file or directory
```
What is strange, was that previous builds of this actually work fine for me when I wasn't generating the libldac-dev package as @EHfive was doing in his PPA. I'm wanting to make sure I generate the same artefacts as the existing PPA for completion sakes 

@EHfive I noticed in your PPA you have also published the pulseaudio build. Are you doing this to fix the pulseaudio version to a know "working" number to avoid breakages due to PA update?

patrolez, 2020-04-06:
>As an update, the container I have is building the debs and installing them fine to test
@berglh just after I have noticed that you are levelling up this project, I have closed a PR :D

GourouLubrik, 2020-04-16:
Mixing all the tips and guide out there, I finally get it working under ubuntu 20.04.

I can't make you a step by step, because I mixed it all... but In the end, SBC/AAC/LDAC & APTX are installed. and LDAC is indeed working on my Sony wh1000xm3.

In the end: this is all the package I installed to get it working... maybe some of them are unused, and surely some of them were already installed.

`sudo apt install pulseaudio libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential python3-dev libdbus-glib-1-dev libgirepository-1.0-1 libgirepository1.0-dev pkg-config libtool sbc-tools libtool pkg-config libtool sbc-tools libtool libavcodec-dev libfdk-aac-dev libldac-dev libavutil-dev libavcodec-extra58 libfdk-aac1 bluez `

When checking out pulseaudio tags I used the 13.99.1 as it correspond to ubuntu 20.04 version.

`git checkout v13.99.1`

I hope some kind of package will reach the ppa for 20.04 soon so that everyone can enjoy this fantastic package. 
Many thanks to @EHfive @berglh and @zylxpl

scanno, 2020-04-17:
Yeah would be nice to have a working PPA for Ubuntu 20.04 LTS once it hits stable. Very happy so far with working APTX on Ubuntu 18.04 LTS.

[46 more replies]

closed 2020-09-14 by stale[bot]