EHfive/pulseaudio-modules-bt #87: Add support for Bluetooth Absolute Volume

c79, 2019-10-22:


EHfive, 2019-10-22:
https://github.com/EHfive/pulseaudio-modules-bt/tree/absolute_volume

It's reserved, because not all of devices work perfectly with absolute volume.

closed 2019-10-22 by EHfive

chrismin13, 2019-11-11:
Is there any way we could bring this branch up to date? For all of my headphones, I can only adjust the volume using the following command:
```dbus-send --print-reply --system --dest=org.bluez /org/bluez/hci0/dev_1C_52_16_5C_8A_AF org.bluez.MediaControl1.VolumeUp```
(replace of course the MAC Address with yours!)

It would be great if we could have this as an option for users with headphones that have too low of a volume by default. Is there any way that a toggle could be added to switch the feature on or off? Alternatively, could a separate module be built to enable absolute volume?

Thanks!

c79, 2019-11-14:
I agree with @chrismin13 . I'm facing the same problem of low volume. 

Perhaps add absolute volume as a PulseAudio module paramete/option.

Anyways, Windows 10, Mac OS X, iOS, and Android devices all enabled absolute volume by default and face little problem reports. I think issues such as these are rare.

chrismin13, 2019-11-14:
@c79 You're right, every other OS has absolute volume support without any issues. Android and Windows also offer ways of switching off Absolute Volume in case you encounter any problems. This trend will only continue, as most modern bluetooth headphones consume less power when the volume is lower.

@EHfive Do you have any cases of devices not working with absolute volume? Searching for past issues in this repo regarding volume brings up no relevant results. It could just be that something was misconfigured elsewhere or that a specific device is problematic. For the most part, devices have more issues without absolute volume.

Is there any way we could contribute to bring the branch up to date and have it merged? I don't have any experience with C, only Java, but I could probably find a way to help out if needed. Could we have some general guidelines?

Thanks in advance!

artizirk, 2020-04-14:
For future reference, there is slow work going on in main pulseaudio repo to make Absolute Volume work. https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/239 

Also it looks like the really old patch from [this gist](https://gist.github.com/EHfive/e2a28d0279a6247fab4bac93d73b8571) still works, only small fixes we required
[0001-A2DP-Absolute-Volume.txt](https://github.com/EHfive/pulseaudio-modules-bt/files/4474981/0001-A2DP-Absolute-Volume.txt)