EHfive/pulseaudio-modules-bt #89: No package 'dbus-1' found

fankertiban, 2019-12-06:
Operating System: KDE neon 5.17
KDE Plasma Version: 5.17.4
KDE Frameworks Version: 5.64.0
Qt Version: 5.13.2
Kernel Version: 5.0.0-36-generic

# cmake -DFORCE_LARGEST_PA_VERSION=ON CODEC_APTX_FF=ON CODEC_APTX_HD_FF=ON CODEC_AAC_FDK=ON CODEC_LDAC=ON ..

-- Checking for module 'dbus-1'
--   No package 'dbus-1' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  CMakeLists.txt:42 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/opt/pulseaudio-modules-bt/build/CMakeFiles/CMakeOutput.log".

Have no idea what to do)

EHfive, 2019-12-07:
Try installl `libdbus-1-dev`

closed 2019-12-07 by EHfive

fankertiban, 2019-12-07:
Thanks a lot! Worked for me