EHfive/pulseaudio-modules-bt #91: No sound through BT to DAC output on RPi Zero

yoGhurrt1, 2020-01-06:
Hi,

I configured mi RPi Zero in base on orrpan [topic]( https://github.com/EHfive/pulseaudio-modules-bt/issues/55) (with some additions) but I cannot get audio from phone, through BT to DAC output. When I'll try play something through phone - there is nothing on output.

I build my config with bluez 5.52 and pulseaudio 13.0 on my raspbian (buster) instead of 5.50 (there was recommendation to use newer build anyway) and pulse 12.2. 

My pulseaudio -v has some errors, but I'm not sure, how much is that related and could confirm it on other websites.

`I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
I: [pulseaudio] core-util.c: Failed to acquire high-priority scheduling: Permission denied
I: [pulseaudio] main.c: This is PulseAudio 13.0
I: [pulseaudio] main.c: Page size is 4096 bytes
I: [pulseaudio] main.c: Machine ID is b6a16c8d8c9d4642a5dea0a1595aac9c.
I: [pulseaudio] main.c: Session ID is c2.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/pi/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
I: [pulseaudio] main.c: System supports high resolution timers
I: [pulseaudio] cpu-arm.c: CPU flags: V6 V7 VFP EDSP
I: [pulseaudio] svolume_arm.c: Initialising ARM optimized volume functions.
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-stream-volumes'.
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #1; argument: "").
I: [pulseaudio] module-card-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-card-database'.
I: [pulseaudio] module.c: Loaded "module-card-restore" (index: #2; argument: "").
I: [pulseaudio] module.c: Loaded "module-augment-properties" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-port-available" (index: #4; argument: "").
E: [pulseaudio] ltdl-bind-now.c: Failed to open module module-alsa-card.so: module-alsa-card.so: cannot open shared object file: No such file or directory
E: [pulseaudio] module.c: Failed to open module "module-alsa-card".
I: [pulseaudio] module-udev-detect.c: Card /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 (alsa_card.usb-SMSL_Inc._SMSL_iDol_DAC-00) failed to load module.
I: [pulseaudio] module-udev-detect.c: Found 1 cards.
I: [pulseaudio] module.c: Loaded "module-udev-detect" (index: #5; argument: "").
I: [pulseaudio] module.c: Loaded "module-esound-protocol-unix" (index: #6; argument: "").
I: [pulseaudio] module.c: Loaded "module-native-protocol-unix" (index: #7; argument: "").
I: [pulseaudio] module.c: Loaded "module-gsettings" (index: #8; argument: "").
I: [pulseaudio] module-default-device-restore.c: No previous default sink setting, ignoring.
I: [pulseaudio] module-default-device-restore.c: No previous default source setting, ignoring.
I: [pulseaudio] module.c: Loaded "module-default-device-restore" (index: #9; argument: "").
I: [pulseaudio] module.c: Loaded "module-rescue-streams" (index: #10; argument: "").
I: [pulseaudio] sink.c: Created sink 0 "auto_null" with sample spec s16le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] sink.c:     device.description = "Dummy Output"
I: [pulseaudio] sink.c:     device.class = "abstract"
I: [pulseaudio] sink.c:     device.icon_name = "audio-card"
I: [pulseaudio] source.c: Created source 0 "auto_null.monitor" with sample spec s16le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] source.c:     device.description = "Monitor of Dummy Output"
I: [pulseaudio] source.c:     device.class = "monitor"
I: [pulseaudio] source.c:     device.icon_name = "audio-input-microphone"
I: [null-sink] util.c: Failed to acquire real-time scheduling: Permission denied
I: [pulseaudio] core.c: default_source: (unset) -> auto_null.monitor
I: [pulseaudio] core.c: default_sink: (unset) -> auto_null
I: [pulseaudio] module.c: Loaded "module-null-sink" (index: #12; argument: "sink_name=auto_null sink_properties='device.description="Dummy Output"'").
I: [pulseaudio] module.c: Loaded "module-always-sink" (index: #11; argument: "").
I: [pulseaudio] module.c: Loaded "module-intended-roles" (index: #13; argument: "").
I: [pulseaudio] module.c: Loaded "module-suspend-on-idle" (index: #14; argument: "").
I: [pulseaudio] module.c: Loaded "module-position-event-sounds" (index: #15; argument: "").
I: [pulseaudio] module.c: Loaded "module-role-cork" (index: #16; argument: "").
I: [pulseaudio] module.c: Loaded "module-filter-heuristics" (index: #17; argument: "").
I: [pulseaudio] module.c: Loaded "module-filter-apply" (index: #18; argument: "").
I: [pulseaudio] module.c: Loaded "module-bluez5-discover" (index: #20; argument: "a2dp_config="ldac_eqmid=sq"").
I: [pulseaudio] module.c: Loaded "module-bluetooth-discover" (index: #19; argument: "a2dp_config="ldac_eqmid=sq"").
I: [pulseaudio] main.c: Daemon startup complete.
E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.ofono.Error.InUse: The resource is currently in use
I: [pulseaudio] module-suspend-on-idle.c: Sink auto_null idle for too long, suspending ...
`

It's probably not a problem with DAC, because it's recognized correctly and I can hear sound while test with speaker-test.

Could someone help me, how to track down this issue?

EHfive, 2020-01-07:
What is "BT to DAC output" mean ? 
Give me more details about your audio devices system/setup?

Besides, assuming that is your last logs, you are not get your BT audio device connected.

yoGhurrt1, 2020-01-07:
Sure, a bit more of clarification. With "BT to DAC output" I meant sending audio from phone through Bluetooth USB dongle, that's connected to RPi Zero, to USB DAC (that's also connected to RPi) and finally to anything with minijack that's connected to that USB DAC. My USB DAC is SMSL iDol+ - I have a plan to buy HifiBerry DAC+ Zero, but that's for future. For now, I want to make it work and sure that's worth it to go further.

Yes, it were my lastest logs, so here's new ones, when I connected my phone to RPi

`I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
I: [pulseaudio] core-util.c: Failed to acquire high-priority scheduling: Permission denied
I: [pulseaudio] main.c: This is PulseAudio 13.0
I: [pulseaudio] main.c: Page size is 4096 bytes
I: [pulseaudio] main.c: Machine ID is b6a16c8d8c9d4642a5dea0a1595aac9c.
I: [pulseaudio] main.c: Session ID is c2.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/pi/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
I: [pulseaudio] main.c: System supports high resolution timers
I: [pulseaudio] cpu-arm.c: CPU flags: V6 V7 VFP EDSP
I: [pulseaudio] svolume_arm.c: Initialising ARM optimized volume functions.
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-stream-volumes'.
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #1; argument: "").
I: [pulseaudio] module-card-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-card-database'.
I: [pulseaudio] module.c: Loaded "module-card-restore" (index: #2; argument: "").
I: [pulseaudio] module.c: Loaded "module-augment-properties" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-port-available" (index: #4; argument: "").
E: [pulseaudio] ltdl-bind-now.c: Failed to open module module-alsa-card.so: module-alsa-card.so: cannot open shared object file: No such file or directory
E: [pulseaudio] module.c: Failed to open module "module-alsa-card".
I: [pulseaudio] module-udev-detect.c: Card /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 (alsa_card.usb-SMSL_Inc._SMSL_iDol_DAC-00) failed to load module.
I: [pulseaudio] module-udev-detect.c: Found 1 cards.
I: [pulseaudio] module.c: Loaded "module-udev-detect" (index: #5; argument: "").
I: [pulseaudio] module.c: Loaded "module-esound-protocol-unix" (index: #6; argument: "").
I: [pulseaudio] module.c: Loaded "module-native-protocol-unix" (index: #7; argument: "").
I: [pulseaudio] module.c: Loaded "module-gsettings" (index: #8; argument: "").
I: [pulseaudio] module-default-device-restore.c: No previous default sink setting, ignoring.
I: [pulseaudio] module-default-device-restore.c: No previous default source setting, ignoring.
I: [pulseaudio] module.c: Loaded "module-default-device-restore" (index: #9; argument: "").
I: [pulseaudio] module.c: Loaded "module-rescue-streams" (index: #10; argument: "").
I: [pulseaudio] sink.c: Created sink 0 "auto_null" with sample spec s16le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] sink.c:     device.description = "Dummy Output"
I: [pulseaudio] sink.c:     device.class = "abstract"
I: [pulseaudio] sink.c:     device.icon_name = "audio-card"
I: [pulseaudio] source.c: Created source 0 "auto_null.monitor" with sample spec s16le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] source.c:     device.description = "Monitor of Dummy Output"
I: [pulseaudio] source.c:     device.class = "monitor"
I: [pulseaudio] source.c:     device.icon_name = "audio-input-microphone"
I: [null-sink] util.c: Failed to acquire real-time scheduling: Permission denied
I: [pulseaudio] core.c: default_source: (unset) -> auto_null.monitor
I: [pulseaudio] core.c: default_sink: (unset) -> auto_null
I: [pulseaudio] module.c: Loaded "module-null-sink" (index: #12; argument: "sink_name=auto_null sink_properties='device.description="Dummy Output"'").
I: [pulseaudio] module.c: Loaded "module-always-sink" (index: #11; argument: "").
I: [pulseaudio] module.c: Loaded "module-intended-roles" (index: #13; argument: "").
I: [pulseaudio] module.c: Loaded "module-suspend-on-idle" (index: #14; argument: "").
I: [pulseaudio] module.c: Loaded "module-position-event-sounds" (index: #15; argument: "").
I: [pulseaudio] module.c: Loaded "module-role-cork" (index: #16; argument: "").
I: [pulseaudio] module.c: Loaded "module-filter-heuristics" (index: #17; argument: "").
I: [pulseaudio] module.c: Loaded "module-filter-apply" (index: #18; argument: "").
I: [pulseaudio] module.c: Loaded "module-bluez5-discover" (index: #20; argument: "a2dp_config="ldac_eqmid=sq"").
I: [pulseaudio] module.c: Loaded "module-bluetooth-discover" (index: #19; argument: "a2dp_config="ldac_eqmid=sq"").
I: [pulseaudio] main.c: Daemon startup complete.
E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.ofono.Error.InUse: The resource is currently in use
I: [pulseaudio] module-suspend-on-idle.c: Sink auto_null idle for too long, suspending ...
`

After a quick look I cannot see a difference between those two logs, what would mean that pulseaudio cannot really see, what's going on around bluez and if is there new audio device connected though BT.

nicosea, 2020-01-14:
I dream to use a pi-zero or pi 3 as an Music LDAC bluetooth streamer (ideally source squeezelite or squeezeplay player), do you know how to do that?
I read you are trying to use pizero as a receiver, if it works, it should also work in case I want to use as streamer?

yoGhurrt1, 2020-01-18:
@nicosea in my first post, you have link to topic with tutorial how to install everything on Pi. Yes, it will work in way you expect, of course if you have headphones able to use LDAC.

yoGhurrt1, 2020-01-25:
Ok, I made some research and honestly I'm stuck, but still I think this is related to bt module.

Firstly, log from pulseaudio - it's slightly different. Now, pulse detects devices and there is actually sth when I did some tries to play music from phone. 

```
I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
I: [pulseaudio] core-util.c: Failed to acquire high-priority scheduling: Permission denied
I: [pulseaudio] main.c: This is PulseAudio 13.0
D: [pulseaudio] main.c: Compilation host: armv6l-unknown-linux-gnueabihf
D: [pulseaudio] main.c: Compilation CFLAGS: -g -O2 -Wall -W -Wextra -pipe -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -fno-common -fdiagnostics-show-option -fdiagnostics-color=auto
D: [pulseaudio] main.c: Running on host: Linux armv6l 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019
D: [pulseaudio] main.c: Found 1 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: no
D: [pulseaudio] main.c: Running from build tree: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.
I: [pulseaudio] main.c: Machine ID is b6a16c8d8c9d4642a5dea0a1595aac9c.
I: [pulseaudio] main.c: Session ID is c8.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/pi/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
I: [pulseaudio] main.c: System supports high resolution timers
D: [pulseaudio] memblock.c: Using shared memfd memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65496
I: [pulseaudio] cpu-arm.c: CPU flags: V6 V7 VFP EDSP
I: [pulseaudio] svolume_arm.c: Initialising ARM optimized volume functions.
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-stream-volumes'.
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1 added for object /org/pulseaudio/stream_restore1
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #1; argument: "").
I: [pulseaudio] module-card-restore.c: Successfully opened database file '/home/pi/.config/pulse/b6a16c8d8c9d4642a5dea0a1595aac9c-card-database'.
I: [pulseaudio] module.c: Loaded "module-card-restore" (index: #2; argument: "").
I: [pulseaudio] module.c: Loaded "module-augment-properties" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-port-available" (index: #4; argument: "").
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-13.0/modules/module-udev-detect.so': success
D: [pulseaudio] module-udev-detect.c: /dev/snd/controlC1 is accessible: yes
D: [pulseaudio] module-udev-detect.c: /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 is busy: no
D: [pulseaudio] module-udev-detect.c: Loading module-alsa-card with arguments 'device_id="1" name="usb-SMSL_Inc._SMSL_iDol_DAC-00" card_name="alsa_card.usb-SMSL_Inc._SMSL_iDol_DAC-00" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1"'
E: [pulseaudio] ltdl-bind-now.c: Failed to open module module-alsa-card.so: module-alsa-card.so: cannot open shared object file: No such file or directory
E: [pulseaudio] module.c: Failed to open module "module-alsa-card".
I: [pulseaudio] module-udev-detect.c: Card /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card1 (alsa_card.usb-SMSL_Inc._SMSL_iDol_DAC-00) failed to load module.
I: [pulseaudio] module-udev-detect.c: Found 1 cards.
I: [pulseaudio] module.c: Loaded "module-udev-detect" (index: #5; argument: "").
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-13.0/modules/module-jackdbus-detect.so': failure
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-13.0/modules/module-esound-protocol-unix.so': success
I: [pulseaudio] module.c: Loaded "module-esound-protocol-unix" (index: #6; argument: "").
I: [pulseaudio] module.c: Loaded "module-native-protocol-unix" (index: #7; argument: "").
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-13.0/modules/module-gsettings.so': success
I: [pulseaudio] module.c: Loaded "module-gsettings" (index: #8; argument: "").
I: [pulseaudio] module-default-device-restore.c: No previous default sink setting, ignoring.
[... cut ...]
    Direct sample mixing device
dsnoop:CARD=DAC,DEV=0
    SMSL iDol DAC, USB Audio
    Direct sample snooping device
hw:CARD=DAC,DEV=0
    SMSL iDol DAC, USB Audio
    Direct hardware device without any conversions
plughw:CARD=DAC,DEV=0
    SMSL iDol DAC, USB Audio
    Hardware device with all software conversions
usbstream:CARD=DAC
    SMSL iDol DAC
    USB Stream Output
**** List of PLAYBACK Hardware Devices ****
card 1: DAC [SMSL iDol DAC], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

closed 2020-01-28 by yoGhurrt1