EHfive/pulseaudio-modules-bt #92: no sound APTx (was working fine previously)

MeisterTea, 2020-01-10:
Hi, my aptX playback doesn't work anymore. I can only use HSP/HFP

It was working fine previously.
Can you help me out at least to debug it ?

my versions: 

* Arch
* Linux 5.4.8.arch1-1
* pulseaudio-modules-bt-git-r143.b9d3bd3-1
* pulseaudio 13.0-2

Headset: Audio Technica ATH-M50xBT

avfinfo:

```
Connecting ... 
Stream End-Point #5: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Data: 0x32
		Vendor Specific Value (aptX)
			Frequencies: 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #3: Audio Sink 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #37: Audio Source 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Data: 0x32
		Vendor Specific Value (aptX)
			Frequencies: 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #35: Audio Source 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #33: Audio Source 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T

```

EHfive, 2020-01-10:
is your `ffmpeg` still installed

EHfive, 2020-01-10:
If modules A2DP endpoints successfully registered
```
journalctl -fu bluetooth
-- Logs begin at Thu 2019-12-19 21:53:56 CST. --
1月 10 19:56:11 eh5-arch bluetoothd[884]: Endpoint registered: sender=:1.134 path=/MediaEndpoint/A2DPSink/VENDOR/APTXHD
1月 10 19:56:11 eh5-arch bluetoothd[884]: Endpoint registered: sender=:1.134 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
1月 10 19:56:11 eh5-arch bluetoothd[884]: Endpoint registered: sender=:1.134 path=/MediaEndpoint/A2DPSink/AAC
1月 10 19:56:11 eh5-arch bluetoothd[884]: Endpoint registered: sender=:1.134 path=/MediaEndpoint/A2DPSink/SBC
```

Your pulseaudio bluetooth card & profiles info
```
$ LANGUAGE=C pactl list cards | grep bluez_card -A20
        Name: bluez_card.70_26_05_00_00_00
        Driver: module-bluez5-device.c
        Owner Module: 29
        Properties:
                device.description = "WH-1000XM2"
                device.string = "70:26:05:00:00:00"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_70_26_05_00_00_00"
                bluez.class = "0x240404"
                bluez.alias = "WH-1000XM2"
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
        Profiles:
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: yes)
```

MeisterTea, 2020-01-10:
ffmpeg is still installed: 1:4.2.1-4

the A2DP module endpoints are registered successfully
```
janv. 10 11:08:24 glados bluetoothd[1024]: Endpoint registered: sender=:1.911 path=/MediaEndpoint/A2DPSource/VENDOR/APTX
janv. 10 11:08:24 glados bluetoothd[1024]: Endpoint registered: sender=:1.911 path=/MediaEndpoint/A2DPSource/AAC
janv. 10 11:08:24 glados bluetoothd[1024]: Endpoint registered: sender=:1.911 path=/MediaEndpoint/A2DPSource/SBC
janv. 10 11:08:24 glados bluetoothd[1024]: Endpoint registered: sender=:1.911 path=/MediaEndpoint/A2DPSink/VENDOR/APTXHD
janv. 10 11:08:24 glados bluetoothd[1024]: Endpoint registered: sender=:1.911 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
janv. 10 11:08:24 glados bluetoothd[1024]: Endpoint registered: sender=:1.911 path=/MediaEndpoint/A2DPSink/AAC
janv. 10 11:08:24 glados bluetoothd[1024]: Endpoint registered: sender=:1.911 path=/MediaEndpoint/A2DPSink/SBC
janv. 10 11:08:27 glados bluetoothd[1024]: /org/bluez/hci0/dev_00_0A_45_0B_1B_A5/sep5/fd10: fd(42) ready
janv. 10 12:50:03 glados bluetoothd[1024]: /org/bluez/hci0/dev_00_0A_45_0B_1B_A5/sep5/fd11: fd(42) ready
janv. 10 14:48:04 glados bluetoothd[1024]: /org/bluez/hci0/dev_00_0A_45_0B_1B_A5/sep5/fd12: fd(41) ready
```

It doesn't seem to be any any problem here as well
```
        Name: bluez_card.00_0A_45_0B_1B_A5
        Driver: module-bluez5-device.c
        Owner Module: 29
        Properties:
                device.description = "ATH-M50xBT    "
                device.string = "00:0A:45:0B:1B:A5"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headphone"
                bluez.path = "/org/bluez/hci0/dev_00_0A_45_0B_1B_A5"
                bluez.class = "0x240418"
                bluez.alias = "ATH-M50xBT    "
                device.icon_name = "audio-headphones-bluetooth"
        Profiles:
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
                a2dp_source_sbc: High Fidelity Capture (A2DP Source: SBC) (sinks: 0, sources: 1, priority: 20, available: yes)
                a2dp_source_aac: High Fidelity Capture (A2DP Source: AAC) (sinks: 0, sources: 1, priority: 20, available: yes)
                a2dp_source_aptx: High Fidelity Capture (A2DP Source: APTX) (sinks: 0, sources: 1, priority: 20, available: yes)
                a2dp_source_aptx_hd: High Fidelity Capture (A2DP Source: aptX HD) (sinks: 0, sources: 1, priority: 20, available: no)
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)

```

EHfive, 2020-01-10:
might relates 
https://github.com/FFmpeg/FFmpeg/commit/675f62a202be7cfe9576b48679b8e6bc1183e84e & https://github.com/FFmpeg/FFmpeg/commit/98a257c3235bdc18151534134148845728418248

You could try downgrade `ffmpeg`. (using [downgrade](https://aur.archlinux.org/packages/downgrade/) , or fetch older package & install it)

asavah, 2020-01-10:
I have build pulseaudio-modules-bt literally yesterday with ffmpeg git master - no issues here with aptx.

MeisterTea, 2020-01-10:
Is there anything I can do to pinpoint where is the problem on this install?

EHfive, 2020-01-10:
There might be useful info ouputed by `ffmpeg` in pulseaudio logs
```
$ systemctl --user stop pulseaudio.socket
$ systemctl --user stop pulseaudio.service
$ pulseaudio --verbose
```

MeisterTea, 2020-01-10:
I got this:

```
I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: This is PulseAudio 13.0-dirty
I: [pulseaudio] main.c: Page size is 4096 bytes
I: [pulseaudio] main.c: Machine ID is 3960d79e93ca47e486fb311eff91dbc5.
I: [pulseaudio] main.c: Session ID is 1.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/chell/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
I: [pulseaudio] main.c: System supports high resolution timers
I: [pulseaudio] cpu-x86.c: CPU flags: CMOV MMX SSE SSE2 SSE3 SSSE3 SSE4_1 SSE4_2
I: [pulseaudio] svolume_mmx.c: Initialising MMX optimized volume functions.
I: [pulseaudio] remap_mmx.c: Initialising MMX optimized remappers.
I: [pulseaudio] svolume_sse.c: Initialising SSE2 optimized volume functions.
I: [pulseaudio] remap_sse.c: Initialising SSE2 optimized remappers.
I: [pulseaudio] sconv_sse.c: Initialising SSE2 optimized conversions.
I: [pulseaudio] svolume_orc.c: Initialising ORC optimized volume functions.
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/home/chell/.config/pulse/3960d79e93ca47e486fb311eff91dbc5-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/home/chell/.config/pulse/3960d79e93ca47e486fb311eff91dbc5-stream-volumes'.
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #1; argument: "").
I: [pulseaudio] module-card-restore.c: Successfully opened database file '/home/chell/.config/pulse/3960d79e93ca47e486fb311eff91dbc5-card-database'.
I: [pulseaudio] module.c: Loaded "module-card-restore" (index: #2; argument: "").
I: [pulseaudio] module.c: Loaded "module-augment-properties" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-port-available" (index: #4; argument: "").
I: [pulseaudio] (alsa-lib)utils.c: could not open configuration file /usr/share/alsa/ucm2/HDA Intel PCH/HDA Intel PCH.conf
I: [pulseaudio] (alsa-lib)parser.c: error: could not parse configuration for card HDA Intel PCH
I: [pulseaudio] (alsa-lib)main.c: error: failed to import HDA Intel PCH use case configuration -2
I: [pulseaudio] alsa-ucm.c: UCM not available for card HDA Intel PCH
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL front:0
I: [pulseaudio] alsa-util.c: Unable to attach to mixer front:0: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] alsa-util.c: Error opening PCM device iec958:0: No such file or directory
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL front:0
I: [pulseaudio] alsa-util.c: Unable to attach to mixer front:0: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] (alsa-lib)pcm_route.c: Found no matching channel map
I: [pulseaudio] alsa-util.c: Error opening PCM device surround21:0: Invalid argument
I: [pulseaudio] alsa-util.c: Failed to set hardware parameters on plug:surround40:0: Invalid argument
I: [pulseaudio] (alsa-lib)pcm_route.c: Found no matching channel map
I: [pulseaudio] alsa-util.c: Error opening PCM device surround41:0: Invalid argument
I: [pulseaudio] (alsa-lib)pcm_route.c: Found no matching channel map
I: [pulseaudio] alsa-util.c: Error opening PCM device surround50:0: Invalid argument
I: [pulseaudio] alsa-util.c: Failed to set hardware parameters on plug:surround51:0: Invalid argument
I: [pulseaudio] alsa-util.c: Failed to set hardware parameters on plug:surround71:0: Invalid argument
I: [pulseaudio] alsa-util.c: Error opening PCM device iec958:0: No such file or directory
I: [pulseaudio] (alsa-lib)pcm.c: Unknown PCM a52:0
I: [pulseaudio] alsa-util.c: Error opening PCM device a52:0: No such file or directory
I: [pulseaudio] (alsa-lib)pcm.c: Unknown PCM a52:0
I: [pulseaudio] alsa-util.c: Error opening PCM device a52:0: No such file or directory
I: [pulseaudio] (alsa-lib)pcm.c: Unknown PCM dca:0
I: [pulseaudio] alsa-util.c: Error opening PCM device dca:0: No such file or directory
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL hdmi:0
I: [pulseaudio] alsa-util.c: Unable to attach to mixer hdmi:0: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL hdmi:0
I: [pulseaudio] alsa-util.c: Unable to attach to mixer hdmi:0: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL hdmi:0
I: [pulseaudio] alsa-util.c: Unable to attach to mixer hdmi:0: No such file or directory
I: [pulseaudio] alsa-util.c: Successfully attached to mixer 'hw:0'
I: [pulseaudio] (alsa-lib)pcm.c: Unknown PCM dcahdmi:0
I: [pulseaudio] alsa-util.c: Error opening PCM device dcahdmi:0: No such file or directory
I: [pulseaudio] (alsa-lib)control.c: Invalid CTL hdmi:0,1
I: [pulseaudio] alsa-util.c: Unable to attach to mixer hdmi:0,1: No such file or directory
[... 265 lines cut ...]
I: [pulseaudio] client.c: Created 2 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Got credentials: uid=1000 gid=1000 success=1
[... cut ...]
I: [pulseaudio] sink-input.c:     application.process.binary = "chrome"
I: [pulseaudio] sink-input.c:     application.language = "en_US.UTF-8"
I: [pulseaudio] sink-input.c:     window.x11.display = ":0"
I: [pulseaudio] sink-input.c:     application.process.machine_id = "3960d79e93ca47e486fb311eff91dbc5"
I: [pulseaudio] sink-input.c:     application.process.session_id = "1"
I: [pulseaudio] sink-input.c:     module-stream-restore.id = "sink-input-by-application-name:Chromium"

```

Is there anything out of the ordinary ? :/

closed 2020-01-13 by MeisterTea