EHfive/pulseaudio-modules-bt #93: Switchin A2DP profiles does not work with B&W headphones

gergap, 2020-01-29:
First of all thanks for your great work.

I installed this new bluetooth module to get the best sound quality.
Not that it was bad before, but I found the sound via the Android phone (aptX HD)
better than from my Debian 10 based laptop.

Installation went fine and new profiles showed up in pavucontrol.
It didn't note any difference in sound quality, but I also didn't know what codec was used before.

Before installation there were only HSP and A2DP profiles.
Headset Head Unit (HSP/HPF): For headsets, sounds like crap for music
High Fidelity Playback (A2DP) - sounds good, no more information about codecs

After installing the new module it changed to this:
Headset Head Unit (HSP/HPF): For headsets, sounds like crap for music
High Fidelity Playback (A2DP Sink: aptX HD): sounds good
High Fidelity Playback (A2DP Sink: aptX): no difference
High Fidelity Playback (A2DP Sink: ACC): no difference
High Fidelity Playback (A2DP Sink: SBC): no difference

I didn't hear any difference. There was not even a click noise when switching, so I got curious
if there is something happening at all.

So I fired up Wireshark to checked what was going on.
1.) Reverting back to the original.
I could see in Wireshark that the A2DP protocol was using the SBC codec.
OK, sounds not as bad is I was reading in the internet.
But I find it sounds different to my mobile phone which uses aptX HD.

2.) Again I installed the git module.
In Wireshark I could see a new protocol layer,
RTP inside A2DP, that was not there before, interesting.
Codec: non-A2DP (0xff), seems to be some sort of extension I guess
Vendor codec: 0x0024 (which is aptX HD according to the sources)

When switching to different A2DP profiles absolutely nothing happens.
Still 0x0024 in wireshark

Then I found in the logs (should have read this before btw), that switching profiles fails. But nothing shows up in the GUI which is bad, but probably this is not the fault of this module.

The log says this:

```
Jan 29 21:26:00 hostname pulseaudio[19134]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aptx: Unavailable profile
Jan 29 21:26:10 hostname pulseaudio[19134]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aac: Unavailable profile
Jan 29 21:26:12 hostname pulseaudio[19134]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_sbc: Unavailable profile
```

pactl list shows this:

```
...
Card #4
	Name: bluez_card.EC_66_D1_A3_71_29
	Driver: module-bluez5-device.c
	Owner Module: 26
	Properties:
		device.description = "Bowers & Wilkins PX"
		device.string = "EC:66:D1:A3:71:29"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headphone"
		bluez.path = "/org/bluez/hci0/dev_EC_66_D1_A3_71_29"
		bluez.class = "0x240418"
		bluez.alias = "Bowers & Wilkins PX"
		device.icon_name = "audio-headphones-bluetooth"
	Profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink_aptx_hd
	Ports:
		headphone-output: Headphone (priority: 0, latency offset: 0 usec, available)
			Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd
		headphone-input: Bluetooth Input (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit
```

So the codecs are available. What is reported here also fits to the user manual of the head phones.

Any ideas why this switching does not work?
Let me know if you need more information.

gergap, 2020-01-29:
just noted when reading my own post that there is "available: no" for the codecs that don't work.
But why?

maboroshinokiseki, 2020-02-19:
I am having the same issue, want to know why and how to fix it.

GRbit, 2020-03-05:
The same thing with bluetooth headset.

Before install pulseaudio-modules-bt I've used this headset with SBC codec. To check this I've runned `sudo hcidump avdtp` and in logs I've clearly seend that SBC codec was used.

After installing pulseaudio-modules-bt, in hcidump I've seen that now aptX-HD codec is using. 

So, I definetely know that my device is capable with SBC and aptX-HD, but in `pactl list` I see this:
```
Card #6
	Name: bluez_card.00_02_5B_00_FF_01
	Driver: module-bluez5-device.c
	Owner Module: 29
	Properties:
		device.description = "ARTEXTREME"
		device.string = "00:02:5B:00:FF:01"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headphone"
		bluez.path = "/org/bluez/hci0/dev_00_02_5B_00_FF_01"
		bluez.class = "0x240418"
		bluez.alias = "ARTEXTREME"
		device.icon_name = "audio-headphones-bluetooth"
	Profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
		a2dp_source_sbc: High Fidelity Capture (A2DP Source: SBC) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aac: High Fidelity Capture (A2DP Source: AAC) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aptx: High Fidelity Capture (A2DP Source: APTX) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aptx_hd: High Fidelity Capture (A2DP Source: aptX HD) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: headset_head_unit
	Ports:
		headphone-output: Headphone (priority: 0, latency offset: 0 usec, available)
			Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd
		headphone-input: Bluetooth Input (priority: 0, latency offset: 0 usec, available)
			Part of profile(s): headset_head_unit, a2dp_source_sbc, a2dp_source_aac, a2dp_source_aptx, a2dp_source_aptx_hd
```
When I try to change profile
```
$ pactl set-card-profile 6 a2dp_sink_sbc
Failure: Invalid argument
```

But I can change profile to `a2dp_sink_aptx_hd` or `headset_head_unit`
```
$ pactl set-card-profile 6 a2dp_sink_aptx_hd
$ pactl set-card-profile 6 headset_head_unit
```

The problem is, that with aptX-HD the effective distance is way shorter than with sbc. Unfortunately I will remove `pulseaudio-modules-bt` until profile selection will be fixed =(

If some tests or dumps needed I can do any debugging, compiling, stracign or something like this.

Also here is hcidump of bluetooth pairing with my headset:

```
$ sudo hcidump avdtp
HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0x400
< AVDTP(s): Discover cmd: transaction 11 nsp 0x00
> AVDTP(s): Discover rsp: transaction 11 nsp 0x00
    ACP SEID 6 - Audio Sink
    ACP SEID 7 - Audio Sink
    ACP SEID 5 - Audio Sink
    ACP SEID 3 - Audio Sink
    ACP SEID 53 - Audio Source
[... 136 lines cut ...]
> AVDTP(s): Open rsp: transaction 9 nsp 0x00
< AVDTP(s): Start cmd: transaction 10 nsp 0x00
    ACP SEID 7
> AVDTP(s): Start rsp: transaction 10 nsp 0x00
< AVDTP(m): ver 2 cc 0 pt 96 seqn 0 time 0 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 1 time 588 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 2 time 1176 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 3 time 1764 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 4 time 2352 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 5 time 2940 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 6 time 3528 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 7 time 4116 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 8 time 4704 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 9 time 5292 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 10 time 5880 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 11 time 6468 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 12 time 7056 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 13 time 7644 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 14 time 8232 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 15 time 8820 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 16 time 9408 ssrc 1
...
...
...
```

KangoV, 2020-04-25:
Just installed the two .deb packages link at the bottom of #85 and am now listening to my B&O H9i's using the AAC sink. Sounds very good.

EHfive, 2020-06-14:
It's all in wiki. https://github.com/EHfive/pulseaudio-modules-bt/wiki#how-to-choose-codec-
Install bluez >= 5.51 for codec switching features.

closed 2020-06-14 by EHfive