EHfive/pulseaudio-modules-bt #97: Limited effective range of volume control

hmorris94, 2020-03-07:
I am using my Android phone to sink audio and only the very low end of the volume slider has an impact (roughly the first 10%, as if this module has a maximum of 10 and rounds 11-100 down). Under iOS, it's binary; anything above zero is effectively maximum volume. The behavior in the absolute_volume branch is the same.

EHfive, 2020-03-07:
Because sink (transfer audio from cellphone to pc ) absolute volume is not implemented in this module.

You have to manually disable `absolute volume` in your Android/IOS devices.

hmorris94, 2020-03-07:
Would the issue not have gone away using the absolute_volume branch? Disabling that feature is not necessarily an option on COTS devices - certainly not the ones I've tried, unfortunately.

EHfive, 2020-03-07:
> Would the issue not have gone away using the absolute_volume branch?

No. It's only activated when transfer audio data from PC/Pulseaudio.

hmorris94, 2020-03-11:
I see. Looking at Bluetooth HCI logs, you're right about absolute volume.

In contrast to the [previous absolute volume issue](https://github.com/EHfive/pulseaudio-modules-bt/issues/87), I think this fits mainline as sink volume control ties directly to Pulse and not device support. Where would one look towards adding this functionality?

hmorris94, 2020-05-10:
Fighting stale bot.

EHfive, 2020-06-07:
I would try to add an option which allows setting device volume gain after that device connected to module.

Or merge https://github.com/EHfive/pulseaudio-modules-bt/tree/absolute_volume but with some modifications if I got free times.

slgray, 2020-06-10:
Have same low volume issue on Airpods too, unfortunately Airpods do not have "button" to control the internal volume.

EHfive, 2021-01-18:
This project has been deprecated. #154

I would try proposing a fix for this on PipeWire.