EHfive/pulseaudio-modules-bt #100: Is AptX Low Latency supported?

katsar0v, 2020-03-25:
A lot of devices do not support normal aptX or aptX HD, they support aptX LL instead or FastStream. Are these planned, is it possible to have them includes?

EHfive, 2020-03-26:
Not planned.

closed 2020-03-26 by EHfive

ValdikSS, 2020-04-05:
@EHfive, actually aptX Low Latency is a subset of aptX, it's not a separate codec but just a special configuration with very low buffer value. It could should be trivial to implement.
Contact me if you need more details, I have a traffic dump somewhere.

katsar0v, 2020-04-09:
@ValdikSS Thank you a lot for your support. @EHfive aptX LL is actually quite a big percentage of the products using aptX so it could be a big feature. Thanks!

OkropNick, 2020-04-09:
If @ValdikSS is right, it would be very nice feature for activities like watching concerts or games playing.

pali, 2020-05-03:
Btw, I have already pulseaudio patches for aptX-LL support based on my pull request:
https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/227
I will publish these aptX-LL patches in new pull request after above pull request (on which they are based) is merged.

nohajc, 2020-09-28:
> I will publish these aptX-LL patches in new pull request after above pull request (on which they are based) is merged.

It would be great if you could publish those changes. I already use pulseaudio built from your pull request to set different SBC profiles (another super-useful feature!) and I'd like to try aptX Low Latency too.