EHfive/pulseaudio-modules-bt #102: install to user location instead of overwriting system modules

mwalle, 2020-04-01:
It would be better to install that into a user location. Eg. `$HOME/.local/lib/pulse-12.2/modules`. Then you can add the following line to your `$HOME/.config/pulse/daemon.conf`:

```
dl-search-path = /home/<user>/.local/lib/pulse-12.2/modules:/usr/lib/pulse-12.2/modules
```

Virtually all pulseaudio daemons runs as the logged in user. Doing it this way:

- you don't overwrite the files provided by the distribution
- you don't need superuser permissions at all (you can also install the ldacBT to ~/.local)

EHfive, 2020-04-02:
You can install them to specific folder by
```
$ make DESTDIR=$HOME/.local/local-root install
```

About ldacBT, you also need to set several [cmake-options](https://github.com/EHfive/ldacBT#cmake-options) and environment variable `PKG_CONFIG_PATH`

mwalle, 2020-04-02:
yes I know, but you also need to set the runtime linker path. It was just a suggestion to mention this install method in the README for other users because I think its a cleaner solution for systems with only one user.

closed 2020-06-06 by EHfive