EHfive/pulseaudio-modules-bt #103: Low volume on WH-1000XM3 headphones when using any other codec then HSP/HFP

adraz98, 2020-04-20:
I installed the module with the following parameter

load-module module-bluetooth-discover a2dp_config="ldac_eqmid=hq ldac_fmt=f32"

If you need any more detailed information, just tell me which command to insert on the terminal and I will provide it to you.

easyrider, 2020-04-26:
On my machine using LDAC profile it's pretty loud above volume > 80% (tidal).
AptX HD was similar in my case.
Some YT videos are quiet though.

adraz98, 2020-05-02:
I tested YouTube, Spotify and many other sources. In alle settings my audio was set to 100% on system and application level.

I use the endeavouros-distro on arch basis. Is there any way to increase the audio volume on the ldac profile?

berglh, 2020-05-05:
@adraz98 As far as I know, PulseAudio isn't aware of your Sony headphones volume level. You will need to manually change this on your headphones to an acceptable level. if your headphones are on a high level already, check your PulseAudio Volume Control volume levels. Each application in the newer version have discrete volume levels in the, for instance, you could listen to Spotify on loud but have Chrome browser on a lower volume this is set in the Playback tab. Each output device also has discrete volume levels in the Output Devices tab. If you have another default device, like plugged in speakers, just using the main volume control may be adjusting that default device rather than the device volume of your Sony headphones.

l0wl3vel, 2020-05-07:
I have the same headphones and they indeed have an internal volume, which is not exposed in pulseaudio. You can change it via dbus:

1. Open a D-Bus explorer like D-Feet

hbriese, 2020-05-10:
@l0wl3vel Thank you! Although I couldn't find such a D-Bus method (the only method in org.bluez.MediaControl1 being SetConfiguration) increasing the internal volume on my earphones did indeed work.

Prior to Ubuntu 20.04 the device's internal volume wasn't able to be changed (set to it's max?) - essentially just controllable through pulseaudio

berglh, 2020-05-10:
@hbriese So looking at the "manual", the normal method via bluetooth is to swipe up on the touch surface. When you do this, does it not actually change the volume when connected via bluez/PulseAudio? That is a pretty annoying design feature if that's the case. :disappointed: 

https://helpguide.sony.net/mdr/wh1000xm3/v1/en/contents/TP0001703116.html

My Sennheiser Momentum 3 have physical volume up and down buttons, and they adjust the internal headphone volume even if the call via bluetooth to adjust volume doesn't work on the bluetooth audio device - i.e. I don't see the any changes in PulseAudio mixer volumes when I press the buttons, but it increases the volume up to an internal max level anyway.

YihaoPeng, 2020-05-26:
In Android, there is a option called "Bluetooth absolute volume", when it is enabled, you can adjust the system volume by sliding on the headset. When it is disabled, the headset has an independent internal volume.

EHfive, 2020-06-07:
Move to #97

closed 2020-06-07 by EHfive