EHfive/pulseaudio-modules-bt #105: Headphones do not connect with pulseaudio-modules-bt (but work with default pulseaudio bluetooth module)

ildella, 2020-04-27:
I just installed Ubuntu 20.04

The MOMENTUM M2 AEBT worked ok, this was the hcidump:

```
HCI sniffer - Bluetooth packet analyzer ver 5.53
device: hci0 snap_len: 1500 filter: 0x400
< AVDTP(s): Discover cmd: transaction 11 nsp 0x00
> AVDTP(s): Discover rsp: transaction 11 nsp 0x00
    ACP SEID 3 - Audio Sink
    ACP SEID 1 - Audio Sink
< AVDTP(s): Set config cmd: transaction 12 nsp 0x00
    ACP SEID 1 - INT SEID 2
    Media Transport
    Media Codec - SBC
      44.1kHz 
      JointStereo 
      16 Blocks
      8 Subbands
      Loudness 
      Bitpool Range 2-53
```

Then I've installed 

```
sudo apt install libavcodec-extra libavcodec-extra58 pulseaudio-modules-bt
```

and after reboot the Momentum are there, but connect brings up the infamous `Failed to connect: org.bluez.Error.Failed`.

Tried unpair/pair, same result. 

Maybe can be related to the `libavcodec` packages I've installed? The `pulseaudio-modules-bt` apt package suggest to install libavcodec which does not exists, so I installed those 2.

I've re-installed `pulseaudio-module-bluetooth` and they're back. 

Please let me know what else I can provide for debug. 

Thanks.

ildella, 2020-04-27:
Oh, I see now Ubuntu 20.04 has v1.1.99 of this module, not the latest ones... I should probably build it myself then. 

This is probably a dup of https://github.com/EHfive/pulseaudio-modules-bt/issues/85

ildella, 2020-04-27:
I confirm it works perfectly with the latest .dev packages, recap is this bash script: 

```
wget https://f.bytefuse.net/pulseaudio-modules-bt-ppa/focal/libldac_2.0.2.3-1_amd64.deb
wget https://f.bytefuse.net/pulseaudio-modules-bt-ppa/focal/pulseaudio-modules-bt_1.3-1_amd64.deb
sudo apt install libavcodec-extra58 libfdk-aac1 -y
sudo dpkg -i libldac_2.0.2.3-1_amd64.deb
sudo dpkg -i pulseaudio-modules-bt_1.3-1_amd64.deb
```

Reboot and it worked for me.

closed 2020-04-27 by ildella

zedidous, 2020-05-03:
Man you made my day... I was battling so hard since i've upgraded my OS to have my bluetooth headset back working as before... Pfiou! Thanks again