EHfive/pulseaudio-modules-bt #106: LDAC Codec no longer works after upgrade to Ubuntu 20.04

daveman1010221, 2020-04-27:
Hello. Apologies in advance if this is the wrong place for this. A while back, I got this working under Ubuntu 19.10. After upgrading to 20.04, I get a message that the codec is unavailable. I've tried re-installing the codec but it doesn't seem to help. The bluetooth manager sees the device and attempts to connect but fails. Let me know if I can provide any other useful information, I'm just not sure how to troubleshoot this. Thanks.

libussa, 2020-05-01:
You can take a look here https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-618870211

berglh, 2020-05-05:
I am working for a PPA for `eoan` and `focal`, I will add a merge request for instructions on this once everything is tested as working. Maybe this issue should be closed now @daveman1010221.

daveman1010221, 2020-05-06:
It works for me, now. The problem I was having was a combination of issues. It seems the configuration of bluetooth codecs for the newer Pulseaudio has changed in signficant ways, too. Thanks for looking at this! I'm super happy to be able to use my Sony headphones again. :-)

closed 2020-05-06 by daveman1010221