EHfive/pulseaudio-modules-bt #107: Detect available codecs of a device

howl, 2020-05-03:
Perhaps this is not an issue of this project, but the default modes HSP and A2DP (SBC) get detected so the only available ones are shown in gnome sound control configuration.

When connecting a device it seems that the best mode is autoselected, I have an aptX headset and another aptX HD one, and when connecting one or another the corresponding codec is defaulted despite aptX or AAC could be used for the aptX HD one.

On both of them all the codecs are shown, and even you cannot tell if you could set up a codec that is not available for the devices, if you select for example LDAC for any of them, silently in the journal appears that is not supported, but the codec remains selected in the combo box.

So the issue is just to show in the combo the available codecs. I don't know if this is more a general PulseAudio issue or gnome volume control problem.

EHfive, 2020-05-11:
Unlike KDE, gnome volume control shows unavailable profiles in profile menu.

EHfive, 2020-06-07:
Due to the complexity of original pulseaudio bluetooth module source which this repo based on, it's not easy to change its design in state mutations, I am not going to change this behavior.

closed 2020-06-07 by EHfive