EHfive/pulseaudio-modules-bt #108: PPA Ubuntu 20.04 LTS

snevas, 2020-05-04:
Could the PPA include the packages for the latest released LTS release "Focal Fossa"?

lucasfijen, 2020-05-04:
See the following comment: [#85 (comment](https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-618870211)

berglh, 2020-05-05:
I am working on a PPA for `eoan` and `focal`. When I have the packages published I will need some people to test, then we can add the PPA to the instructions for `eoan` and `focal`.

snevas, 2020-05-05:
I am at your disposal to test when you are ready.

berglh, 2020-05-06:
@snevas I have published `focal` and `eaon` on my personal PPA here: https://launchpad.net/~berglh/+archive/ubuntu/pulseaudio-a2dp

Please do the purging and removal of existing packages first on this comment: https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-592915350

Then try to add the PPA using the instructions on the PPA page then:
```
sudo apt update
sudo apt install pulseaudio-modules-bt libldac
```
Then reboot, I think.

closed 2020-05-06 by snevas

Moosieboy, 2020-05-17:
> @snevas I have published `focal` and `eaon` on my personal PPA here: https://launchpad.net/~berglh/+archive/ubuntu/pulseaudio-a2dp
> 
> Please do the purging and removal of existing packages first on this comment: [#85 (comment)](https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-592915350)
> 
> Then try to add the PPA using the instructions on the PPA page then:
> 
> ```
> sudo apt update
> sudo apt install pulseaudio-modules-bt libldac
> ```
> 
> Then reboot, I think.
> 
> Please report back on issue #85

Works great. Many thanks for packaging!

teranex, 2020-05-22:
Thank you @berglh for packaging!