EHfive/pulseaudio-modules-bt #111: Sony WH-1000XM3 only a2dp_sink_sbc available

chillje, 2020-05-14:
Hi
I have installed pulseaudio-modules-bt and ldac successfully on my debian testing system.
Now I can see the A2DP codecs "ldac, aptx, aptx hd, aac" and not only the sbc.

But I cant change the profile from sbc (or hsp) to one of the others. I get "Failed to change profile" from blueman-applet.

pacmd shows the profiles are not available for my device:

```
    index: 8
	name: <bluez_card.38_18_4C_7D_51_24>
	driver: <module-bluez5-device.c>
	owner module: 30
	properties:
		device.description = "WH-1000XM3"
		device.string = "38:18:4C:7D:51:24"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_38_18_4C_7D_51_24"
		bluez.class = "0x240404"
		bluez.alias = "WH-1000XM3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (priority 30, available: unknown)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (priority 40, available: unknown)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (priority 40, available: no)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (priority 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (priority 40, available: no)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (priority 40, available: no)
		off: Off (priority 0, available: yes)
	active profile: <a2dp_sink_sbc>
	sinks:
		bluez_sink.38_18_4C_7D_51_24.a2dp_sink/#10: WH-1000XM3
	sources:
		bluez_sink.38_18_4C_7D_51_24.a2dp_sink.monitor/#16: Monitor of WH-1000XM3
	ports:
		headset-output: Headset (priority 0, latency offset 0 usec, available: unknown)
			properties:
				
		headset-input: Headset (priority 0, latency offset 0 usec, available: unknown)
			properties:
```

I tried to do it via pacmd
```
pacmd set-card-profile 8 a2dp_sink_ldac 
Failed to set card profile to 'a2dp_sink_ldac'.
```

Any ideas what I can try to get it to work?

mexus, 2020-05-14:
Hm, on my Arch it works:

```
    index: 6
        name: <bluez_card.38_18_4C_02_DE_F5>
        driver: <module-bluez5-device.c>
        owner module: 32
        properties:
                device.description = "WH-1000XM3"
                device.string = "38:18:4C:02:DE:F5"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_38_18_4C_02_DE_F5"
                bluez.class = "0x240404"
                bluez.alias = "WH-1000XM3"
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
        profiles:
                headset_head_unit: Headset Head Unit (HSP/HFP) (priority 30, available: unknown)
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (priority 40, available: unknown)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (priority 40, available: unknown)
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (priority 40, available: unknown)
                a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (priority 40, available: unknown)
                a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (priority 40, available: unknown)
                off: Off (priority 0, available: yes)
        active profile: <a2dp_sink_ldac>
        sinks:
                bluez_sink.38_18_4C_02_DE_F5.a2dp_sink/#7: WH-1000XM3
        sources:
                bluez_sink.38_18_4C_02_DE_F5.a2dp_sink.monitor/#10: Monitor of WH-1000XM3
        ports:
                headset-output: Headset (priority 0, latency offset 0 usec, available: unknown)
                        properties:

                headset-input: Headset (priority 0, latency offset 0 usec, available: unknown)
                        properties:
```

pulseaudio 13.0, linux-kernel 5.6.8

EHfive, 2020-05-15:
Switch to "High quality mode" in "Sony Headphones" IOS/Android app.

sylv-io, 2020-05-17:
Looks like @EHfive is right. You have to enable "High quality mode" first.

chillje, 2020-05-25:
Perfect works great after changing the profile in android app.

Short question before I close this issue.
Any chance to have the mic working in Linux with acceptable sound quality (not like HSP/HFP)?

YihaoPeng, 2020-05-26:
> Any chance to have the mic working in Linux with acceptable sound quality (not like HSP/HFP)?

Unless we can write the headset firmware ourselvse, I think we have no chance. The current headset firmware only supports specific Bluetooth profiles: HSP / HFP / A2DP, and multiple profiles cannot be used at the same time.

HSP / HFP is required for microphone access. But once you switch to HSP / HFP, the A2DP audio will be interrupted immediately.

In addition, the A2DP function of the headset is unidirectional (cannot be transmitted from the headset microphone to the computer).

As for HSP / HFP, I understand that it has only two codec: **CVSD** (required, sampling rate: 8kHz), **mSBC** (optional, sampling rate: 16kHz). **mSBC** should get better sound quality than **CVSD**, but much worse than any **A2DP** codec including **SBC**.

And the worse news is that I browsed all the official information of Sony, but I couldn't find the statement that WH-1000XM3 supports mSBC. So 8kHz CVSD may be the only microphone option available.

But even if it supports mSBC is far from good enough. Even with mSBC, HSP / HFP is **mono** in both directions!

So we still can only use it to talk, not for multiplayer games. Therefore, there is currently no official Bluetooth profile for multiplayer gamers and professional recording users. This is a question that Bluetooth Special Interest Group needs to consider.

ipeacocks, 2020-05-28:
@chillje check this https://askubuntu.com/a/1120106/217142
I've tried that with mine Sony WH1000XM3 and seems it's working but e.g. Zoom crashes when this option is on. Try it maybe it will be better for you.

closed 2020-06-06 by EHfive