EHfive/pulseaudio-modules-bt #112: Mpow Flame S, can't use aptX only aptX HD on ubuntu but can on arch?

ObeseChicken, 2020-05-21:
I have 2 systems. 1 is ubuntu 18.0.4 and one is arch. I've installed the codecs into both.

On my arch system with KDE I can switch between aptx, aptx HD and SBC. On ubuntu it fails when I try anything besides aptX HD.

Same thing with Flame 2, it would only let me use AAC on ubuntu.

```
HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0x400
< AVDTP(s): Discover cmd: transaction 7 nsp 0x00
> AVDTP(s): Discover rsp: transaction 7 nsp 0x00
    ACP SEID 7 - Audio Sink
    ACP SEID 5 - Audio Sink
    ACP SEID 1 - Audio Sink
< AVDTP(s): All Capabilities cmd: transaction 8 nsp 0x00
    ACP SEID 7
> AVDTP(s): All Capabilities rsp: transaction 8 nsp 0x00
    Media Transport
    Media Codec - non-A2DP (Unknown)
      32 00 00 00 00 
    Content Protection
      02 00 
    Delay Reporting
< AVDTP(s): All Capabilities cmd: transaction 9 nsp 0x00
    ACP SEID 5
> AVDTP(s): All Capabilities rsp: transaction 9 nsp 0x00
    Media Transport
    Media Codec - non-A2DP (aptX)
      44.1kHz 48kHz 
      Stereo 
    Content Protection
      02 00 
    Delay Reporting
< AVDTP(s): All Capabilities cmd: transaction 10 nsp 0x00
    ACP SEID 1
> AVDTP(s): All Capabilities rsp: transaction 10 nsp 0x00
    Media Transport
    Media Codec - SBC
      16kHz 32kHz 44.1kHz 48kHz 
      Mono DualChannel Stereo JointStereo 
      4 8 12 16 Blocks
      4 8 Subbands
      SNR Loudness 
      Bitpool Range 2-53
    Content Protection
      02 00 
    Delay Reporting
< AVDTP(s): Set config cmd: transaction 11 nsp 0x00
    ACP SEID 7 - INT SEID 2
    Media Transport
    Media Codec - non-A2DP (Unknown)
      22 00 00 00 00 
> AVDTP(s): Set config rsp: transaction 11 nsp 0x00
< AVDTP(s): Open cmd: transaction 12 nsp 0x00
    ACP SEID 7
> AVDTP(s): Open rsp: transaction 12 nsp 0x00
< AVDTP(s): Start cmd: transaction 13 nsp 0x00
    ACP SEID 7

```

Am I missing something?

ObeseChicken, 2020-05-21:
This is the error: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aptx: Unavailable profile

EHfive, 2020-06-06:
Maybe you need to install `ffmpeg`.

closed 2020-06-06 by EHfive