EHfive/pulseaudio-modules-bt #114: PPA Ubuntu 20.04

gvecchicert, 2020-06-01:
Ubuntu 20.04 is out. Will it be supported via PPA?

abach, 2020-06-03:
https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-551390000
https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-567175392

Using the recipes from the above comments on issue #85 about Ubuntu 19.10, I was able to build and use modules, changing step 3 in @berglh's instructions to:

```shell
sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential python3-dev libdbus-glib-1-dev
sudo apt install pulseaudio libavcodec58 libldac pulseaudio-modules-bt libbluetooth-dev libsbc-dev libpulse-dev git virtualenv build-essential python3-dev libdbus-glib-1-dev libgirepository1.0-dev pkg-config libtool sbc-tools libtool pkg-config libtool sbc-tools libtool cmake
```

and adding 

 ```shell
sudo apt install libavcodec-dev libfdk-aac-dev libldac-dev libavutil-dev
```

before step 7 from @zylxpl.

berglh, 2020-06-03:
@gvecchicert @abach I have created a PPA for 19.10 and 20.04, instructions are here: https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-625634213

EHfive, 2020-06-07:
Hey @berglh, thanks a lot for your contribution.

You could just fire an new issue to hold feeds, I would assign it to you.

closed 2020-06-07 by EHfive

gvecchicert, 2020-06-08:
> @gvecchicert @abach I have created a PPA for 19.10 and 20.04, instructions are here: _Edit_ [#85 (comment)](https://github.com/EHfive/pulseaudio-modules-bt/issues/85#issuecomment-624503956)
> The PPA: https://launchpad.net/~berglh/+archive/ubuntu/pulseaudio-a2dp

Thanks a lot.
May this be merged with official ubuntu binaries sooner or later?

berglh, 2020-06-08:
@gvecchicert At this stage I've only seen support for this being mentioned for AptX, I believe there is an ongoing license conflict between Sony LDAC (Apache) and pulseaudio (GPL), although I'm not entirely sure what the solution for this will be long term. Other codecs are being worked on by the looks of things :)

https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/227/commits

gvecchicert, 2020-06-08:
> @gvecchicert At this stage I've only seen support for this being mentioned for AptX, I believe there is an ongoing license conflict between Sony LDAC (Apache) and pulseaudio (GPL), although I'm not [...]

Thanks for your feedback!

mhalano, 2020-06-25:
> @gvecchicert At this stage I've only seen support for this being mentioned for AptX, I believe there is an ongoing license conflict between Sony LDAC (Apache) and pulseaudio (GPL), although I'm not [...]

The package libldac ara available for Debian Unstable (Sid) and Ubuntu N+1 (Groovy)

berglh, 2020-07-07:
@mhalano In the comments it says this:
```
This merge request does not support ldac codec.
And libldac (in Debian or Ubuntu) is irrelevant here as its license is incompatible with pulseaudio. So it cannot be used in pulseaudio for ldac support.
```
We will probably still continue building the pulseaudio-modules for LDAC support, but can hopefully not need to package libldac in the pulseaudio-modules-bt PPA :)

plumlis, 2020-09-21:
Will this ppa be upgrade to lastest version ?
it seems many bug is fixed and performance is improved.
https://github.com/EHfive/pulseaudio-modules-bt/commits/master

berglh, 2020-09-21:
@plumlis Sure thing, I'll do an updated build for v1.4 tag. Unfortunately, I'd prefer to wait for final version tags posted by @EHfive so that I can reference them via tagging rather than just building master which is just an arbitrary point of commits. When @EHfive is happy for a new release for the commits after March, I'll make another build then.

I also noticed in libLDAC repo, that some changes were made by @EHfive but no new tags there either. I'll need to set something up to update these on tag pushes.

berglh, 2020-09-21:
@plumlis The focal 20.04 build is being done now on LaunchPad: [https://launchpad.net/~berglh/+archive/ubuntu/pulseaudio-a2dp/+packages](https://launchpad.net/~berglh/+archive/ubuntu/pulseaudio-a2dp/+packages). When the v1.4 build is complete you should be able to install it and try it out - it seemed to load all modules in PulseAudio with my test build container. I will try it out on my laptop, let me know if you have any troubles.

berglh, 2020-09-21:
@plumlis OK, v1.4 is published and installable via the PPA. I saw an error in PulseAudio trying to load ofono:
```
Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files.
```
I'm not sure if this is related to any fixes that you were expecting or if this is just my machine specifically. Anyway, LDAC is working with my Sony speaker, so perhaps open an issue if you are having issues and tag me in it.

berglh, 2020-09-21:
Just a comment, I tried to update for `eoan`, but it's now considered End of Life and Ubuntu PPA won't let me build new packages for it. Sorry `eoan` users, please upgrade to `focal` ^_^.