EHfive/pulseaudio-modules-bt #117: Can not install module on Ubutntu 20.04 LTS

megalloid, 2020-07-30:
Distro: Linux megalloid-ubuntu 5.4.0-40-generic #44-Ubuntu SMP Tue Jun 23 00:01:04 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

-- Checking for module 'dbus-1'
--   No package 'dbus-1' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  CMakeLists.txt:44 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/megalloid/pulseaudio-modules-bt/build/CMakeFiles/CMakeOutput.log".

EHfive, 2020-07-30:
Install any `*-devel` packages relating to items in https://github.com/EHfive/pulseaudio-modules-bt#general-installation.

By the way, you could use https://github.com/EHfive/pulseaudio-modules-bt/wiki/Packages#ppaberglhpulseaudio-a2dp.

closed 2020-07-30 by EHfive

sergsagan, 2020-09-30:
Executing the command cmake:

cmake ..
-- Checking for module 'sbc'
--   No package 'sbc' found

help solve the problem