EHfive/pulseaudio-modules-bt #118: Infinite reconnect of headphones

anonymix007, 2020-08-07:
After installing module I'm getting infinite reconnection and this error in logs:

Process 14179 (pulseaudio) crashed and dumped core.

This usually indicates a programming error in the crashing program and
should be reported to its vendor as a bug.

Sometimes I was able to see headphones as output device in sound settings, but never aptX HD codec choice (headphones support it)

anonymix007, 2020-08-07:
Logs:
[log messages.txt](https://github.com/EHfive/pulseaudio-modules-bt/files/5042514/log.messages.txt)
[log messages1.txt](https://github.com/EHfive/pulseaudio-modules-bt/files/5042518/log.messages1.txt)

/var/lib/systemd/coredump/ directory contains about half GiB of lz4 files, zip: https://yadi.sk/d/LN1KTPZfZQrqRQ

EHfive, 2020-08-08:
Have you missed build step below ?
```
$ git -C pa/ checkout v`pkg-config libpulse --modversion|sed 's/[^0-9.]*\([0-9.]*\).*/\1/'`
$ git -C pa/ checkout v12.2 # it's v12.2 based on your log
```

anonymix007, 2020-08-08:
Maybe. 
Anyway, after second reinstall it worked well.
Thanks for support.

closed 2020-08-08 by anonymix007