EHfive/pulseaudio-modules-bt #119: Installing this breaks non-LDAC A2DP

ashic, 2020-08-09:
I installed this, and in blueman manager in archlinux, I can select my new Sony WH-1000XM4 headphones, and after connecting, can change the profile to LDAC. However, non-LDAC A2DP has stopped working. This is useful in two cases:

1. If I set the XM4 to connect to two devices, then LDAC isn't an option. In this case, it's flipping to headset mode with awful sound. 
2. Connecting another bluetooth set that has A2DP support but not LDAC means it can only connect in headset mode. 

With pulseaudio-modules-bt installed, I have the profiles listed in blueman-manager (AAC, SBC, aptX, LDAC), but it can't switch to any of them. With pulseaudio-bluetooth, there are only two options -  

* Headset Unit (HSP/HFP)
* High Fidelity Playback (A2DP sink)

With pulseaudio-modules-bt, there are more options, but they don't seems to work unless I'm connecting an LDAC headset. 

Any thought on how I might be able to have LDAC for the XM4, and the "default" A2DP for the other headphones?

EHfive, 2020-08-09:
Dup #107 in some point.

EHfive, 2020-08-09:
Use `pactl list cards` to list available profiles.

closed 2020-08-09 by EHfive

ashic, 2020-08-09:
I'm not sure this is entirely a duplicate. 

If I install this package, and the connect the Sony WF-1000XM3 earbuds, they connect as HSP/HFP. I see the other A2DP options this package adds, but none of them work. Running `pactl list cards` gives the following:

```Card #1
	Name: bluez_card.38_18_4C_81_B3_5A
	Driver: module-bluez5-device.c
	Owner Module: 23
	Properties:
		device.description = "LE_WF-1000XM3"
		device.string = "38:18:4C:81:B3:5A"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_38_18_4C_81_B3_5A"
		bluez.class = "0x240404"
		bluez.alias = "LE_WF-1000XM3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: headset_head_unit
	Ports:
		headset-output: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac
		headset-input: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit
```

If I then install the pulseaudio-bluetooth module (which removes this one), I get A2DP Sink back. `pactl list cards` returns this:

```
Card #1
	Name: bluez_card.38_18_4C_81_B3_5A
	Driver: module-bluez5-device.c
	Owner Module: 23
	Properties:
		device.description = "WF-1000XM3"
		device.string = "38:18:4C:81:B3:5A"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_38_18_4C_81_B3_5A"
		bluez.class = "0x240404"
		bluez.alias = "WF-1000XM3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
		a2dp_sink: High Fidelity Playback (A2DP Sink) (sinks: 1, sources: 0, priority: 40, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink
	Ports:
		headset-output: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit, a2dp_sink
		headset-input: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit

```

Is there something I can do so that when using this module, the "standard" A2DP sink is used for the WF-1000XM3?

EHfive, 2020-08-09:
The name `LE_WF-1000XM3` suggest that that device is using BLE instead of BR/EDR which A2DP depends on.
Maybe other bluetooth host like Android/IOS device is still connected with that WF-1000XM3 via BR/EDR.

ashic, 2020-08-09:
The name is LE_WF-1000XM3 while it is not connected. Once connecting, it changes to WF-1000XM3, and then once connected, it goes back to LE_WF-1000XM3. The audio profile is definitely A2DP sink though, as the non-A2DP is quite horrendous if manually selected. The XM3s can't connect to two devices simultaneously, and I've also tested with the phone turned completely off. The results are the same and repeatable. With pulseaudio-bluetooth, there's two profiles available - HSP/HSF and A2DP. Both work (A2DP being substantially better). With pulseaudio-modules-bt, there's more A2DP profiles available, but none of them work. What this means is that to use LDAC, I have to install pulseaudio-modules-bt, but then I lose any use of the WF-1000XM3 earbuds until I install pulseaudio-bluetooth. If there were some way to "tell" pulseaudio to use the "standard" A2DP sink for the XM3, that'd be sufficient for me to not have to worry about switching packages when switching headphones.

EHfive, 2020-08-10:
That might be a bug. Please file a [new issue](https://github.com/EHfive/pulseaudio-modules-bt/issues/new?assignees=EHfive&labels=bug&template=bug_report.md&title=%5BBUG%5D) for your "WF-1000XM3" issues.