EHfive/pulseaudio-modules-bt #121: [BUG] HSP/HSF is not available for my bluetooth headset.

ElbinP, 2020-08-15:
**Describe the bug**
HSP/HSF is not available for my bluetooth headset.
```
Card #2
        Name: bluez_card.00_1E_7C_92_1D_B6
        Driver: module-bluez5-device.c
        Owner Module: 24
        Properties:
                device.description = "Infinity Glide 500"
                device.string = "00:1E:7C:92:1D:B6"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_00_1E_7C_92_1D_B6"
                bluez.class = "0x240404"
                bluez.alias = "Infinity Glide 500"
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
        Profiles:
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink_sbc
        Ports:
                headset-output: Headset (priority: 0, latency offset: 0 usec, available)
                        Part of profile(s): a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac, headset_head_unit
                headset-input: Headset (priority: 0, latency offset: 0 usec, not available)
                        Part of profile(s): headset_head_unit
```
```
alsa-plugins-pulseaudio-1.2.2-1.fc32.i686
alsa-plugins-pulseaudio-1.2.2-1.fc32.x86_64
pulseaudio-libs-13.99.1-4.fc32.x86_64
pulseaudio-qt-1.2-1.fc32.x86_64
pulseaudio-module-x11-13.99.1-4.fc32.x86_64
pulseaudio-utils-13.99.1-4.fc32.x86_64
pulseaudio-libs-13.99.1-4.fc32.i686
pulseaudio-module-gsettings-13.99.1-4.fc32.x86_64
pulseaudio-libs-glib2-13.99.1-4.fc32.x86_64
pulseaudio-module-gconf-13.99.1-4.fc32.x86_64
kde-settings-pulseaudio-32.0-3.fc32.noarch
pulseaudio-13.99.1-4.fc32.x86_64
pulseaudio-module-bluetooth-freeworld-1.4-1.fc32.x86_64
```

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
Need to be able to select HSP/HSF for the headset profile.

**Environment (please complete the following information):**
 - Distribution  Fedora 32 
 - Pulseaudio Version 13.99
 - Version 1.41
 - Installing via rpmfusion repos

**Pulseaudio logs:** 
Logs attached

**`avinfo` of device:**
```
[elbin@acer-laptop ~]$ avinfo 00:1E:7C:92:1D:B6
Connecting ... 
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-52
Stream End-Point #2: Audio Sink 
        Media Codec: MPEG24
                Object Types: MPEG-2 AAC LC 
                Frequencies: 44.1kHz 48kHz 
                Channels: 1 2 
                Bitrate: 264630
                VBR: Yes
```
[journalctl-dump.txt](https://github.com/EHfive/pulseaudio-modules-bt/files/5078405/journalctl-dump.txt)

**Additional context**
Add any other context about the problem here.

EHfive, 2020-08-26:
In my case, it's
```
D: [pulseaudio][modules/bluetooth/module-bluez5-device.c:684 transport_acquire()] Acquiring transport /org/bluez/hci0/dev_38_18_4C_XX_XX_XX/fd22
I: [pulseaudio][modules/bluetooth/backend-native.c:152 sco_do_connect()] doing connect
E: [pulseaudio][modules/bluetooth/backend-native.c:155 sco_do_connect()] connect(): Function not implemented
D: [pulseaudio][../pulseaudio/src/pulsecore/card.c:338 pa_card_set_profile()] bluez_card.38_18_4C_BE_92_E5: active_profile: a2dp_sink_ldac -> off

```

casta, 2020-09-02:
Hello,

I have the same issue. Plain pulseaudio has the same issue.
There is a patch against pulseaudio-13 that fixes this and make possible to switch to HFP profile (found here : https://bugzilla.redhat.com/show_bug.cgi?id=1311238 but this refers to some pull requests referenced).

I confirm the fix, but with this, I loose apt-Xwith A2DP  provided by your bt moduels fork.

You may eventually port this patch to your fork ?

marek141017, 2020-10-11:
Hello

I need some help
My Ubuntu is 20.04 LTS
I did as was suggested here
But, when I connect my MPOW I got:
<pre>
	Name: bluez_card.F8_8A_3C_31_E2_31
	Driver: module-bluez5-device.c
	Owner Module: 14
	Properties:
		device.description = "Mpow X3"
		device.string = "F8:8A:3C:31:E2:31"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_F8_8A_3C_31_E2_31"
		bluez.class = "0x2c0404"
		bluez.alias = "Mpow X3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		a2dp_source_sbc: High Fidelity Capture (A2DP Source: SBC) (sinks: 0, sources: 1, priority: 20, available: yes)
		a2dp_source_aac: High Fidelity Capture (A2DP Source: AAC) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aptx: High Fidelity Capture (A2DP Source: APTX) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aptx_hd: High Fidelity Capture (A2DP Source: aptX HD) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink_aac
	Ports:
		headset-output: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac, headset_head_unit
		headset-input: Headset (priority: 0, latency offset: 0 usec, not available)
			Part of profile(s): a2dp_source_sbc, a2dp_source_aac, a2dp_source_aptx, a2dp_source_aptx_hd, headset_head_unit
</pre>

The most important part
headset_head_unit: available: no

I have only one source a2dp_source_sbc: High Fidelity Capture
When I try to change profile, I got
Failed to change profile to a2dp_source_sbc

Need some advise.
Thank you

sschuhmann, 2020-12-14:
I still have this issue

Mic92, 2021-01-03:
I had to follow this hack to make it available: https://wiki.archlinux.org/index.php/Bluetooth_headset#HFP_not_working_with_PulseAudio

EHfive, 2021-01-18:
This project has been deprecated. #154

There will be no bug fixing commits anymore.