EHfive/pulseaudio-modules-bt #124: [BUG] A2DP profiles unavailable for WH-1000XM4 when multipoint is enabled

nloewen, 2020-09-07:
When my WH-1000XM4 headphones have multipoint enabled, all A2DP profiles are unavailable. When multipoint is disabled on the headphones, SBC, AAC, and LDAC are available as expected. A2DP does work with pulseaudio-modules-bluetooth when multipoint is enabled.

Distro: Arch
Pulseaudio: 13.0
pulseaudio-modules-bt: 1.4
Installed from: AUR

<details>
<summary>Pulseaudio logs (multipoint off)</summary>

```
-- Logs begin at Thu 2020-08-06 16:58:52 CDT, end at Mon 2020-09-07 18:37:42 CDT. --
Sep 07 18:35:24 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Properties changed in device /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] backend-native.c: dbus: path=/Profile/HSPAGProfile, interface=org.bluez.Profile1, member=NewConnection
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] backend-native.c: dbus: NewConnection path=/org/bluez/hci0/dev_94_DB_56_63_D2_C6, fd=34, profile headset_head_unit
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: I: [pulseaudio] backend-native.c: doing listen
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_94_DB_56_63_D2_C6/fd34 state: disconnected -> idle
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] backend-native.c: Transport /org/bluez/hci0/dev_94_DB_56_63_D2_C6/fd34 available for profile headset_head_unit
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep1 found
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Codec: 0
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Capabilities size:4
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep1 valid; Codec name: SBC
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep3 found
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Codec: 0
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Capabilities size:4
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep3 valid; Codec name: SBC
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep4 found
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Codec: 2
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Capabilities size:6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep4 valid; Codec name: AAC
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep2 found
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Codec: 2
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Capabilities size:6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep2 valid; Codec name: AAC
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep5 found
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Codec: 2
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Capabilities size:6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep5 valid; Codec name: AAC
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep6 found
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Codec: 255
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Capabilities size:8
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep6 valid; Codec name: LDAC
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: dbus: path=/MediaEndpoint/A2DPSource/VENDOR/LDAC, interface=org.bluez.MediaEndpoint1, member=SelectConfiguration
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: selecing configuration
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 07 18:35:25 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.MediaTransport1 found, skipping
[... 601 lines cut ...]
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #3: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T
Stream End-Point #4: Audio Sink 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
```
</details>

lucasfijen, 2020-09-08:
I think that is a problem on the side of the headphone, from what I read at least ldac is disabled as soon as you turn on multipoint. You should get a pop up in the Sony app about that when enabling multipoint

EHfive, 2020-09-08:
Pulseaudio logs (multipoint on)
```
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: dbus: path=/MediaEndpoint/A2DPSource/VENDOR/LDAC, interface=org.bluez.MediaEndpoint1, member=SelectConfiguration
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: selecing configuration
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep1 removed
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep3 removed
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep4 removed
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep2 removed
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep5 removed
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep6 removed
Sep 07 18:47:05 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Properties changed in device /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 07 18:47:08 nathan-laptop pulseaudio[134261]: D: [pulseaudio] bluez5-util.c: Timeout expired, and device /org/bluez/hci0/dev_94_DB_56_63_D2_C6 still has disconnected profiles: a2dp_sink
```

`SelectConfiguration` is a dbus method called by bluez when A2DP negotiation started.

Based on your "`avinfo` of device (multipoint on)", LDAC is unavailable when "multipoint" is on, but bluez is still calling `SelectConfiguration` on LDAC endpoint (`/MediaEndpoint/A2DPSource/VENDOR/LDAC`).

It might be working with SBC and AAC endpoints registered only, try it by remove libldac and ffmpeg (`pacman -R libldac ffmpeg`).

nloewen, 2020-09-08:
I am not expecting LDAC to be available with multipoint on, but was expecting the other codecs to work. I uninstalled libldac and I can now make use of AAC with multipoint turned on. Switching to SBC breaking things and I have to reconnect to get the headphones working again, but that should be an avoidable issue, haha. I haven't tested without ffmpeg, because I need it for other things.

Here's the logs with multipoint on, and libldac uninstalled in case you feel it's worth looking into this more.
<details>
<summary>Pulseaudio logs (multipoint on, no libldac)</summary>

```
-- Logs begin at Thu 2020-08-06 16:58:52 CDT, end at Tue 2020-09-08 09:18:49 CDT. --
-- Logs begin at Thu 2020-08-06 16:58:52 CDT, end at Tue 2020-09-08 09:18:49 CDT. --
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Properties changed in device /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] backend-native.c: dbus: path=/Profile/HSPAGProfile, interface=org.bluez.Profile1, member=NewConnection
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] backend-native.c: dbus: NewConnection path=/org/bluez/hci0/dev_94_DB_56_63_D2_C6, fd=22, profile headset_head_unit
Sep 08 09:10:31 pulseaudio[188831]: I: [pulseaudio] backend-native.c: doing listen
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_94_DB_56_63_D2_C6/fd22 state: disconnected -> idle
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] backend-native.c: Transport /org/bluez/hci0/dev_94_DB_56_63_D2_C6/fd22 available for profile headset_head_unit
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep1 found
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Codec: 0
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Capabilities size:4
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep1 valid; Codec name: SBC
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep3 found
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Codec: 0
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Capabilities size:4
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep3 valid; Codec name: SBC
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep4 found
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Codec: 2
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Capabilities size:6
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep4 valid; Codec name: AAC
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep2 found
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: UUID: 0000110b-0000-1000-8000-00805f9b34fb
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Codec: 2
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Capabilities size:6
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Device: /org/bluez/hci0/dev_94_DB_56_63_D2_C6
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Sep endpoint /org/bluez/hci0/dev_94_DB_56_63_D2_C6/sep2 valid; Codec name: AAC
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
Sep 08 09:10:31 pulseaudio[188831]: D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
[... cut ...]
Sep 08 09:15:45 pulseaudio[188831]: I: [pulseaudio] remap.c: Using stereo to mono remapping
Sep 08 09:15:45 pulseaudio[188831]: D: [pulseaudio] memblockq.c: memblockq requested: maxlength=33554432, tlength=0, base=4, prebuf=0, minreq=1 maxrewind=0
Sep 08 09:15:45 pulseaudio[188831]: D: [pulseaudio] memblockq.c: memblockq sanitized: maxlength=33554432, tlength=33554432, base=4, prebuf=0, minreq=4 maxrewind=0
Sep 08 09:15:45 pulseaudio[188831]: I: [pulseaudio] source-output.c: Created output 11 "Peak detect" on blu

nloewen, 2020-09-08:
Not sure if this is related or a new issue. AAC audio cuts out periodically, and I have to switch profiles to OFF and back to AAC to get it going again.

<details>
<summary>pulseaudio logs</summary>

```
-- Logs begin at Thu 2020-08-06 16:58:52 CDT, end at Tue 2020-09-08 12:05:13 CDT. --
Sep 08 12:03:10 nathan-laptop pulseaudio[188831]: D: [bluetooth] sink-input.c: Requesting rewind due to corking
Sep 08 12:03:10 nathan-laptop pulseaudio[188831]: D: [pulseaudio] sink.c: bluez_sink.94_DB_56_63_D2_C6.a2dp_sink: state: RUNNING -> IDLE
Sep 08 12:03:10 nathan-laptop pulseaudio[188831]: D: [pulseaudio] sink.c: set_port() operation not implemented for sink 12 "bluez_sink.94_DB_56_63_D2_C6.a2dp_sink"
Sep 08 12:03:13 nathan-laptop pulseaudio[188831]: D: [bluetooth] protocol-native.c: Requesting rewind due to end of underrun.
Sep 08 12:03:13 nathan-laptop pulseaudio[188831]: D: [bluetooth] sink-input.c: Requesting rewind due to uncorking
Sep 08 12:03:13 nathan-laptop pulseaudio[188831]: D: [pulseaudio] module-suspend-on-idle.c: Sink bluez_sink.94_DB_56_63_D2_C6.a2dp_sink becomes busy, resuming.
Sep 08 12:03:13 nathan-laptop pulseaudio[188831]: D: [pulseaudio] sink.c: bluez_sink.94_DB_56_63_D2_C6.a2dp_sink: state: IDLE -> RUNNING
Sep 08 12:03:13 nathan-laptop pulseaudio[188831]: D: [pulseaudio] sink.c: set_port() operation not implemented for sink 12 "bluez_sink.94_DB_56_63_D2_C6.a2dp_sink"
Sep 08 12:04:53 nathan-laptop pulseaudio[188831]: D: [bluetooth] module-bluez5-device.c: Skipping 52834 us (= 9320 bytes) in audio stream
Sep 08 12:04:53 nathan-laptop pulseaudio[188831]: D: [bluetooth] protocol-native.c: Implicit underrun of 'Playback'
Sep 08 12:04:53 nathan-laptop pulseaudio[188831]: D: [bluetooth] protocol-native.c: Requesting rewind due to end of underrun.
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: I: [pulseaudio] protocol-native.c: Application "PulseAudio Volume Control" requests card profile change. card = bluez_card.94_DB_56_63_D2_C6, profile = off
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Device removed from object /org/pulseaudio/core1/sink12
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Sink removed from object /org/pulseaudio/core1/sink12
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.DevicePort removed from object /org/pulseaudio/core1/sink12/port0
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] sink-input.c: Starting to move sink input 40 from 'bluez_sink.94_DB_56_63_D2_C6.a2dp_sink'
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Stream removed from object /org/pulseaudio/core1/record_stream70
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: I: [pulseaudio] source-output.c: Freeing output 70 "Peak detect"
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] ratelimit.c: 140 events suppressed
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [bluetooth] sink-input.c: Have to rewind 11708 bytes on render memblockq.
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [bluetooth] sink-input.c: Have to rewind 23416 bytes on implementor.
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [bluetooth] sink.c: Requesting rewind due to started move
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: D: [pulseaudio] sink.c: bluez_sink.94_DB_56_63_D2_C6.a2dp_sink: state: RUNNING -> IDLE
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: I: [pulseaudio] sink-input.c: Trying to change sample spec
Sep 08 12:05:03 nathan-laptop pulseaudio[188831]: I: [pulseaudio] sink.c: Cannot update sample spec, SINK_IS_RUNNING, will keep using s16le and 44100 Hz
[... cut ...]
Sep 08 12:05:05 nathan-laptop pulseaudio[188831]: D: [pulseaudio] memblockq.c: memblockq sanitized: maxlength=4194304, tlength=4194304, base=4, prebuf=4, minreq=4 maxrewind=0
Sep 08 12:05:05 nathan-laptop pulseaudio[188831]: I: [pulseaudio] protocol-native.c: Final latency 88.22 ms = 40.00 ms + 48.22 ms
Sep 08 12:05:05 nathan-laptop pulseaudio[188831]: D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Stream added for object /org/pulseaudio/core1/record_stream74
Sep 08 12:05:13 nathan-laptop pulseaudio[188831]: I: [pulseaudio] module-card-restore.c: Synced.
```
</details>

dbenjaminmiller, 2020-09-09:
I have the same issue with libldac installed.

nikil, 2020-09-18:
Same here, with LDAC installed multipoint breaks A2DP for all codecs, but when I build pulseaudio-modules-bt with `cmake -DCODEC_LDAC=OFF ..` multipoint works fine with AAC.  It would be nice if somehow bluez was smart enough to not try to negotiate with an unavailable endpoint.

I haven't had any issues with AAC cutting out, but I just bought these headphones yesterday.  I'll report back if I see any issues.

Frederick888, 2020-11-07:
I'm experiencing something totally different under Arch.

When multipoint is enabled, I can use A2DP/AAC but not other profiles:
```
profiles:
        headset_head_unit: Headset Head Unit (HSP/HFP) (priority 30, available: unknown)
        a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (priority 40, available: unknown)
        a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (priority 40, available: yes)
        a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (priority 40, available: no)
        a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (priority 40, available: no)
        a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (priority 40, available: unknown)
        off: Off (priority 0, available: yes)
```

When multipoint is disabled however, I can only use HSP/HFP and there's actually no sound coming from the headphones.

If I run `pacmd set-card-profile bluez_card.94_DB_56_xx_xx_xx a2dp_sink_ldac`, the headphones then disappear from the available devices list. I had to re-connect to the headphones to use them again.

I also tried `-DCODEC_LDAC=OFF` but it made no difference.

Another problem with multipoint is that once I play something on my laptop, the headphones never switch back to my phone unless I disconnect them from my laptop.

EHfive, 2020-12-20:
Seems it's a bluez bug.
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/479#note_735270

EHfive, 2021-01-18:
This project has been deprecated. #154 

Please report this to bluez community. 

Close.

closed 2021-01-18 by EHfive

rene-dev, 2021-01-18:
https://github.com/bluez/bluez/issues/85