EHfive/pulseaudio-modules-bt #130: [BUG]

smirnov, 2020-10-07:
**Describe the bug**
Connecting the bluetooth headphones to laptop running Fedora 33 with Pulseaudio 13.99.0 and pulseaudio-module-bluetooth-freeworld 1.4-2 succeeds, then connection quickly drops before any sound is routed to the headphones.

**To Reproduce**
Steps to reproduce the behavior:
Pair the headphones, observe succesful pairing operation and a quick disconnect after that

**Expected behavior**
Headphones are paired and remain paired and are receiving sound from the laptop

**Environment (please complete the following information):**
 - Distribution Fedora 33
 - Pulseaudio Version 13.99.0
 - Version 1.4-2.fc33
 - Installing via rpmfusion repository

**Pulseaudio logs:**
(set `log-level` to `4` in `/etc/pulse/daemon.conf`)

Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSource/VENDOR/APTXHD
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSource/VENDOR/APTX
Oct 07 17:32:53 hostname pulseaudio[31202]: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name is not activatable
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSource/AAC
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSource/SBC
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSink/VENDOR/APTXHD
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSink/AAC
Oct 07 17:32:53 hostname bluetoothd[1088]: Endpoint registered: sender=:1.4520 path=/MediaEndpoint/A2DPSink/SBC
Oct 07 17:32:54 hostname systemd-coredump[31187]: Process 31003 (pulseaudio) of user 1000 dumped core.
                                                       
                                                       Stack trace of thread 31003:
                                                       #0  0x00007f50d39c0bc5 raise (libc.so.6 + 0x3dbc5)
                                                       #1  0x00007f50d39a98a4 abort (libc.so.6 + 0x268a4)
                                                       #2  0x00007f50d3ca6c05 pa_hashmap_put (libpulsecommon-13.99.so + 0x2ec05)
                                                       #3  0x00007f50c193af8f module_bluez5_device_LTX_pa__init (module-bluez5-device.so + 0xaf8f)
                                                       #4  0x00007f50d3d9d919 pa_module_load (libpulsecore-13.99.so + 0x2a919)
                                                       #5  0x00007f50c1ae8510 device_connection_changed_cb (module-bluez5-discover.so + 0x2510)
                                                       #6  0x00007f50d3d9a253 pa_hook_fire (libpulsecore-13.99.so + 0x27253)
                                                       #7  0x00007f50d3d22b73 pa_mainloop_dispatch (libpulse.so.0 + 0x26b73)
                                                       #8  0x00007f50d3d2327e pa_mainloop_iterate (libpulse.so.0 + 0x2727e)
                                                       #9  0x00007f50d3d23320 pa_mainloop_run (libpulse.so.0 + 0x27320)
                                                       #10 0x000055b7eabe2728 main (pulseaudio + 0x7728)
                                                       #11 0x00007f50d39ab1a2 __libc_start_main (libc.so.6 + 0x281a2)
                                                       #12 0x000055b7eabe330e _start (pulseaudio + 0x830e)
                                                       
                                                       Stack trace of thread 31012:
                                                       #0  0x00007f50d3a79b0e ppoll (libc.so.6 + 0xf6b0e)
                                                       #1  0x00007f50d3daaa04 pa_rtpoll_run (libpulsecore-13.99.so + 0x37a04)
                                                       #2  0x00007f50c1a9b1c8 thread_func.lto_priv.1 (libalsa-util.so + 0x331c8)
                                                       #3  0x00007f50d3cd07c8 internal_thread_func (libpulsecommon-13.99.so + 0x587c8)
                                                       #4  0x00007f50d36383f9 start_thread (libpthread.so.0 + 0x93f9)
                                                       #5  0x00007f50d3a84b03 __clone (libc.so.6 + 0x101b03)
                                                       
                                                       Stack trace of thread 31006:
                                                       #0  0x00007f50d3a79b0e ppoll (libc.so.6 + 0xf6b0e)
                                                       #1  0x00007f50d3daaa04 pa_rtpoll_run (libpulsecore-13.99.so + 0x37a04)
                                                       #2  0x00007f50c1a9b1c8 thread_func.lto_priv.1 (libalsa-util.so + 0x331c8)
                                                       #3  0x00007f50d3cd07c8 internal_thread_func (libpulsecommon-13.99.so + 0x587c8)
                                                       #4  0x00007f50d36383f9 start_thread (libpthread.so.0 + 0x93f9)
                                                       #5  0x00007f50d3a84b03 __clone (libc.so.6 + 0x101b03)
                                                       
                                                       Stack trace of thread 31009:
                                                       #0  0x00007f50d3a79b0e ppoll (libc.so.6 + 0xf6b0e)
                                                       #1  0x00007f50d3daaa04 pa_rtpoll_run (libpulsecore-13.99.so + 0x37a04)
                                                       #2  0x00007f50c1a99cef thread_func.lto_priv.0 (libalsa-util.so + 0x31cef)
                                                       #3  0x00007f50d3cd07c8 internal_thread_func (libpulsecommon-13.99.so + 0x587c8)
                                                       #4  0x00007f50d36383f9 start_thread (libpthread.so.0 + 0x93f9)
                                                       #5  0x00007f50d3a84b03 __clone (libc.so.6 + 0x101b03)
░░ Subject: Process 31003 (pulseaudio) dumped core

**`avinfo` of device:**
(guide: https://github.com/EHfive/pulseaudio-modules-bt/issues/31#issuecomment-462717049)

**Additional context**
Add any other context about the problem here.

perceival, 2020-10-08:
Have the same problem plus attempt to dnf install end up with unsuccessful installation despite the --allowerasing switch:
Error: Transaction test error:
  file /usr/lib64/pulse-13.99/modules/libbluez5-util.so from install of pulseaudio-module-bluetooth-13.99.2-1.fc33.x86_64 conflicts with file from package pulseaudio-module-bluetooth-freeworld-1.4-2.fc33.x86_64
  file /usr/lib64/pulse-13.99/modules/module-bluetooth-discover.so from install of pulseaudio-module-bluetooth-13.99.2-1.fc33.x86_64 conflicts with file from package pulseaudio-module-bluetooth-freeworld-1.4-2.fc33.x86_64
  file /usr/lib64/pulse-13.99/modules/module-bluetooth-policy.so from install of pulseaudio-module-bluetooth-13.99.2-1.fc33.x86_64 conflicts with file from package pulseaudio-module-bluetooth-freeworld-1.4-2.fc33.x86_64
  file /usr/lib64/pulse-13.99/modules/module-bluez5-device.so from install of pulseaudio-module-bluetooth-13.99.2-1.fc33.x86_64 conflicts with file from package pulseaudio-module-bluetooth-freeworld-1.4-2.fc33.x86_64
  file /usr/lib64/pulse-13.99/modules/module-bluez5-discover.so from install of pulseaudio-module-bluetooth-13.99.2-1.fc33.x86_64 conflicts with file from package pulseaudio-module-bluetooth-freeworld-1.4-2.fc33.x86_64

smirnov, 2020-10-08:
@perceival getting around the second error could be done with

dnf shell
install pulseaudio-module-bluetooth
remove pulseaudio-module-bluetooth-freeworld

perceival, 2020-10-08:
Thanks. Was doing that so far with rpm -e -nodeps.

vrossum, 2020-10-10:
I have similar issues with pulseaudio 13.99 on Ubuntu 20.10 (beta).
Bluetooth connects (receiver beeps), but then disconnects again.
I don't have the coredump.

The relevant part in the logs seems to be:
ct 10 20:51:07 localhost pulseaudio[4344]: dbus: path=/MediaEndpoint/A2DPSource/VENDOR/APTXHD, interface=org.bluez.MediaEndpoint1, member=SetConfiguration
Oct 10 20:51:07 localhost pulseaudio[4344]: Transport /org/bluez/hci0/dev_FC_58_FA_00_BB_65/sep7/fd0 state: disconnected -> idle
Oct 10 20:51:07 localhost pulseaudio[4344]: Loading module-bluez5-device path=/org/bluez/hci0/dev_FC_58_FA_00_BB_65 autodetect_mtu=0 a2dp_config="ldac_eqmid=hq ldac_fmt=f32"
Oct 10 20:51:07 localhost pulseaudio[4344]: Assertion 'h' failed at pulsecore/hashmap.c:138, function pa_hashmap_put(). Aborting.
Oct 10 20:51:07 localhost systemd[4291]: pulseaudio.service: Main process exited, code=killed, status=6/ABRT
Oct 10 20:51:07 localhost systemd[4291]: pulseaudio.service: Failed with result 'signal'.
Oct 10 20:51:08 localhost systemd[4291]: pulseaudio.service: Scheduled restart job, restart counter is at 1.
Oct 10 20:51:08 localhost pulseaudio[12243]: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
Oct 10 20:51:08 localhost pulseaudio[12243]: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
Oct 10 20:51:08 localhost pulseaudio[12243]: Timer slack is set to 50 us.
Oct 10 20:51:08 localhost pulseaudio[12243]: RealtimeKit worked.
Oct 10 20:51:08 localhost pulseaudio[12243]: Successfully gained nice level -11.

iafilatov, 2020-10-13:
FWIW I had the same crash (I think) and fixed it my checking out pulseaudio that exactly matched the installed version.
```
pulseaudio -vvv
```
Connect to a BT audio device
```
...
D: [pulseaudio] module-bluez5-discover.c: Loading module-bluez5-device path=/org/bluez/hci0/dev_00_06_78_33_F6_07 autodetect_mtu=0 a2dp_config=""
E: [pulseaudio] hashmap.c: Assertion 'h' failed at pulsecore/hashmap.c:138, function pa_hashmap_put(). Aborting.
```
Find out system pa version and check out
```
pulseaudio --version  # => pulseaudio 13.99.1
git -C pa/ checkout v13.99.1
```
Build and install normally. Restart pa
```
pulseaudio -k
```
Or with `-vvv`
```
...
D: [pulseaudio] module-bluez5-discover.c: Loading module-bluez5-device path=/org/bluez/hci0/dev_00_06_78_33_F6_07 autodetect_mtu=1 a2dp_config=""
I: [pulseaudio] module-card-restore.c: Restoring port latency offsets for card bluez_card.00_06_78_33_F6_07.
D: [pulseaudio] card.c: Looking for initial profile for card bluez_card.00_06_78_33_F6_07
D: [pulseaudio] card.c: a2dp_sink_sbc availability no
D: [pulseaudio] card.c: a2dp_sink_aac availability unknown
D: [pulseaudio] card.c: a2dp_sink_aptx availability no
D: [pulseaudio] card.c: a2dp_sink_aptx_hd availability no
D: [pulseaudio] card.c: a2dp_sink_ldac availability no
D: [pulseaudio] card.c: off availability yes
I: [pulseaudio] card.c: bluez_card.00_06_78_33_F6_07: active_profile: a2dp_sink_aac
I: [pulseaudio] card.c: Created 1 "bluez_card.00_06_78_33_F6_07"
...
```

vrossum, 2020-10-13:
Thank you.
Unfortunately
 git -C pa/ checkout v13.99.2
says that 13.99.2 is not on git (yet?). 
So I'll wait.

iafilatov, 2020-10-13:
It's there
```
git clone https://github.com/EHfive/pulseaudio-modules-bt.git
Cloning into 'pulseaudio-modules-bt'...
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 1021 (delta 1), reused 0 (delta 0), pack-reused 1010
Receiving objects: 100% (1021/1021), 384.00 KiB | 938.00 KiB/s, done.

cd pulseaudio-modules-bt/
git submodule update --init 
Submodule 'pa' (https://github.com/pulseaudio/pulseaudio.git) registered for path 'pa'
Cloning into '/home/iafilatov/pulseaudio-modules-bt/pa'...
Submodule path 'pa': checked out 'cdeac178019ed3e73840702cb1223df5edd9df43'

git -C pa/ checkout v13.99.2
Previous HEAD position was cdeac1780 format: Add support for Dolby TrueHD and DTS-HD HBR passthrough
HEAD is now at 870598036 build-sys: Bump gettext dependency
```

I mean, if you have 13.99.2 installed then it must've come from _somewhere_. And that's the official repo.

asavah, 2020-10-13:
You might want to learn about git submodules a bit:
```
git -C pa/ tag | grep v13
v13.0
v13.99.1
v13.99.2
```

Update pa/ submodule first eg.
```
git submodule update --remote
```

smirnov, 2020-10-13:
Well, then it seems that the modules were simply not built with correct PA. Latest version in RPMFusion 1.4.3 works just fine with PA 13.99.2, so this bug can be closed, especially since it's not a bug of the project itself, more of a delayed build.

closed 2020-10-13 by smirnov