EHfive/pulseaudio-modules-bt #131: [BUG] Pulse 13.99.2 profiles not work

IDeathByte, 2020-10-13:
After connecting headphones, all profiles exclude HFP not work

**To Reproduce**
1-pair headphones
2-try to switch profile

**Environment**
 - Distribution Manjaro (unstable branch) /Arch (stable)
 - Pulseaudio Version - 13.99.2
 - Version 1.4
 - Installing via AUR

**Pulseaudio logs:**

> D: [pulseaudio] module-suspend-on-idle.c: Source bluez_source.00_16_94_31_8F_B3.headset_head_unit becomes idle, timeout in 5 seconds.
D: [pulseaudio] module-switch-on-connect.c: Trying to switch to new source bluez_source.00_16_94_31_8F_B3.headset_head_unit
I: [pulseaudio] core.c: configured_default_source: echoCancel_source -> bluez_source.00_16_94_31_8F_B3.headset_head_unit
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.
I: [pulseaudio] core.c: default_source: echoCancel_source -> bluez_source.00_16_94_31_8F_B3.headset_head_unit
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.
D: [pulseaudio] backend-native.c: RFCOMM >> +VGM=15
I: [pulseaudio] module.c: Loaded "module-bluez5-device" (index: #30; argument: "path=/org/bluez/hci0/dev_00_16_94_31_8F_B3 autodetect_mtu=0 a2dp_config="sbc_min_bp=53 sbc_freq=44k sbc_cmode=dual sbc_sbands=8 sbc_blen=16"").
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module30
D: [pulseaudio] backend-native.c: RFCOMM << AT+IPHONEACCEV=2,1,9,2,0
I: [pulseaudio] module-suspend-on-idle.c: Source bluez_source.00_16_94_31_8F_B3.headset_head_unit idle for too long, suspending ...
D: [pulseaudio] source.c: bluez_source.00_16_94_31_8F_B3.headset_head_unit: suspend_cause: (none) -> IDLE
D: [pulseaudio] source.c: bluez_source.00_16_94_31_8F_B3.headset_head_unit: state: IDLE -> SUSPENDED
I: [pulseaudio] module-suspend-on-idle.c: Sink bluez_sink.00_16_94_31_8F_B3.headset_head_unit idle for too long, suspending ...
D: [bluetooth] module-bluez5-device.c: Releasing transport /org/bluez/hci0/dev_00_16_94_31_8F_B3/fd21
I: [bluetooth] backend-native.c: Transport /org/bluez/hci0/dev_00_16_94_31_8F_B3/fd21 released
D: [bluetooth] module-bluez5-device.c: Audio stream torn down
D: [pulseaudio] sink.c: bluez_sink.00_16_94_31_8F_B3.headset_head_unit: suspend_cause: (none) -> IDLE
D: [pulseaudio] sink.c: bluez_sink.00_16_94_31_8F_B3.headset_head_unit: state: IDLE -> SUSPENDED
D: [pulseaudio] source.c: bluez_sink.00_16_94_31_8F_B3.headset_head_unit.monitor: suspend_cause: (none) -> IDLE
D: [pulseaudio] source.c: bluez_sink.00_16_94_31_8F_B3.headset_head_unit.monitor: state: IDLE -> SUSPENDED
I: [pulseaudio] core.c: All sinks and sources are suspended, vacuuming memory
D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_00_16_94_31_8F_B3/fd21 state: playing -> idle
D: [pulseaudio] card.c: Setting card bluez_card.00_16_94_31_8F_B3 profile headset_head_unit to availability status unknown
D: [pulseaudio] device-port.c: Setting port headset-output to status unknown
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.
D: [pulseaudio] device-port.c: Setting port headset-input to status unknown
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.
I: [pulseaudio] protocol-native.c: Application "QPulse" requests card profile change. card = bluez_card.00_16_94_31_8F_B3, profile = a2dp_sink_sbc
W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_sbc: Not connected

**`avinfo` of device:**
Connect failed. Connection refused(111)

On version 13.0 all works fine
Similar reports on AUR pages:
https://aur.archlinux.org/packages/pulseaudio-modules-bt
https://aur.archlinux.org/packages/pulseaudio-modules-bt-git

For one user helped remove and re-pair headphones, but in my case it didn`t work

vrossum, 2020-10-13:
See bug #130.
Did you get 
git -C pa/ checkout v13.99.2
to work?

IDeathByte, 2020-10-13:
> Did you get
 git -C pa/ checkout v13.99.2
 to work?

yep i try. no changes

AwesomeIbex, 2020-10-15:
Same issue on manjaro KDE

IDeathByte, 2020-10-29:
I fixed this after install bluez-qt-git. all works fine now 
Manjaro KDE

EHfive, 2020-12-20:
Dose it work with pulseaudio 14.0?

IDeathByte, 2020-12-20:
Yep. Not sure about non-git bluez-qt

EHfive, 2020-12-20:
Surely it's not related with bluez-qt except potential dependency issues.

closed 2020-12-20 by EHfive