EHfive/pulseaudio-modules-bt #132: Add groovy package in ppa

soredake, 2020-10-15:
Ubuntu 20.10 will release in 22 october.
@berglh

berglh, 2020-10-17:
@EHfive You can assign this to me. @soredake There appear to be some issues with functionality against PulseAudio 13.99.2, which I am assuming might be present in Groovy. I was thinking about this earlier this week I need to package it up and try it out, lets track the build here. Can you add 20.10 and Ubuntu to the title to make it easy for everyone to see?

EHfive, 2020-10-17:
@berglh Great. Thanks for your effort.

berglh, 2020-10-20:
Ok, initial build is looking kind of promising.  I will try to push to the PPA this evening after work. :crossed_fingers: 

```bash
root@docker:/build# hciconfig
hci0:	Type: Primary  Bus: Virtual
	BD Address: 00:AA:01:00:00:23  ACL MTU: 192:1  SCO MTU: 0:0
	UP RUNNING 
	RX bytes:0 acl:0 sco:0 events:86 errors:0
	TX bytes:2948 acl:0 sco:0 commands:86 errors:0

root@docker:/build# pulseaudio -D
W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
root@docker:/build# systemctl status -l bluetooth 
* bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/bluetooth.service.d
             `-btvirt-dep.conf
     Active: active (running) since Tue 2020-10-20 02:07:23 UTC; 31s ago
       Docs: man:bluetoothd(8)
   Main PID: 177 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 19107)
     Memory: 2.3M
     CGroup: /system.slice/containerd.service/system.slice/bluetooth.service
             `-177 /usr/lib/bluetooth/bluetoothd

Oct 20 02:07:23 docker bluetoothd[177]: Bluetooth management interface 1.14 initialized
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSource/VENDOR/APTXHD
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSource/VENDOR/APTX
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSource/AAC
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSource/SBC
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSink/VENDOR/APTXHD
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSink/AAC
Oct 20 02:07:40 docker bluetoothd[177]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSink/SBC
root@docker:/build# cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.10
DISTRIB_CODENAME=groovy
DISTRIB_DESCRIPTION="Ubuntu Groovy Gorilla (development branch)"
root@docker:/build# pulseaudio --version
pulseaudio 13.99.2
root@docker:/build# bluetoothd --version
5.55
```

berglh, 2020-10-20:
@soredake @strawberrypie 

I've now updated the PPA to include a build of `pulseaudio_modules_bt` at version `1.4` and `libldac` at version `2.0.2.3`. The `pulseaudio_modules_bt` package built successfully and is just currently being published, so it should be available sometime in the next hour or so.

Please follow instructions for the `berglh` PPA to test and report back to me if you have any problems :ok_hand: 
https://github.com/EHfive/pulseaudio-modules-bt/wiki/Packages#ppaberglhpulseaudio-a2dp

berglh, 2020-10-20:
Looks like there is now an official libldac for Groovy, so might deprecate that from future builds.

ikeboy, 2020-10-22:
This doesn't work for me on 20.10. Package worked fine on 20.04. 

In pulseaudio, it shows the aptx formats but shows as (unavailable). 

[image]

berglh, 2020-10-23:
> This doesn't work for me on 20.10. Package worked fine on 20.04.
> 
> In pulseaudio, it shows the aptx formats but shows as (unavailable).
> 
> [image]

Hi Ikeboy, are you sure that the Mpox X3 supports AptX? Do they work with with SBC or AAC mode?

Thanks for testing :+1:

berglh, 2020-10-23:
@ikeboy Can you show us the output of `pactl list` where the item is the headphones in question?

ikeboy, 2020-10-23:
It worked in aptx under 20.04, or at least didn't say unavailable. Output:
```

Card #8
	Name: bluez_card.F8_8A_3C_32_3F_61
	Driver: module-bluez5-device.c
	Owner Module: 34
	Properties:
		device.description = "Mpow X3"
		device.string = "F8:8A:3C:32:3F:61"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_F8_8A_3C_32_3F_61"
		bluez.class = "0x2c0404"
		bluez.alias = "Mpow X3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		a2dp_source_sbc: High Fidelity Capture (A2DP Source: SBC) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aac: High Fidelity Capture (A2DP Source: AAC) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aptx: High Fidelity Capture (A2DP Source: APTX) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_source_aptx_hd: High Fidelity Capture (A2DP Source: aptX HD) (sinks: 0, sources: 1, priority: 20, available: no)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink_aac
	Ports:
		headset-output: Headset (type: Unknown, priority: 0, latency offset: -200000 usec, available)
			Part of profile(s): a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac, headset_head_unit
		headset-input: Headset (type: Unknown, priority: 0, latency offset: 0 usec, not available)
			Part of profile(s): a2dp_source_sbc, a2dp_source_aac, a2dp_source_aptx, a2dp_source_aptx_hd, headset_head_unit

```

ikeboy, 2020-10-23:
It works with SBC and AAC. 

Maybe it never worked with aptx but just didn't show the error on 20.04?

berglh, 2020-10-23:
@ikeboy I couldn't really tell you. There were issues with devices remembering previous profiles which was fixed with newer versions. It's possible that it wasn't communicated to you via the interface. Regardless, given the specifications of the device, I would expect it to fail to apply that profile due to the fact the device isn't advertising that profile as a capability (the available: no|yes section in the pactl output). Anyway, I'm glad to hear that you've got some sound working, that is a good start.

bmn001, 2020-10-23:
I upgraded to 20.10 from 20.04 with this installed.  Now none of my bluetooth devices will connect.

Blueman shows "Connection Failed: Protocol not available" when attempting to connect (or pair) various pairs of true wireless earbuds.

berglh, 2020-10-23:
@bmn001 I'm glad to hear it. I just updated my XPS laptop to 20.10 because of your issue.

I uncommented line one for the PPA in the `/etc/apt/sources.list.d/berglh-ubuntu-pulseaudio-a2dp-focal.list` file and both LDAC and AptX are working for me on `pulseaudio 13.99.2`.

Additionally, I tested `HSP/HFP` (still sounds terrible), `SBC`, `AAC` and it's all working just fine for me once I set the codec profile I wanted for each device.

@EHfive I think this issue is resolve now - I'm happy that all the primary codecs are working from my PPA.

```shell
Jul 22 07:02:53 bxps systemd[1]: Started Bluetooth service.
Jul 22 07:02:53 bxps bluetoothd[988]: Bluetooth management interface 1.15 initialized
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSource/VENDOR/APTXHD
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSource/VENDOR/APTX
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSource/AAC
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSource/SBC
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSink/VENDOR/APTXHD
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSink/AAC
Jul 22 07:02:54 bxps bluetoothd[988]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSink/SBC
```

### LDAC on Sony SRS-HG1 speaker
```shell
Card #1
	Name: bluez_card.E0_37_BF_09_F9_8B
	Driver: module-bluez5-device.c
	Owner Module: 24
	Properties:
		device.description = "bsrshg1"
		device.string = "E0:37:BF:09:F9:8B"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "speaker"
		bluez.path = "/org/bluez/hci0/dev_E0_37_BF_09_F9_8B"
		bluez.class = "0x240414"
		bluez.alias = "bsrshg1"
		device.icon_name = "audio-speakers-bluetooth"
	Profiles:
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: yes)
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink_ldac
	Ports:
		speaker-output: Speaker (type: Unknown, priority: 0, latency offset: 0 usec)
			Part of profile(s): a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac, headset_head_unit
		speaker-input: Bluetooth Input (type: Unknown, priority: 0, latency offset: 0 usec, not available)
			Part of profile(s): headset_head_unit
```

### Sennheiser Momentum 3 AptX LL
```shell
Card #2
	Name: bluez_card.00_1B_66_B1_01_55
	Driver: module-bluez5-device.c
	Owner Module: 25
	Properties:
		device.description = "MOMENTUM 3"
		device.string = "00:1B:66:B1:01:55"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_00_1B_66_B1_01_55"
		bluez.class = "0x240404"
		bluez.alias = "MOMENTUM 3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink_aptx
	Ports:
		headset-output: Headset (type: Unknown, priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac
		headset-input: Headset (type: Unknown, priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit
```

closed 2020-12-04 by soredake