EHfive/pulseaudio-modules-bt #133: Ubuntu 20.04 PPA wants to remove wine-stable and related packages when installing, why is that?

ZZZXXX0110, 2020-10-18:
When attempt to install the packages via the PPA on Ubuntu 20.04, these are the messages:

```
sudo apt install pulseaudio-modules-bt libldac
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cabextract fuseiso gstreamer1.0-plugins-base:i386 i965-va-driver:i386 intel-media-va-driver:i386 libaom0:i386
  libasn1-8-heimdal:i386 libasound2:i386 libasound2-plugins:i386 libasyncns0:i386 libavutil56:i386 libbrotli1:i386
  libcairo-gobject2:i386 libcairo2:i386 libcap2:i386 libcapi20-3 libcapi20-3:i386 libcdparanoia0:i386 libcodec2-0.9:i386
  libcurl3-gnutls:i386 libdatrie1:i386 libexif12:i386 libfaudio0 libflac8:i386 libfontconfig1:i386 libfribidi0:i386
  libgd3:i386 libgdbm-compat4:i386 libgdbm6:i386 libgdk-pixbuf2.0-0:i386 libglib2.0-0:i386 libgomp1:i386
  libgphoto2-6:i386 libgphoto2-port12:i386 libgraphite2-3:i386 libgsm1:i386 libgssapi3-heimdal:i386
  libgstreamer-plugins-base1.0-0:i386 libgstreamer1.0-0:i386 libharfbuzz0b:i386 libhcrypto4-heimdal:i386
  libheimbase1-heimdal:i386 libheimntlm0-heimdal:i386 libhx509-5-heimdal:i386 libicu66:i386 libieee1284-3:i386
  libigdgmm11:i386 libjack-jackd2-0:i386 libjbig0:i386 libjpeg-turbo8:i386 libjpeg8:i386 libkrb5-26-heimdal:i386
  libldap-2.4-2:i386 libltdl7:i386 libmp3lame0:i386 libmpg123-0:i386 libmspack0 libmysqlclient21:i386 libnghttp2-14:i386
  libnuma1:i386 libodbc1:i386 libogg0:i386 libopenal1:i386 libopenjp2-7:i386 libopus0:i386 liborc-0.4-0:i386
  libosmesa6:i386 libpango-1.0-0:i386 libpangocairo-1.0-0:i386 libpangoft2-1.0-0:i386 libpcap0.8:i386 libpci3:i386
  libperl5.30:i386 libpixman-1-0:i386 libpsl5:i386 libpulse0:i386 libroken18-heimdal:i386 librsvg2-2:i386
  librsvg2-common:i386 librtmp1:i386 libsamplerate0:i386 libsane:i386 libsasl2-2:i386 libsasl2-modules:i386
  libsasl2-modules-db:i386 libsdl2-2.0-0:i386 libshine3:i386 libsnappy1v5:i386 libsndfile1:i386 libsndio7.0:i386
  libsnmp35:i386 libsoxr0:i386 libspeex1:i386 libsqlite3-0:i386 libssh-4:i386 libstb0 libstb0:i386 libswresample3:i386
  libthai0:i386 libtheora0:i386 libtiff5:i386 libtwolame0:i386 libusb-1.0-0:i386 libv4l-0:i386 libv4lconvert0:i386
  libva-drm2:i386 libva-x11-2:i386 libva2:i386 libvdpau1:i386 libvisual-0.4-0:i386 libvkd3d1 libvkd3d1:i386
  libvorbis0a:i386 libvorbisenc2:i386 libvpx6:i386 libwavpack1:i386 libwayland-cursor0:i386 libwayland-egl1:i386
  libwebp6:i386 libwebpmux3:i386 libwind0-heimdal:i386 libwrap0:i386 libx264-155:i386 libx265-179:i386
  libxcb-render0:i386 libxcb-shm0:i386 libxcb-xfixes0:i386 libxcomposite1:i386 libxinerama1:i386 libxkbcommon0:i386
  libxml2:i386 libxpm4:i386 libxslt1.1:i386 libxss1:i386 libxvidcore4:i386 libzvbi0:i386 mesa-va-drivers:i386
  mesa-vdpau-drivers:i386 ocl-icd-libopencl1:i386 va-driver-all:i386 vdpau-driver-all:i386 wine-stable-amd64
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libavcodec-extra58 libfdk-aac1
Suggested packages:
  libldac
The following packages will be REMOVED:
  libavcodec58 libavcodec58:i386 libfaudio0:i386 pulseaudio-module-bluetooth wine-stable wine-stable-i386:i386
  winehq-stable winetricks
The following NEW packages will be installed:
  libavcodec-extra58 libfdk-aac1 libldac pulseaudio-modules-bt
0 upgraded, 4 newly installed, 8 to remove and 0 not upgraded.
Need to get 5,355 kB of archives.
After this operation, 459 MB disk space will be freed.
Do you want to continue? [Y/n]
```

Notice that it wants to remove wine-stable, wine-stable-i386:i386, winehq-stable, winetricks.

What does pulseaudio-modules-bt have to do with wine?

ZLima12, 2020-10-21:
I don't have much experience with apt, but it seems like it thinks that wine is only a dependency. If you can mark it as explicitly installed, perhaps it won't try to remove it.

berglh, 2020-10-22:
I will need to check this out in the build environment, installing wine first and then the package. My speculation is that libavcodec58 is a dependency for wine-stable taking.

berglh, 2020-10-23:
@ZZZXXX0110 ZLima12 has a good point, you can check the state of the package using `dpkg -l | grep -i wine`, it should say `ii` to the left of the package.

Looking at this from the other angle where I have the modules already installed, I seem to be able to install `wine-stable` just fine without getting any package conflicts, such as having to remove the modules before installing. Keep in mind I haven't added a WineHQ PPA to my system, this is just coming from the official Ubuntu PPA as far as I know.

```shell
The following additional packages will be installed:
  fonts-wine glib-networking:i386 gstreamer1.0-plugins-base:i386 gstreamer1.0-plugins-good:i386 gstreamer1.0-x:i386 i965-va-driver:i386 intel-media-va-driver:i386 libaa1:i386 libaom0:i386 libaribb24-0:i386
  libasn1-8-heimdal:i386 libasound2:i386 libasound2-plugins:i386 libasyncns0:i386 libavahi-client3:i386 libavahi-common-data:i386 libavahi-common3:i386 libavc1394-0:i386 libavcodec-extra58:i386 libavutil56:i386
  libblkid1:i386 libbrotli1:i386 libbz2-1.0:i386 libcaca0:i386 libcairo-gobject2:i386 libcairo2:i386 libcap2:i386 libcapi20-3 libcapi20-3:i386 libcdparanoia0:i386 libcodec2-0.9:i386 libcom-err2:i386
  libcups2:i386 libcurl3-gnutls:i386 libdatrie1:i386 libdb5.3:i386 libdbus-1-3:i386 libdv4:i386 libexif12:i386 libfaudio0 libfaudio0:i386 libflac8:i386 libfontconfig1:i386 libfreetype6 libfreetype6:i386
  libfribidi0:i386 libgcrypt20:i386 libgd3:i386 libgdbm-compat4:i386 libgdbm6:i386 libgdk-pixbuf2.0-0:i386 libglib2.0-0:i386 libglu1-mesa:i386 libgmp10:i386 libgnutls30:i386 libgomp1:i386 libgphoto2-6:i386
  libgphoto2-port12:i386 libgpm2:i386 libgraphite2-3:i386 libgsm1:i386 libgssapi-krb5-2:i386 libgssapi3-heimdal:i386 libgstreamer-plugins-base1.0-0:i386 libgstreamer-plugins-good1.0-0:i386 libgstreamer1.0-0:i386
  libgudev-1.0-0:i386 libharfbuzz0b:i386 libhcrypto4-heimdal:i386 libheimbase1-heimdal:i386 libheimntlm0-heimdal:i386 libhogweed5:i386 libhx509-5-heimdal:i386 libicu66:i386 libiec61883-0:i386 libieee1284-3:i386
  libigdgmm11:i386 libjack-jackd2-0:i386 libjbig0:i386 libjpeg-turbo8:i386 libjpeg8:i386 libk5crypto3:i386 libkeyutils1:i386 libkrb5-26-heimdal:i386 libkrb5-3:i386 libkrb5support0:i386 liblcms2-2:i386
  libldap-2.4-2:i386 libltdl7:i386 liblz4-1:i386 liblzma5:i386 libmount1:i386 libmp3lame0:i386 libmpg123-0:i386 libmysqlclient21:i386 libncurses6:i386 libncursesw6:i386 libnettle7:i386 libnghttp2-14:i386
  libnuma1:i386 libodbc1:i386 libogg0:i386 libopenal1:i386 libopencore-amrnb0:i386 libopencore-amrwb0:i386 libopenjp2-7:i386 libopus0:i386 liborc-0.4-0:i386 libosmesa6 libosmesa6:i386 libp11-kit0:i386
  libpango-1.0-0:i386 libpangocairo-1.0-0:i386 libpangoft2-1.0-0:i386 libpcap0.8:i386 libpci3:i386 libpcre2-8-0:i386 libpcre3:i386 libperl5.30:i386 libpixman-1-0:i386 libpng16-16:i386 libproxy1v5:i386
  libpsl5:i386 libpulse0:i386 libraw1394-11:i386 libroken18-heimdal:i386 librsvg2-2:i386 librsvg2-common:i386 librtmp1:i386 libsamplerate0:i386 libsane:i386 libsasl2-2:i386 libsasl2-modules:i386
  libsasl2-modules-db:i386 libsdl2-2.0-0:i386 libselinux1:i386 libshine3:i386 libshout3:i386 libslang2:i386 libsnappy1v5:i386 libsndfile1:i386 libsndio7.0:i386 libsnmp35:i386 libsoup2.4-1:i386 libsoxr0:i386
  libspeex1:i386 libsqlite3-0:i386 libssh-4:i386 libssl1.1:i386 libstb0 libstb0:i386 libswresample3:i386 libsystemd0:i386 libtag1v5:i386 libtag1v5-vanilla:i386 libtasn1-6:i386 libthai0:i386 libtheora0:i386
  libtiff5:i386 libtwolame0:i386 libusb-1.0-0:i386 libuuid1:i386 libv4l-0:i386 libv4lconvert0:i386 libva-drm2:i386 libva-x11-2:i386 libva2:i386 libvdpau1:i386 libvisual-0.4-0:i386 libvkd3d1 libvkd3d1:i386
  libvo-amrwbenc0:i386 libvorbis0a:i386 libvorbisenc2:i386 libvpx6:i386 libwavpack1:i386 libwayland-cursor0:i386 libwayland-egl1:i386 libwebp6:i386 libwebpmux3:i386 libwind0-heimdal:i386 libwine libwine:i386
  libwrap0:i386 libx264-155:i386 libx265-179:i386 libxcb-render0:i386 libxcb-shm0:i386 libxcb-xfixes0:i386 libxcomposite1:i386 libxcursor1:i386 libxi6:i386 libxkbcommon0:i386 libxml2:i386 libxpm4:i386
  libxrandr2:i386 libxrender1:i386 libxslt1.1:i386 libxv1:i386 libxvidcore4:i386 libzvbi0:i386 mesa-va-drivers:i386 mesa-vdpau-drivers:i386 ocl-icd-libopencl1:i386 va-driver-all:i386 vdpau-driver-all:i386 wine
  wine32:i386 wine64
Suggested packages:
  gvfs:i386 i965-va-driver-shaders:i386 libdv-bin:i386 oss-compat:i386 rng-tools:i386 libgd-tools:i386 gdbm-l10n:i386 gnutls-bin:i386 gphoto2:i386 gpm:i386 krb5-doc:i386 krb5-user:i386 libvisual-0.4-plugins:i386
  gstreamer1.0-tools:i386 jackd2:i386 libmyodbc:i386 odbc-postgresql:i386 tdsodbc:i386 unixodbc-bin:i386 libportaudio2:i386 opus-tools:i386 libraw1394-doc:i386 librsvg2-bin:i386 hplip:i386
  libsasl2-modules-gssapi-mit:i386 | libsasl2-modules-gssapi-heimdal:i386 libsasl2-modules-ldap:i386 libsasl2-modules-otp:i386 libsasl2-modules-sql:i386 sndiod:i386 speex:i386 gstreamer1.0-libav:i386
  gstreamer1.0-plugins-bad:i386 gstreamer1.0-plugins-ugly:i386 libvdpau-va-gl1:i386 nvidia-vdpau-driver:i386 nvidia-legacy-340xx-vdpau-driver:i386 nvidia-legacy-304xx-vdpau-driver:i386 q4wine winbind winetricks
  playonlinux wine-binfmt dosbox exe-thumbnailer | kio-extras wine32-preloader:i386 wine64-preloader
The following NEW packages will be installed:
  fonts-wine glib-networking:i386 gstreamer1.0-plugins-base:i386 gstreamer1.0-plugins-good:i386 gstreamer1.0-x:i386 i965-va-driver:i386 intel-media-va-driver:i386 libaa1:i386 libaom0:i386 libaribb24-0:i386
[... cut ...]
  libvorbisenc2:i386 libvpx6:i386 libwavpack1:i386 libwayland-cursor0:i386 libwayland-egl1:i386 libwebp6:i386 libwebpmux3:i386 libwind0-heimdal:i386 libwine libwine:i386 libwrap0:i386 libx264-155:i386
  libx265-179:i386 libxcb-render0:i386 libxcb-shm0:i386 libxcb-xfixes0:i386 libxcomposite1:i386 libxcursor1:i386 libxi6:i386 libxkbcommon0:i386 libxml2:i386 libxpm4:i386 libxrandr2:i386 libxrender1:i386
  libxslt1.1:i386 libxv1:i386 libxvidcore4:i386 libzvbi0:i386 mesa-va-drivers:i386 mesa-vdpau-drivers:i386 ocl-icd-libopencl1:i386 va-driver-all:i386 vdpau-driver-all:i386 wine wine-stable wine32:i386 wine64
The following packages will be upgraded:
  libfreetype6
```

closed 2020-12-31 by stale[bot]