EHfive/pulseaudio-modules-bt #134: ffmpeg_libs.c: Cannot open libavcodec library: libavcodec.so.58. /lib/i386-linux-gnu/libxvidcore.so.4: cannot make segment writable for relocation: Operation not permitted

porschemad911, 2020-10-22:
I compiled from source on Debian Sid using script as follows (disabling AAC /LDAC support at build time):

```
#!/bin/bash
# instal build deps
sudo apt install bluez-hcidump pkg-config cmake libtool libpulse-dev libdbus-1-dev libsbc-dev libbluetooth-dev git checkinstall libavcodec-dev

# backup original libraries
echo -e "Backup:\n"
MODDIR=`pkg-config --variable=modlibexecdir libpulse`
sudo find $MODDIR -regex ".*\(bluez5\|bluetooth\).*\.so" -exec cp -v {} {}.bak \;

## compile pulseaudio-modules-bt
temp_compile_dir=$(mktemp -d)
cd "$temp_compile_dir"

git clone https://github.com/EHfive/pulseaudio-modules-bt.git
cd pulseaudio-modules-bt
git submodule update --init
git -C pa/ checkout v`pkg-config libpulse --modversion|sed 's/[^0-9.]*\([0-9.]*\).*/\1/'`
mkdir build
cd build
cmake -DCODEC_AAC_FDK=OFF -DCODEC_LDAC=OFF ..
make

sudo checkinstall -D -d2 --install=yes --pkgname pulseaudio-module-bluetooth

# restart pulseaudio and bluetooth service
pulseaudio -k
sudo systemctl restart bluetooth.service

```

I get several of the following errors in /var/log/syslog after starting bluetooth.service.

> Oct 22 22:40:11 john-n10j pulseaudio[3041]: W: [pulseaudio] ffmpeg_libs.c: Cannot open libavcodec library: libavcodec.so.58. /lib/i386-linux-gnu/libxvidcore.so.4: cannot make segment writable for relocation: Operation not permitted

Kernel version is: 5.9.0-1-686-pae

Runtime libs installed are:
libavcodec58/unstable,now 7:4.3.1-4 i386
libavutil56/unstable,now 7:4.3.1-4 i386
pulseaudio-utils/unstable,now 13.0-5 i386
pulseaudio/unstable,now 13.0-5 i386
bluez-hcidump/unstable,now 5.55-1 i386
bluez-obexd/unstable,now 5.55-1 i386
bluez/unstable,now 5.55-1 i386
dbus/unstable,now 1.12.20-1 i386

porschemad911, 2020-10-22:
My headset is Sennheiser HD 4.50 BTNC. I can connect it via SBC codec and play fine, however aptX is unable to connect.

I am also running Lubuntu 18.04.5, on which I installed pulseaudio-modules-bt from `ppa:eh5/pulseaudio-a2dp`. This works fine for aptX with this headset.

EHfive, 2020-12-20:
Can you run `ldconfig -p | grep avcodec` ?
My outputs:
```
$ ldconfig -p | grep avcodec
        libavcodec.so.58 (libc6,x86-64) => /usr/lib/libavcodec.so.58
        libavcodec.so (libc6,x86-64) => /usr/lib/libavcodec.so
```

EHfive, 2021-01-18:
This project has been deprecated. #154