EHfive/pulseaudio-modules-bt #137: [BUG] bluetooth HSP/HFP does not work with autodetect_mtu=1

ghjp, 2020-11-07:
**Describe the bug**
When the feature autodetect_mtu is switched on (the default setting) no sound can be heard in a bluetooth headset. I have tested this with 5 different BT headsets. After a while the connection to the headset is dropped.

**To Reproduce**
Connect a bluetooth headset and switch pulseaudio card profile to HSP/HFP. Now play music and set the output to the headset.

**Expected behavior**
Sound/music can be heard in the headset.

**Environment (please complete the following information):**
 - Distribution [e.g. Ubuntu, Fedora, Arch ...]: Debian 10 Buster
 - Pulseaudio Version [e.g. v12.2]: 13.0
 - Bluetooth Version: 5.54
 - Installing via [e.g. manually compiled, precompiled package from xxx, ...] package manager

**Pulseaudio logs:**
(set `log-level` to `4` in `/etc/pulse/daemon.conf`)

**`avinfo` of device:**
(guide: https://github.com/EHfive/pulseaudio-modules-bt/issues/31#issuecomment-462717049)

**Additional context**
Before the autodetect_mtu feature was added the system behaved as expected.

ghjp, 2021-01-12:
This bug is still present!

EHfive, 2021-01-18:
This project has been deprecated. #154

There will no bug fixing commits anymore.