EHfive/pulseaudio-modules-bt #138: [BUG] Can't build with libpulse/pulseaudio v13.99.3

zaeph, 2020-11-09:
**Describe the bug**
The package doesn't build with libpulse/pulseaudio v13.99.3.  Downgrading libpulse/pulseaudio to v13.0 fixes the issue.  I didn't bisect to find the exact culprit.

**To Reproduce**
- Update libpulse/pulseaudio to v13.99.3.

**Environment (please complete the following information):**
 - Arch
 - libpulse/pulseaudio v13.99.3
 - [AUR package which fetches upstream](https://aur.archlinux.org/packages/pulseaudio-modules-bt-git/)

asavah, 2020-11-09:
It does build fine with v13.99.3 here, it's not arch but pulse and pulse-audio-modules-bt are both build from sources.
You may want to raise this issue with you distros package maintainer.
And BTW `The package doesn't build` is not a bug report but a waste of time.
If it doesn't build there has to be an error message which you fail to provide.

zaeph, 2020-11-09:
> And BTW The package doesn't build is not a bug report but a waste of time.

Apologies for wasting your time, then.

closed 2020-11-09 by zaeph