EHfive/pulseaudio-modules-bt #139: Module settings ignored? Cannot connect Bose in Dual Channel SBC mode.

aljazzair, 2020-11-24:
Hello and thank you for creating this module.

I am opening this issue and requesting help after reading through all the issues, researching extensively online and wiping out my previous operating system (Centos8.2) just to make this module work using its highest settings.

My new OS is a fresh install of Manjaro with latest kernels.

My headphones are Bose 700.

As it stands, I have tried and installed (pamac)  pulseaudio-bluetooth then pulse-audio-Bluetooth git.

**My goal is to connect my Bose 700 headphones using Dual Channel SBC XQ, but I cannot.**

In pavucontrol, I can see a selection of available codecs: LDAC -AAC - SBC AptX and AptxHD.

Bose 700 support both SBC and AAC and can connect using both.
I have edited **/etc/pulse/default.pa** with the following parameters:

`load-module module-bluetooth-discover a2dp_config="sbc_min_bp=38 sbc_max_bp=53 sbc_cmode=dual sbc_alloc=loudness sbc_sbands=8 sbc_blen=16 sbc_freq=44k"
`

I have played with the above parameters in order to reach stated objective of connecting in Dual Channel mode, to no avail. **It seems as if  this line is ignored by pulse audio.**

Initially I tried locking bitpool to 53 to no avail. I also tried 48K. Then I read online that Dual channel is best negotiated at a bitpool of 37 for 44100hz. So I settled on the range above. 

My headphones report the following supported modes:

 **sudo avinfo 4C:87:5D:XX:XX:XX**
```
Connecting ...
Stream End-Point #3: Audio Sink
        Media Codec: MPEG24
                Object Types: MPEG-2 AAC LC MPEG-4 AAC LC
                Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz
                Channels: 1 2
                Bitrate: 192000
                VBR: Yes
        Content Protection: SCMS-T
Stream End-Point #51: Audio Source
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000000a
                Vendor Specific Codec ID 0x0104
                Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x82 0xee 0x00
        Content Protection: SCMS-T
Stream End-Point #19: Audio Sink
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000000a
                Vendor Specific Codec ID 0x0104
                Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x82 0xee 0x00
        Content Protection: SCMS-T
Stream End-Point #49: Audio Source
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000000a
                Vendor Specific Codec ID 0x0103
                Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
        Content Protection: SCMS-T
Stream End-Point #17: Audio Sink
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000000a
                Vendor Specific Codec ID 0x0103
                Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
        Content Protection: SCMS-T
Stream End-Point #33: Audio Source
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 16Khz 32Khz 44.1Khz 48Khz
                Subbands: 4 8
                Blocks: 4 8 12 16
                Bitpool Range: 2-53
        Content Protection: SCMS-T
Stream End-Point #1: Audio Sink
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 16Khz 32Khz 44.1Khz 48Khz
                Subbands: 4 8
                Blocks: 4 8 12 16
                Bitpool Range: 2-53
        Content Protection: SCMS-T

Upon connection, the following is negotiated:

 **sudo hcidump avdtp**
HCI sniffer - Bluetooth packet analyzer ver 5.55
device: hci0 snap_len: 1500 filter: 0x400
< AVDTP(s): Discover cmd: transaction 14 nsp 0x00
> AVDTP(s): Discover rsp: transaction 14 nsp 0x00
    ACP SEID 3 - Audio Sink
    ACP SEID 51 - Audio Source
    ACP SEID 19 - Audio Sink
    ACP SEID 49 - Audio Source
    ACP SEID 17 - Audio Sink
    ACP SEID 33 - Audio Source
    ACP SEID 1 - Audio Sink
< AVDTP(s): Set config cmd: transaction 15 nsp 0x00
    ACP SEID 1 - INT SEID 5
    Media Transport
    Media Codec - SBC
      44.1kHz
      JointStereo
      16 Blocks
      8 Subbands
      Loudness
      Bitpool Range 2-53
> AVDTP(s): Set config rsp: transaction 15 nsp 0x00
< AVDTP(s): Open cmd: transaction 0 nsp 0x00
    ACP SEID 1
> AVDTP(s): Open rsp: transaction 0 nsp 0x00
< AVDTP(s): Start cmd: transaction 1 nsp 0x00
    ACP SEID 1

```
 So no matter what, my shiny **Bose 700 headpones are stuck in JointStereo,** and never report a Bitpool lock that is in line with the contents of pulseaudio default.pa settings.

 What I am doing wrong? **How can I connect the Bose headpones at maximum quality, Dual Channel SBC @44.1kz?**

Bose is a relatively popular brand, so the findings of this thread will likely be useful for many users in the future.
 Thanks in advance for your kind feedback.

aljazzair, 2020-12-05:
I would like to add that I just tested my hardware in a virtual machine with a distribution called Zenwalk, which incorporates SBC XQ patches to pulseaudio. Using the exact same bluetooth dongle, the Bose 700 headphones do connect in Dual Channel with 16/8 bands/lenghth and a Bitpool of 38. Quality is noticeably superior to regular SBC.
Therefore the headphones and dongle do support the higher SBC quality settings.
I can only conclude that there is an issue with way the software interacts with the hardware, but I am unable to identify the problem.
Any pointers would be much appreciated.
Thanks again for creating this module, and looking forward to any suggestion :)

aljazzair, 2020-12-07:
Hello
It seems no one can help so I am closing this issue.
I decided to use Pali's improvements instead, and they work well without issue for my setup:
https://gitlab.freedesktop.org/kapiwko/pulseaudio/-/issues/1
Thanks again for creating this module.

closed 2020-12-07 by aljazzair

mogenson, 2021-01-15:
Just adding that I have the same headphones and using the settings listed above I see the same issue.

From `btmon`:
```
< ACL Data TX: Handle 256 flags 0x00 dlen 18             #1271 [hci0] 14.903443
      Channel: 1093 len 14 [PSM 25 mode Basic (0x00)] {chan 0}
      AVDTP: Set Configuration (0x03) Command (0x00) type 0x00 label 9 nosp 0
        ACP SEID: 1
        INT SEID: 5
        Service Category: Media Transport (0x01)
        Service Category: Media Codec (0x07)
          Media Type: Audio (0x00)
          Media Codec: SBC (0x00)
            Frequency: 44100 (0x20)
            Channel Mode: Joint Stereo (0x01)
            Block Length: 16 (0x10)
            Subbands: 8 (0x04)
            Allocation Method: Loudness (0x01)
            Minimum Bitpool: 2
            Maximum Bitpool: 53
```

aljazzair, 2021-01-16:
> Just adding that I have the same headphones and using the settings listed above I see the same issue.
>
> From `btmon`:
>
> ```
> < ACL Data TX: Handle 256 flags 0x00 dlen 18             #1271 [hci0] [...]

Pali's solution above works. It involves replacing pulsaudio altogether.
You can also scrap Pulseaudio and go wit blue-alsa. 
Both alternative solutions allow Bose 700 to work in Dual Channel SBC XQ.