EHfive/pulseaudio-modules-bt #140: Incompatible with pulseaudio 14.0 release

andrewrembrandt, 2020-12-01:
Hi EHfive,

Many thanks for a fantastic library / project!

Unfortunately, I've just had to downgrade my pulseaudio install to 13.99.3 (-1) on Arch linux, as was unable to pair bluetooth devices since upgrading to pulseaudio 14.0.

The following log message is all I've gathered so far:
```
Dec 01 14:51:24 xxxxx bluetoothd[1128]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 04:52:C7:F3:75:88: Protocol not available
```
Let me know if you'd like more details / logs etc?

Many thanks,

Andrew

EHfive, 2020-12-01:
You need to rebuild it whenever minor/major version of PulseAudio has changed in your system.

andrewrembrandt, 2020-12-01:
Ah, thank you, works perfectly after rebuild (d'oh!)

Again, really appreciate the project/library. Let me know where I can donate something?

closed 2020-12-01 by andrewrembrandt