EHfive/pulseaudio-modules-bt #141: Packaging issue? Cannot install on Fedora Silverblue 33 using RPM Fusion 33?

JaneSmith, 2020-12-02:
I've just upgraded from Fedora Silverblue 32 to 33. Before the upgrade, I had to remove everything from RPM Fusion, and the RPM Fusion repos itself. After the upgrade, I re-added the latest RPM Fusion repos using the instructions on their website, rebooted, and then attempted to reinstall this using the following command from the wiki:

`rpm-ostree override remove pulseaudio-module-bluetooth --install pulseaudio-module-bluetooth-freeworld`

This results in the following error:

```
error: Could not depsolve transaction; 2 problems detected:
 Problem 1: cannot install both pulseaudio-13.99.2-1.fc33.x86_64 and pulseaudio-14.0-2.fc33.x86_64
  - package pulseaudio-module-bluetooth-freeworld-1.4-3.fc33.x86_64 requires libpulsecore-13.99.so()(64bit), but none of the providers can be installed
  - conflicting requests
 Problem 2: package pulseaudio-module-bluetooth-freeworld-1.4-3.fc33.x86_64 requires libpulsecommon-13.99.so()(64bit), but none of the providers can be installed
  - cannot install both pulseaudio-libs-13.99.2-1.fc33.x86_64 and pulseaudio-libs-14.0-2.fc33.x86_64
  - package gnome-bluetooth-1:3.34.3-1.fc33.x86_64 requires pulseaudio-module-bluetooth, but none of the providers can be installed
  - conflicting requests
```

It sounds like Fedora 33 is using `pulseaudio-14`, but this package expects `pulseaudio-13.99`? Could anyone help explain what's going on here? Thanks.

gombosg, 2020-12-16:
Hi, I'm the Fedora packager. I don't use Silverblue, but is there something like `--allowerasing` there? If the issue persists, please report at bugzilla.rpmfusion.org. Thanks!

JaneSmith, 2020-12-20:
I don't know what `--allowerasing` is, but it doesn't seem to exist. Fedora Silverblue uses the `rpm-ostree` package manager.

In any case, this issue seems to have gone away. I'm not sure why... It was definitely not working before, but now I am able to install it with no problems.

closed 2020-12-20 by JaneSmith