EHfive/pulseaudio-modules-bt #143: unsupported codecs on QCC3020 device

jaccc, 2020-12-05:
**Describe the bug**
all codecs aren't supported on Haylou GT1 Plus

**Environment (please complete the following information):**
Gentoo, net-wireless/bluez-5.55, media-sound/pulseaudio-13.0-r1

**`avinfo` of device:**
Stream End-Point #3: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Data: 0x32
		Vendor Specific Value (aptX)
			Frequencies: 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #2: Audio Sink 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 264630
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T
Stream End-Point #4: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x000000d7
		Vendor Specific Codec ID 0x0025
		Vendor Specific Data: 0x38
Stream End-Point #5: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0104
		Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x84 0x09 0xb6
Stream End-Point #6: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x000000d7
		Vendor Specific Codec ID 0x0026
		Vendor Specific Data: 0x07 0x06 0x00 0x00 0x38 0x17 0x02 0x35
Stream End-Point #7: Audio Source 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x000000d7
		Vendor Specific Codec ID 0x0025
		Vendor Specific Data: 0x38
Stream End-Point #8: Audio Source 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0104
		Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x84 0x09 0xb6
Stream End-Point #9: Audio Source 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x000000d7
		Vendor Specific Codec ID 0x0026
		Vendor Specific Data: 0x07 0x06 0x00 0x00 0x38 0x17 0x02 0x35
Stream End-Point #10: Audio Source 
	Media Codec: SBC
		Channel Modes: Mono 
		Frequencies: 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 16 
		Bitpool Range: 2-53

EHfive, 2020-12-20:
Have you installed ffmpeg & libldac?
Except SBC, aptX, AAC, your device also supports "Vendor Specific Codec ID 0x0025" -> aptX HD and other unknown codecs.

jaccc, 2020-12-20:
It doesn't support aptX HD according to Haylou and Qualcomm pages.
aptX HD should be 0x0024.

jaccc, 2020-12-20:
Found page https://helgeklein.com/blog/2020/09/how-to-check-which-bluetooth-a2dp-audio-codec-is-used-on-windows/
0x0104	- The CSR True Wireless Stereo v3 Codec ID for AAC

EHfive, 2020-12-20:
Oh, yes. Seems those are aptX Adaptive and Qualcomm ANC (Adaptive).

So you want support of those codecs?
Without any spec or en/decoder of those, it's not possible to achieve it.

EHfive, 2021-01-18:
This project has been deprecated. #154

I would try adding/proposing support for this on PipeWire.

closed 2021-01-18 by EHfive