EHfive/pulseaudio-modules-bt #144: [BUG] Unable to checkout 'cdeac178019ed3e73840702cb1223df5edd9df43' in submodule path 'pa'

raffaem, 2020-12-12:
**Describe the bug**
Harmless error in initializing submodule. 

**To Reproduce**
```
raffaele@raffaele-Nitro-AN515-52:/media/raffaele/DATA/data/linux/software$ git clone https://github.com/EHfive/pulseaudio-modules-bt.git
Cloning into 'pulseaudio-modules-bt'...
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 1021 (delta 1), reused 5 (delta 0), pack-reused 1010
Receiving objects: 100% (1021/1021), 384.17 KiB | 341.00 KiB/s, done.
Resolving deltas: 100% (599/599), done.
raffaele@raffaele-Nitro-AN515-52:/media/raffaele/DATA/data/linux/software$ cd pulseaudio-modules-bt
raffaele@raffaele-Nitro-AN515-52:/media/raffaele/DATA/data/linux/software/pulseaudio-modules-bt$ git submodule update --init
Submodule 'pa' (https://github.com/pulseaudio/pulseaudio.git) registered for path 'pa'
Cloning into '/media/raffaele/DATA/data/linux/software/pulseaudio-modules-bt/pa'...
error: unable to create file src/modules/alsa/mixer/samples/USB Device 0x46d:0x9a4--USB Mixer: Invalid argument
Unable to checkout 'cdeac178019ed3e73840702cb1223df5edd9df43' in submodule path 'pa'
```

**Expected behavior**
No error. However, the error is harmless. Following the subsequent instructions allow to compile and install the module. LDAC support is successfully installed

**Environment (please complete the following information):**
 - Distribution [e.g. Ubuntu, Fedora, Arch ...]: Kubuntu 20.10
 - Pulseaudio Version [e.g. v12.2]: pulseaudio 13.99.2
 - Version [e.g. v1.2]: version of what?
 - Installing via [e.g. manually compiled, precompiled package from xxx, ...]: git clone

**Pulseaudio logs:**
Tell me if those are necessary

**`avinfo` of device:**
(guide: https://github.com/EHfive/pulseaudio-modules-bt/issues/31#issuecomment-462717049)

The package `bluez-tools` does not include `avinfo`

```
raffaele@raffaele-Nitro-AN515-52:/media/raffaele/DATA/data/linux/software/pulseaudio-modules-bt$ dpkg-query -L bluez-tools
/.
/usr
/usr/bin
/usr/bin/bt-adapter
/usr/bin/bt-agent
/usr/bin/bt-device
/usr/bin/bt-network
/usr/bin/bt-obex
/usr/share
/usr/share/doc
/usr/share/doc/bluez-tools
/usr/share/doc/bluez-tools/README
/usr/share/doc/bluez-tools/changelog.Debian.gz
/usr/share/doc/bluez-tools/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/bt-adapter.1.gz
/usr/share/man/man1/bt-agent.1.gz
/usr/share/man/man1/bt-device.1.gz
/usr/share/man/man1/bt-network.1.gz
/usr/share/man/man1/bt-obex.1.gz
```

EHfive, 2020-12-12:
It probably related with file system. 
Assumed that `/media/raffaele/DATA/data/linux/software/` is on NTFS, the character `:` inside filename "USB Device 0x46d:0x9a4--USB Mixer" is invalid under NTFS.

closed 2020-12-12 by EHfive

raffaem, 2020-12-12:
Right, thanks!

But the module works anyway .. do I have to recompile?