EHfive/pulseaudio-modules-bt #147: Pipewire coming to Fedora

gombosg, 2020-12-16:
Hi @EHfive! I wanted to have a conversation here about pipewire.
It is coming to Fedora 34 next spring.
https://fedoraproject.org/wiki/Changes/DefaultPipeWire

Now, I think there are quite a lot of Fedora users for this package.
I'm unable to test Pipewire yet (I use F33 and can't install it properly).

Do you know how this package interacts with `pipewire-pulseaudio`?
Do you know if pipewire supports aptX already?

I heard that https://github.com/pali/libopenaptx is compatible with pipewire, so maybe I will start packaging it.

soredake, 2020-12-16:
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/249

EHfive, 2020-12-16:
> Do you know if pipewire supports aptX already?

aptX, LDAC (https://gitlab.freedesktop.org/pipewire/pipewire/-/merge_requests/341) is working on PipeWire . aptX HD still have problems. AAC support has not been finished yet.

> Do you know how this package interacts with `pipewire-pulseaudio`?
Basically `pipewire-pulseaudio` is just a meta package with a flag file (which enables pipewire's alternative pulseaudio server) inside.
In Arch, pipewire-pulseaudio provides (conflict with)  `pulseaudio` and `pulseaudio-bluetooth`.

> I heard that https://github.com/pali/libopenaptx is compatible with pipewire, so maybe I will start packaging it.
Absolutely.

gombosg, 2020-12-16:
Thanks for the details!
@soredake - wow, this is so recent... like it's being developed right now.
I personally think it's too early for F34 next spring, but they pushed through the change and looks like it will make it.

And we should provide some upgrade path for users...

> In Arch, pipewire-pulseaudio provides (conflicts with) pulseaudio and pulseaudio-bluetooth.
In Fedora, too.

So basically:
- aptX, LDAC - all OK
- aptX HD, AAC - keep an eye on it

And announce to users somehow during F34 upgrade which one is supported at that point?

gombosg, 2020-12-16:
@EHfive it's great to see you contributing to pipewire! I hope you'll have a much better experience than Pali had with PA.

EHfive, 2020-12-16:
> And we should provide some upgrade path for users...

I think it's quite easy to upgrade as long as bluez5 support is enabled by default on packaging.
End users can transparently replace pulseaudio with pipewire without any configuration requirements.

IDeathByte, 2020-12-21:
i hope extended sbc profiles will be supported in pipewire too :)
i have better experience on my headphones than with aptx (no quantum-noises and higher bitrate)

Thanks for your work :)

gombosg, 2020-12-21:
Well, for Fedora we have libldac and I started packaging [libopenaptx](https://bugzilla.redhat.com/show_bug.cgi?id=1908922). 
Looks like this `pulseaudio-modules-bt` will be removed during the upgrade, because pipewire replaces pulseaudio, and hence `pulseaudio-modules-bluetooth` as well.

soredake, 2020-12-21:
Only blocker for migrating to pipewire for me is [this ](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/495)and [this](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/348)

soredake, 2020-12-21:
Seems ubuntu and debian not ready for using pipewire as pulseaudio replacement :(
[image]

soredake, 2020-12-30:
https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/440