EHfive/pulseaudio-modules-bt #152: [BUG] error: pathspec 'v14.0' did not match any file(s) known to git

EvanCarroll, 2021-01-15:
Getting an error when I copy-paste commands,

```
$ git -C pa/ checkout v`pkg-config libpulse --modversion|sed 's/[^0-9.]*\([0-9.]*\).*/\1/'`
error: pathspec 'v14.0' did not match any file(s) known to git
```

This can be fixed with,

```
$ git submodule sync --recursive
```

After the `submodule init` to ensure that if already initialized, that the submodules are still in sync.

ZLima12, 2021-01-15:
If you solved your issue, why not share how you did it? That way, others who have the same issue will be able to solve it as well.

EvanCarroll, 2021-01-15:
Because I'm not sure of what actually solved the problem @ZLima12. I isolated the problem to having an out of date submodule. That could be fixed by removing the pa/ directory and rerunning `--init` on it. I think you can also run `git submodule sync --recursive`. I don't know that either of these made it work though, upon building I was getting (from `journalctl --user /usr/bin/pulseaudio`)

    pulseaudio[2523]: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files

and also,

    pulseaudio[147717]: Assertion 'h' failed at pulsecore/hashmap.c:138, function pa_hashmap_put(). Aborting.
    
So alas I removed the `pa/` directory rebuilt the whole thing and restarted. It works now, though I still get the above org.ofono message. but it's working the ldac codec works.

**The documentation should be updated such that it doesn't require a clone each time, but I'm unsure how to do it. Documentation improvements should be to run `git clone --recurse-submodules`, rather than a `git submodule --init`. Then we should have instructions to do `git submodule sync --recursive`, but with all that I was not able to confirm that it would fix my problem (though it should have worked). And now I can't recreate it.**

closed 2021-01-17 by EvanCarroll