EHfive/pulseaudio-modules-bt #154: Deprecation Announcement

EHfive, 2021-01-18:
Since I have already switched to PipeWire and I don't want to pay further effort on this. I am going to deprecate this project. The repo will be archived in January 25.

Although the modules are still working on current PulseAudio Releases(12.0 - 14.1) and later API compatible releases. I would recommend you to switch to vanilla PulseAudio or PipeWire. 
There is a WIP MR adding extra A2DP codecs support on PulseAudio https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/440?commit_id=e474e965e82e99cb21da0d77f0b0310a2b9aad5c, PulseAudio users may count on this.

soredake, 2021-01-18:
I hope ppa will be updated for 21.04, new pulseaudio version is not released, so 21.04 will have vanilla pulseaudio without codecs.

soredake, 2021-01-18:
@berglh will you update ppa to 21.04 for the last time?

bcremer, 2021-01-18:
Thanks for all the work you put in this on the last 2+ years.

berglh, 2021-01-18:
> @berglh will you update ppa to 21.04 for the last time?
@soredake Absolutely. I had a feeling that LDAC wasn't being included in the original MR for PulseAudio, but it now looks like the MR has a lease of new life. We'll keep an eye on this MR, if it's landed and functional for 21.04, I won't bother though.

lucasfijen, 2021-01-19:
@EHfive and @berglh Thank you so much for your effort on this repo! I've been enjoying ldac so much!

Glad it'll now be implemented in a more official way so that more people will be able to enjoy good audio:)

soredake, 2021-01-19:
> if it's landed and functional for 21.04

Feature freeze will happen [soon](https://discourse.ubuntu.com/t/hirsute-hippo-release-schedule/18539) :(

berglh, 2021-01-19:
@EHfive Thanks for all your personal help and guidance for bringing your excellent work to the Ubuntu users.
@soredake I will build closer to the release date, open a new issue and tag me in it when it's launch day. I'll upgrade my laptop and get it tested and built.

artizirk, 2021-01-19:
PulseAudio codecs support was just merged upstream! https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/440

soredake, 2021-01-20:
https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/476

Roger-ZRZ, 2021-01-22:
Sorry to ask, but are there stable alternatives we can use to stream AptX or LDAC for now?

EHfive, 2021-01-22:
> Sorry to ask, but are there any other stable alternatives we can use to stream AptX or LDAC for now? 

PipeWire.

https://gitlab.freedesktop.org/pipewire/pipewire/-/blob/master/NEWS

IDeathByte, 2021-01-25:
PipeWire supports extended SBC options now?